ipbus/ipbus-software #30: Allow readConfigurationSpace method to be detected in Python for IPbus-specific ClientInterface-derived classes

tswilliams, 2017-04-11:
Follow up from issue #19 

Currently, the `readConfigurationSpace` method is defined in the IPbus-specific `ClientInterface`-derived classes (specifically the `IPbusCore` class), but it cannot be used directly in Python when an IPbus-specific client is accessed through methods that return a `ClientInterface` reference/pointer - e.g for `HwInterface` object `myHw`, with URI protocol `ipbusudp-2.0`:
```
>>> myHw.getClient().readConfigurationSpace()
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'ClientInterface' object has no attribute 'readConfigurationSpace'
```

As a temporary fix, I introduced the Python-bindings-specific `readIPbusConfigurationSpace` function, with C++ signature:
```
uhal::ValWord< uint32_t > readIPbusConfigurationSpace (uhal::IPbusCore& aClient, const uint32_t& aAddr )
```

However, if possible, this Python-bindings-specific function should be removed, and succeeded by a language-agnostic solution

tswilliams, 2017-04-11:
Solution found by Alessandro: Declare the classes that are derived `ClientInterface` in the Python bindings, e.g:
```
class_<uhal::TCP<uhal::ControlHub<uhal::IPbus<(unsigned char)2, (unsigned char)0, 350u, false> >, 3ul>, bases<uhal::IPbusCore>,
         boost::noncopyable, /* no to-python converter (would require a copy CTOR) */
         boost::shared_ptr< uhal::TCP<uhal::ControlHub<uhal::IPbus<(unsigned char)2, (unsigned char)0, 350u, false> >, 3ul> > // /* all instances are held within boost::shared_ptr */
          >("TCP_CH_IPbus_2_0", no_init /* no CTORs */)
```

tswilliams, 2017-04-21:
Addressed in pull request #32

closed 2017-04-21 by tswilliams

alessandrothea, 2017-05-07:
Out of curiosity, why prepend the python class names with underscores, i.e. `_UDP_IPbus_1_3`, `_UDP_IPbus_2_0`, `_TCP_IPbus_1_3`, `_TCP_IPbus_2_0`, `_TCP_ControlHub_IPbus_1_3`, `_TCP_ControlHub_IPbus_2_0`?

tswilliams, 2017-05-07:
Re: prepending class name with underscores ... A leading underscore is the Python convention for private methods/classes - the purpose is to indicate to end-users that they should never have to use those classes directly, or be aware of their existance.

alessandrothea, 2017-05-07:
Ok, I see your point, but I don't see how to prevent users from using them directly.
Now that they are declared in python,`HwInterface.getClient()` returns the derived type, not a `ClientInterface` instance anymore.

```
In [11]: udp2.getClient()
Out[11]: <uhal._core._UDP_IPbus_2_0 at 0x3095fa0>
```