ipbus/ipbus-software #62: Provide a mechanism for building IPBus on non-RPM systems ( e.g. Ubuntu )

DavidCussans, 2017-10-06:
I would like to make use of IPBus on Linux distributions that aren't based on RPM . Mainly Ubuntu.

Would it be possible to provide a "Cmake" or similar mechanism for building Cactus / uHAL ?

arummler, 2017-10-06:
I am working with David on this project (eudaq/tlu) that requires uhal and I support the idea to go away from hand-made Makefiles tailored to specific distributions. I would be willing to look into a cmake based solution if you like (and as far time allows).

tswilliams, 2017-10-06:
Hi,

As the library-/executable-building sections of the uHAL makefiles are not performing any actions specific to the RPM-based linux distributions, it should be relatively easy to update the makefiles to build on ubuntu. When I tried building the current tip of the `master` branch on ubuntu 16, I noticed a couple of problems that were quickly fixed by adding a few extra arguments to the make command; there was one remaining problem when linking the libraries, which I could not quickly address, so I'll let you know once that's fixed.

Regarding which buildsystem to use, we are considering various options to improve upon the current situation (including reorganised version of the existing makefiles, cmake, and scons), but we've yet to decide between the different options.

Cheers,
Tom

arummler, 2017-10-08:
Out of curiosity I made a cmake prototype:
https://github.com/arummler/ipbus-software/tree/master-cmake
It is only tested on Ubuntu 16.04 and works with eudaq:
https://github.com/arummler/eudaq/tree/master-fmctlu-using-cmaked-uhal
In order to make an out-of-source build create a folder e.g. uhal-build, cd into it:

cmake -DCMAKE_INSTALL_PREFIX=/home/rummler/Software/ipbus-software-install -Dpugixml_DIR=/home/rummler/Software/ipbus-software/extern/pugixml/pugixml-1.8-install/lib/cmake/pugixml/ ../ipbus-software/uhal/

replace install, build and source folders accordingly.
To use with eudaq 

cmake -DCMAKE_PREFIX_PATH=/home/rummler/Software/ipbus-software-install ..

in build folder.

**Restrictions**: Only uhal library, no gui, tests, packaging (cpack) --> Depending on the extend of packaging variants it would take some days. I updated pugixml to the newer release 0.18 as it has a proper cmake config file, there did not seem to be any side effects during compilation, due a lack of hardware and tests I could test more. I used the system boost. Both boost and pugixml are already available packaged but override is not difficult if required. In ProtocolTCP.hxx and ProtocolUDP.hxx there was a problem with the signature of write_callback and read_callback, which requires two arguments. In one case I think I found the size in the other I put in a place  holder. It is only used in the error case so it should work anyhow.

DavidCussans, 2017-10-09:
Andre,

   Great! Many thanks. We will attempt to try this out in DESY this week.

David

Dr David Cussans                    University of Bristol,
office: +44-(117)-95 46879   H.H. Wills Physics Laboratory,
     lab: +44-(117)-33 17199   Tyndall Avenue,
     fax: +44-(117)-925 5624   Bristol BS8 1TL, UK

https://github.com/arummler/ipbus-software/tree/master-cmake
https://github.com/arummler/eudaq/tree/master-fmctlu-using-cmaked-uhal

tswilliams, 2017-10-12:
Hi Andre,

Many thanks for sharing your work. At the moment, I just have a few short comments:
* Regarding pugixml, I think it would probably be simpler to just install the system version using `apt-get`, as you have done with boost. I believe that uHAL should work fine with any versions of pugixml from 1.2 onwards.
 * The `uhal/uhal` package should be compiled with `-DDISABLE_PACKET_COUNTER_HACK -DRUN_ASIO_MULTITHREADED` in the arguments for g++ ; with these additional macros defined, there should be no need to modify `ProtocolTCP.hxx` or `ProtocolUDP.hxx` (These extra flags are easier to spot in a cleaned up version of the makefiles that can be found at https://github.com/ipbus/ipbus-software/tree/task/rationalise-uhal-makefiles )
 * If you do manage to get the test package compiling with cmake, then you can run some of the unit tests as follows:
{{{
source uhal/tests/setup.sh
uhal_test_suite.py -v -c uhal/tests/etc/uhal/tests/dummy_connections.xml -s "2.0 udp"
}}}

Cheers,
Tom

tswilliams, 2018-01-08:
Hi,

I have added a linux-distribution-agnostic install target to the IPbus software Makefiles, and merged this change into the `master` branch.

Essentially, after running `make` to compile the software, you can now run `sudo make install` to install the software. The default installation directory is `/opt/cactus`, but this can be changed by specifying make variables in the `make install` command. Instructions (including how to set a non-default installation directory) can be found at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/compile.html#instructions 

Please let me know if you encounter any problems when following these instructions, or if you have any other requests/suggestions.

Cheers,
Tom

tswilliams, 2018-01-10:
Closing now proactively, since "install" command has been implemented.

Please re-open and/or post a comment if you encounter any problems when following the instructions at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/compile.html#instructions , or if you have any other requests/suggestions

closed 2018-01-10 by tswilliams