ipbus/ipbus-software #118: Exception NoBranchFoundWithGivenUID on Ubuntu 16.04

darcato, 2018-06-15:
Hi,
we are trying to execute ipbus-software on ubuntu 16.04. 
The program fails at startup with the following exception:
```
terminate called after throwing an instance of 'uhal::exception::NoBranchFoundWithGivenUID'
  what():  No branch found with ID-path " ����" from node "TOP.sl1"
Not even a partial match found for ID-path " ����" from node "TOP.sl1". If this address looks correct, please check for leading, trailing and stray whitespace.

Aborted (core dumped)
```
The string is formatted correctly and same code on centos 7 is working fine. 

Do you have any suggestion?
Thank you,
Davide

tswilliams, 2018-06-15:
Hi,

I haven't encountered any problems like this before, so I need to ask a few questions:
 * Which version of uHAL are you using on each of the two operating systems?
 * Are you using the same address table as you used on CentOS7?
 * Are you calling the uHAL functions from C++ code, or Python?
 * Can you re-run your executable/script with the uHAL log threshold set to the 'info' level, and post the resulting output?

In C++, you can set the log threshold to 'info' as follows:
```
uhal::setLogLevelTo(uhal::Info());
```
... and in Python:
```
uhal.setLogLevelTo( uhal.LogLevel.INFO )
```

Cheers,
Tom

darcato, 2018-06-18:
Hi,
thanks for the answer. 
- We are using version v2.6.1 installed following the official guide.
- Yes, the xml file is the exactly the same.
- We are using C++
- The message error is:
```
15-06-18 16:36:49.738013 [7f3a05500740] ERROR - No branch found with ID-path "�"�w�" from node "TOP.sl1"
15-06-18 16:36:49.738040 [7f3a05500740] ERROR - Not even a partial match found for ID-path "�"�w�" from node "TOP.sl1". If this address looks correct, please check for leading, trailing and stray whitespace.
15-06-18 16:36:49.738048 [7f3a05500740] ERROR - Tree structure is:
+ Node "sl1", Address 0x00000080, Tags "slave1", Description "memory space", Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "FMC_CTRL_REG", SINGLE register, Address 0x000000D3, Mask 0xFFFFFFFF, Permissions rw, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "FMC_DATA1", SINGLE register, Address 0x000000D4, Mask 0xFFFFFFFF, Permissions r-, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "FMC_DATA2", SINGLE register, Address 0x000000D5, Mask 0xFFFFFFFF, Permissions r-, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "FMC_DATA3", SINGLE register, Address 0x000000D6, Mask 0xFFFFFFFF, Permissions r-, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "FMC_DATA4", SINGLE register, Address 0x000000D7, Mask 0xFFFFFFFF, Permissions r-, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "BPM_CTRL_REG", SINGLE register, Address 0x000000D8, Mask 0xFFFFFFFF, Permissions rw, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "BPM_RESET_REG", SINGLE register, Address 0x000000D9, Mask 0xFFFFFFFF, Permissions rw, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "BPM_ENABLE_REG", SINGLE register, Address 0x000000DA, Mask 0xFFFFFFFF, Permissions rw, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
  + Node "BPM_DELAY", SINGLE register, Address 0x000000DB, Mask 0xFFFFFFFF, Permissions rw, Module "/diagnostic/iocBoot/iocdiagnostic/ipbus_connection/ioc_regs.xml"
terminate called after throwing an instance of 'uhal::exception::NoBranchFoundWithGivenUID'
  what():  No branch found with ID-path "�"�w�" from node "TOP.sl1"
Not even a partial match found for ID-path "�"�w�" from node "TOP.sl1". If this address looks correct, please check for leading, trailing and stray whitespace.

Aborted (core dumped)
``` 
which sadly is not much more informative. 
Today we further investigated the problem and it seem to be related to the piece of code:
```
ConnectionManager manager(file1);
HwInterface hw = manager.getDevice("IOC");

std::vector<std::string> v = hw.getNode("sl1").getNodes();
for(auto s : v){
    printf("INS %s - %s \n", "sl1", s);
}
```
which prints:
```
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g�� 
INS sl1 - g��
```
And at each run the strings printed are different (but always garbage). 
The same code written in python, is not affected by the problem:
```
    manager = uhal.ConnectionManager("file://" + args.path)
    hw = manager.getDevice(args.id)

    s = hw.getNode("sl1").getNodes()
    print s
```
In fact the printings are correct:
```
['BPM_DELAY', 'BPM_ENABLE_REG', 'BPM_RESET_REG', 'BPM_CTRL_REG', 'FMC_DATA4', 'FMC_DATA2', 'FMC_DATA1', 'FMC_DATA3', 'FMC_CTRL_REG']
```
Are we doing something wrong?
Thank you in advance,
Davide

tswilliams, 2018-06-18:
Hi,

This is quite strange. Can you send us your address files? Then I'll attempt to reproduce the problem on my machine.

Cheers,
Tom

darcato, 2018-06-18:
Here they are!
[ioc_.zip](https://github.com/ipbus/ipbus-software/files/2111653/ioc_.zip)

tswilliams, 2018-06-18:
Thanks! Which versions of pugixml & boost are you using on your ubuntu 16.04 setup? Did you use any special command-line arguments when compiling uHAL, or did you simply type `make` with no further arguments?

darcato, 2018-06-18:
The problem first occurred on the personal pc of one of my collegues. Now I am testing inside a docker container, so I can show you the complete setup:
```
# estracted from a gitlab-ci file
  image: ubuntu:16.04

    - apt update 
    - apt -y install git build-essential libreadline-dev re2c make erlang g++ libboost-all-dev libpugixml-dev python-all-dev rsyslog libgsl-dev
    - touch /usr/lib/erlang/man/man1/x86_64-linux-gnu-gcov-tool.1.gz
    - touch /usr/lib/erlang/man/man1/gcov-tool.1.gz
    - git clone --depth=1 -b v2.6.1  https://github.com/ipbus/ipbus-software.git
    - cd ipbus-software
    - make -j5
    - make install
    - export LD_LIBRARY_PATH=/opt/cactus/lib:$LD_LIBRARY_PATH
```
libboost-all-dev version 1.58.0.1ubuntu1
libpugixml-dev version 1.7-2

tswilliams, 2018-06-18:
Hi,

I have reproduced the strange output you saw with the following code:
```
ConnectionManager manager(file1);
HwInterface hw = manager.getDevice("IOC");

std::vector<std::string> v = hw.getNode("sl1").getNodes();
for(auto s : v){
    printf("INS %s - %s \n", "sl1", s);
}
```

... but in that case, the strange output is due to a bug in the `printf` statement - the last argument to `printf` should be of type `char *` in order to match the format string `%s` (i.e. `s.data()` rather than `s`).

I have tried to reproduce the original error that you reported by accessing several nodes, however I have not managed to reproduce that problem yet. Can you send me the C++ source code that results in the error that you reported when opening the ticket?