ipbus/ipbus-software #139: Problem with starting ControlHub in systemd.

m-kru, 2019-03-20:
I have installed ipbus-software on Arch Linux following the [Compiling and installing from source](https://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/compile.html#compiling-and-installing-from-source) instructions. Unfortunately ControlHub was not successfully installed to be started automatically by systemd as a daemon.
I have copied `ipbus-software/controlhub/pkg/systemd/controlhub.service` file to the appropriate directory, commented out `EnvironmentFile=` and replaced `__CACTUSROOT__ ` with `/opt/cactus`. When I try to start controlhub via systemctl I get following output:

```
Mar 20 16:37:47 arch systemd[1]: Starting IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)...
Mar 20 16:37:47 arch controlhub[15247]: init terminating in do_boot ({cannot get bootfile,no_dot_erlang.boot})
Mar 20 16:37:47 arch controlhub[15247]: [1B blob data]
Mar 20 16:37:47 arch controlhub[15247]: Crash dump is being written to: erl_crash.dump...done
Mar 20 16:37:47 arch controlhub[15247]: Using config file: /etc/controlhub/default/sys.config
Mar 20 16:37:48 arch controlhub[15247]: ERROR in ControlHub configuration files! Details below.
Mar 20 16:37:48 arch controlhub[15247]: {"init terminating in do_boot",{'cannot get bootfile','no_dot_erlang.boot'}}
Mar 20 16:37:48 arch controlhub[15247]: init terminating in do_boot ({cannot get bootfile,no_dot_erlang.boot})
Mar 20 16:37:48 arch controlhub[15247]: [1B blob data]
Mar 20 16:37:48 arch controlhub[15247]: Crash dump is being written to: erl_crash.dump...done
Mar 20 16:37:48 arch systemd[1]: controlhub.service: Control process exited, code=exited status=4
Mar 20 16:37:48 arch systemd[1]: controlhub.service: Failed with result 'exit-code'.
Mar 20 16:37:48 arch systemd[1]: Failed to start IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications).
```
I lack ideas what might be wrong.
I have installed erlang system package (ver. 21.3-1) and file `no_dot_erlang.boot` is located in 2 places:
```
/usr/lib/erlang/bin/no_dot_erlang.boot
/usr/lib/erlang/releases/21/no_dot_erlang.boot
```

I have tried copying each of them to the `/opt/cactus/lib/controlhub/bin` directory where controlhub script is located and I get a slightly different error:
```
Mar 20 16:46:40 arch systemd[1]: Starting IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)...
Mar 20 16:46:40 arch controlhub[15849]: Using config file: /etc/controlhub/default/sys.config
Mar 20 16:46:40 arch controlhub[15849]: Starting ControlHub .. ERROR
Mar 20 16:46:40 arch controlhub[15849]: Exec: /opt/cactus/lib/controlhub/erts-10.2/bin/erlexec  -boot /opt/cactus/lib/controlhub/releases/2.6.0/-detached -mode embedded -config /etc/controlhub/default/sys.config -controlhub config_file "/etc/controlhub/default/sys.config" -args_file /opt/cactus/lib/controlhub/releases/2.6.0/vm.args -- ''
Mar 20 16:46:40 arch controlhub[15849]: Root: /opt/cactus/lib/controlhub
Mar 20 16:46:40 arch controlhub[15849]: {"init terminating in do_boot",{'cannot get bootfile','/opt/cactus/lib/controlhub/releases/2.6.0/-detached.boot'}}
Mar 20 16:46:40 arch controlhub[15849]: init terminating in do_boot ({cannot get bootfile,/opt/cactus/lib/controlhub/releases/2.6.0/-detached.boot})
Mar 20 16:46:40 arch controlhub[15849]: [1B blob data]
Mar 20 16:46:40 arch controlhub[15849]: Crash dump is being written to: erl_crash.dump...done
Mar 20 16:46:40 arch systemd[1]: controlhub.service: Control process exited, code=exited status=1
Mar 20 16:46:40 arch systemd[1]: controlhub.service: Failed with result 'exit-code'.
Mar 20 16:46:40 arch systemd[1]: Failed to start IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
```

I have also tried to execute the controlhub script from the directory where I have write and read access but it also ends up with `init terminating in do_boot ({cannot get bootfile,no_dot_erlang.boot})` error.

I would really appreciate if you could give me some advice what might be wrong.

tswilliams, 2019-03-23:
Which version of Arch Linux are you using?

m-kru, 2019-03-23:
@tswilliams 
Linux arch 4.19.12-arch1-1-ARCH #1 SMP PREEMPT Fri Dec 21 13:56:54 UTC 2018 x86_64 GNU/Linux

tswilliams, 2019-04-22:
Hi @m-kru 

Sorry for not getting back to you with a solution for this sooner.

I've reproduced this problem in an archlinux docker container, and I observed the same problem in a Fedora 30 docker container - the Erlang version provided by default on both of these systems is v21.

In version 21 of Erlang, the default name for the Erlang bootfile changed from `start_clean.boot` to `no_dot_erlang.boot`. I've now patched the `master` branch of this repo to support version 21 of Erlang (pull request #141), and I've checked that this solves the problem on my archlinux docker container.

Can you try the `master` branch of the repository - if you still interested in a fix to this issue - and let me know if the problem still occurs for you?

Cheers,
Tom

m-kru, 2019-04-28:
● controlhub.service - IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
   Loaded: loaded (/etc/systemd/system/controlhub.service; enabled; vendor preset: disabled)
   Active: **active (running)** since Sun 2019-04-28 10:35:30 CEST

tswilliams, 2019-04-29:
Hi @m-kru 
Thanks; based on your last message, I assume that the ControlHub is now running successfully for you, using the latest tip of the `master` branch; so, I'll close this ticket now.

If the ControlHub still isn't working due to the problem described above, then please feel free to re-open the ticket.

Cheers,
Tom

closed 2019-04-29 by tswilliams