ipbus/ipbus-software #140: Debugging tools for controlhub?

PaoloGB, 2019-03-21:
Is there any way to enable debugging messages for control hub/ipbus?
A colleague has found an issue where the communication seems to hang at the very beginning of operations and we cannot find any clue on what is going wrong.
I am wondering if there is any way to see the IPBus packets being sent/received.

tswilliams, 2019-03-23:
Could you clarify what you mean by "communication seems to hang"? Even if some reply packets are not being received by uHAL or the ControlHub, uHAL should report a timeout; what log messages do you see in uHAL?

Debugging messages can be enabled for both the ControlHub and for uHAL. The uHAL log level threshold is the easiest to change - if you're using uHAL in C++, then you can enable all log messages as follows:
```
uhal::setLogLevelTo(uhal::Debug());
```
... or enable just messages at `info` level and above as follows:
```
uhal::setLogLevelTo(uhal::Info());
```
... and in Python, instead you would write:
```
uhal.setLogLevelTo( uhal.LogLevel.DEBUG )
```
... or:
```
uhal.setLogLevelTo( uhal.LogLevel.INFO )
```
Even without changing the ControlHub log level, it should write log messages if it doesn't receive any response from a device - you can see those log messages in `/var/log/controlhub.log`. Do you see any error messages in that file, at the same time as the communication problems that you observe.

PaoloGB, 2019-03-25:
Hi,
thanks for the reply.
That is the weird thing: in my experience I always had some message, if only a timeout.
This colleague is working at CERN. In his setup, after her sends a configuration command the system seems to halt. No messages from ControlHub, not even a timeout.
I tried to log in remotely and I can confirm the behaviour. 

The log level is currently set to 2 (Error). I will play around to see if there is any warning or debug message that can provide some hint on what is going on.

Thanks.

tswilliams, 2019-03-25:
OK, thanks. To clarify, when you state that "the system seems to halt", is an exception is thrown by uHAL? If you can post what uHAL log messages you see (assuming you have the log threshold set to 'error' or below), I may be able to provide more hints about what could be going wrong.

PaoloGB, 2019-03-25:
No, at least at ERROR level, no messages/excepmtion are provided. This is what is making the debug really hard.
We will now attemp to increase the verbosity and I will report back if there is any update.

PaoloGB, 2019-03-27:
Hi again,
after setting the log level to 5, the last messages before the system goes silent are the following:
`26-03-19 14:42:29.056322 [7efda132e700] INFO - Converted IP address string 192.168.200.30:50001" to 192.168.200.30:50001 and converted this to IP 0xC0A8C81E, port 0xC351

26-03-19 14:42:29.057156 [7efda132e700] INFO - Attempting to create TCP connection to 'localhost' port 10203.`

After that nothing happens. Is there any chance you have seen this before and have any idea what is causing it?

tswilliams, 2019-03-27:
Hi! No, I have never observed this problem before. Which version of the IPbus software and operating system are you using? If OS is SLC6 or CentOS7, was the IPbus software installed from the standard YUM repositories (i.e. according to instructions at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/yum.html ) ?

Have you tried restarting the ControlHub - e.g. by running `/opt/cactus/bin/controlhub_stop && /opt/cactus/bin/controlhub_start` ? If not, can you try that and see if you still encounter the same problem again?

arummler, 2019-03-27:
Hi,
I am going to answer here directly. The OS is CC7. I tried first to install the binary package which worked fine but I could not link against our application as it requires a modern gcc (v8 loaded per setup script from CERN LCG) and there was an ABI conflict with the provided binary which uses v4.3 I remember correctly). I built iphub manually following the instructions and that worked without an obvious problem. I stopped and restarted controlhub several times but that did not make any difference. If I use the controlhub I can see with controlhub_stats that the number of total connections increases by one and there is no error but when I check the number of active connections is already 0. There was some standard firewall which is disabled completely.

tswilliams, 2019-03-31:
OK, thanks for the detailed info! Which version of boost are you using? Are the uHAL libraries linked against a copy of the boost libraries that have been built with the same version of gcc as is used for uHAL? (E.g. If you want to use an LCG-distributed copy of the boost libraries, then in the IPbus `make` commands, you should have set the `EXTERN_BOOST_INCLUDE_PREFIX` and `EXTERN_BOOST_LIB_PREFIX` variables accordingly.)

Can you use GDB get a stack trace of the program when it hangs? If your program is a C++ executable, then you can do this by typing `gdb --args` before the command you normally run - i.e. `./myExecutable someArgument` would become `gdb --args ./myExecutable someArgument`. Once inside the `gdb` console, type `r` to run the program, then press `Ctrl+C` when the program hangs, and then finally type `where` when the GDB console appears again in order to get the stack trace. If your program is a Python script, that you normally run by typing `python my_script.py`, then you can run that within GDB by instead typing `gdb --args $(which python) my_script.py`; once inside the GDB, you would then use the same steps for a Python script as for a C++ executable.

tswilliams, 2019-03-31:
(re-opening after accidental close)

tswilliams, 2019-04-16:
Hi @PaoloGB @arummler 

Are you still encountering the same problem, or did you resolve the issue since I posted my last comment?

Cheers,
Tom

tswilliams, 2019-05-10:
Suspected cause: user mistake in installation; closing ticket now.

Please re-open if you're still encountering this problem.

closed 2019-05-10 by tswilliams