ipbus/ipbus-software #145: uhal::exception::UdpTimeout

dinardo, 2019-06-23:
Daer IPbus experts.
I'm developping the Middleware code to read, configure, calibrate the RD53 (pixel chip) for the CMS expertiment.
I'm using an "FC7" intermediate board with FPGA.
When I run a particularly long calibration, very often I get the following IPbus error mesage:

```
terminate called after throwing an instance of 'uhal::exception::UdpTimeout'
  what():  Timeout (1000 milliseconds) occurred for UDP receive from target with URI: ipbusudp-2.0://192.168.1.80:50001
Aborted (core dumped)
```

- Do you know what this is all about?
Looking at ethtool it seems that the network card of my computer is running at 1 Gbit/s, as it should, therefore I do not understnad from where the problem might come from.
I'm using and SFP connector from ABCU-57xxxxZ Family: 5731/5740

Many thanks.
- Mauro.

tswilliams, 2019-06-25:
Hi,

uHAL currently only implements the reliability mechanism (i.e. the method by which it recovers dropped packets) when traffic is routed via the ControlHub. With a URI beginning `ipbusudp-2.0` UDP packets will be sent directly from the uHAL client to the hardware device, so I suspect that these packets are very rarely getting dropped by your network interface card, or some switch/router between your computer and the FC7.

In order to ensure that any dropped packets are recovered (or any out-of-order packets re-ordered), you should route your communication a ControlHub, by using a URI beginning `chtcp-2.0` - e.g. `chtcp-2.0://host-running-controlhub:10203?target=192.168.1.80:50001`

I hope that is clear; if not, or if you encounter any further problems after routing traffic via the ControlHub, please let us know.

Cheers,
Tom

dinardo, 2019-06-26:
Hi Tom,
sounds good. I'll try as you suggested.

Quick curiosty: do you have an idea what could be the problem?
I have an almost peer-to-peer network: my computer mount an extra 1Gbit ethernet card which is connected to a 1Gbit switch which is connected to the FC7.
Nothing else is connected to the switch.
I wonder how this could cause dropped packets.

Many thanks,
- Mauro.

dinardo, 2019-07-04:
Hi,
I've been running recently some quite communication intensive calibration, which require several IPbus transactions for several minutes (~20).
Which it shouldn't be a problem per se, i think.
Unfortunately they end up with a quite disruptive error:

```
terminate called after throwing an instance of 'uhal::exception::ControlHubTargetTimeout'
  what():  The ControlHub did not receive any response from the target with URI: chtcp-2.0://localhost:10203?target=192.168.1.80:50001ControlHub error code is: 4
Aborted (core dumped)
```
Do you know what could be the cause of the problem?
Do you have a suggestion on how to mitigate it?

Many thanks,
- Mauro.

dinardo, 2019-07-08:
Dear Experts,
I was wondering whether you had a chance to have a look at this issue.

Many thanks and kind regards.
- Mauro.

tswilliams, 2019-07-08:
Hi,

Sorry for not replying sooner. The `ControlHubTargetTimeout` exception is only thrown when the ControlHub reports that it has not received a response from the hardware after the ControlHub's timeout period (which is 20 milliseconds by default).

If your machine is just very heavily loaded and the reception of one packet is slightly delayed, you may be able to avoid this error by simply increasing the timeout used by the ControlHub. This can be done by creating a file at `/etc/controlhub/sys.config` with contents (for 500 millisecond timeout):
```
[
  %% Take application-level settings (e.g. for logging) from 'core.config' file
  "/opt/cactus/lib/controlhub/core.config",

  %% Override default values of ControlHub settings
  {controlhub,
    [{device_response_timeout, 500}]
  }
].
```
... and then restarting your ControlHub (`/opt/cactus/bin/controlhub_stop && /opt/cactus/bin/controlhub_start`). Then, before trying your calibrations again, you should check that the new configuration has been picked up by running `/opt/cactus/bin/controlhub_info` and checking that the `timout for replies from board` equals `500ms`.

Apart from the response packet being delayed due to your machine being very busy, the other possibility is that the reply packet corresponding to the timeout is always dropped by some network device when the the ControlHub requests that it be resent. We should be able to determine whether or not that's happening based on the ControlHub logs. Could you send me the ControlHub log file `/var/log/controlhub/controlhub.log` for the machine that's running the ControlHub , and let me know the time that you observed the `uhal::exception::ControlHubTargetTimeout` error (so that I know which log file entries to focus on)? 

Finally, can you let me know which version of the IPbus FW & SW you're using?

Cheers,
Tom

tswilliams, 2019-07-08:
P.S. Forgot to say: More details on the ControlHub configuration file can be found at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/ControlHub.html#configuration-settings

dinardo, 2019-07-08:
Dear Experts,
now I'm getting this error:
```
terminate called after throwing an instance of 'uhal::exception::TcpTimeout'
  what():  Timeout (1000 milliseconds) occurred for receive from ControlHub with URI: chtcp-2.0://localhost:10203?target=192.168.1.80:50001
```
Do you have any suggestion?
Do you think i simply need to increase the TcpTimeout?

Many thanks.
- Mauro.

tswilliams, 2019-07-08:
Hi,

Yes, my recommendation would be to see if this reoccurs with a timeout of 10 seconds. (With high-throughput traffic, depending on the specs of your machine, it could simply be that occasionally user-space software takes slightly longer than the default timeout period to receive a reply packet from the OS / drivers). You can set this timeout period by calling the `setTimeoutPeriod` method of the `uhal::HwInterface` or `uhal::ClientInterface` classes - i.e. `hw.setTimeoutPeriod(10000)`.

If the error still occurs, and send us the ControlHub log files (and let us know the time at which the exception is thrown, so that we can find the log messages from around the same time).

Cheers,
Tom

dinardo, 2019-08-16:
Dear Experts,
always in this same thread I still get the following error:
```
Error while writing the following parameters: user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe, user.cmd_regs.ctrl_reg, user.cmd_regs.data0_reg, user.ctrl_regs.fast_cmd_reg_1.cmd_strobe,
terminate called after throwing an instance of 'uhal::exception::ControlHubTargetTimeout'
  what():  The ControlHub did not receive any response from the target with URI: chtcp-2.0://localhost:10203?target=192.168.1.80:50001
ControlHub error code is: 4
Aborted (core dumped)
```
This error occures during a long calibration where several messages are sent from the computer to the FPGA.
If I run the following command I get:

[electroniclab@quark runDir]$ ps -aux | grep hub    
```
root      1822  0.0  0.0  11624   604 ?        S    Aug13   0:01 /opt/cactus_v2_6/lib/controlhub/erts-5.10.2/bin/epmd -daemon
root      1905  5.0  0.4 442724 80840 ?        Sl   Aug13 243:17 /opt/cactus_v2_6/lib/controlhub/erts-5.10.2/bin/beam.smp -zdbbl 2097151 -sbt db -scl false -sbwt very_long -swt low -- -root /opt/cactus_v2_6/lib/controlhub -progname controlhub -- -home $CACTUSROOT -- -boot /opt/cactus_v2_6/lib/controlhub/releases/2.5.2/controlhub -mode embedded -config /opt/cactus_v2_6/lib/controlhub/releases/2.5.2/sys.config -sname controlhub -setcookie ch_cookie -- console -noshell -noinput ''
electro+ 25986  0.0  0.0 114780   996 pts/5    S+   16:08   0:00 grep --color=auto hub
```
Though if I run the command:
[electroniclab@quark runDir]$ /opt/cactus/bin/controlhub_info
```
The ControlHub is down.
```
And my program, which obviously uses IPbus, works a part from the error message which I was mentioning.
Any further suggestion? Please let me know if you need further information.
Many thanks,
- Mauro.

tswilliams, 2019-08-16:
Hi,

Which version of the IPbus firmware & software are you using? Also which OS are you using? 

Based on the output of `ps aux | grep hub`, the ControlHub appears to be running using files under `/opt/cactus_v2_6`. But based on the path of the `controlhub_info` script that you ran, you also appear to have an installation of the ControlHub under `/opt/cactus` (the default installation location). Depending on the nature of these two installations, some conflict between them may explain the strange output of the `controlhub_info` command.

Cheers,
Tom

dinardo, 2019-08-16:
Dear @tswilliams,
this is the content of my /opt/ directory:
electroniclab@quark opt]$ ll
```
total 4
lrwxrwxrwx.  1 root root   12 Jul 20  2018 cactus -> cactus_v2_6/
drwxr-xr-x.  6 root root   54 Nov 20  2018 cactus_v2_6
drwxr-xr-x.  3 root root   20 May  9  2018 google
drwxr-xr-x.  7 root root   81 May 25  2018 puppetlabs
drwxr-xr-x.  3 root root   26 Jan  7  2019 rh
drwxr-xr-x. 13 root root 4096 Mar 23  2018 VirtualBox
```
Concerning the operating system: I'm using "CENTOS7"
Concerning instead the FW version, I need to ask. I'll get back to you as soon as I know.
- Mauro.

dinardo, 2019-08-17:
Dear @tswilliams,
we are using this formware version:
https://github.com/ipbus/ipbus-firmware/releases/tag/ipbus_2_0_v1

Do you think we should upgrade to a newer verion?
- Mauro.

tswilliams, 2019-08-17:
Hi,

Regarding the firmware, the `ipbus_2_0_v1` tag contained a bug in the IPv4 checksum calculation that only occurred for certain IP addresses and packet sizes, and could lead to the kind of symptoms that you describe above. This was fixed in version 1.1 of the FW, and the latest version of the FW is 1.4, so I would recommend rebuilding your firmware with either the `v1.1` tag or `v1.4` tag.

Regarding the version of the software that you have installed, I see that the ControlHub command in the output of `ps -aux | grep hub` contains `releases/2.5.2`, a directory that I would not expect in version 2.6.x of the software. To make sure that there's no ambiguity about which version of the software is being run, could you remove those directories, and reinstall the IPbus SW following the instructions at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/yum.html ?

Cheers,
Tom

tswilliams, 2020-01-07:
Hi,

As there hasn't been any response on this ticket for several months, I assume the problem was resolved, and so I'm closing it. If the problem reported here still persists, please re-open and/or comment!

Cheers,
Tom

closed 2020-01-07 by tswilliams