ipbus/ipbus-software #149: DummyHardware::SetEndpoint() became private ?

sly74fr, 2019-09-11:
Dear IPbus team,

First thank you very much for all your great work on IPbus !

I rely on UDPDummyHardware to create an hardware mockup of the board we develop.
I used SetEndpoint() for my mockup to reply specific values for identified register addresses.
But now I can't anymore as this method became private meanwhile.

Maybe there is another way around this ?
Otherwise how would you recommend doing this kind of custom dummy hardware ?
I would love being able to load some address:value tables to simulate my board easily :)
What do you think ?

Best regards,
Sly

PS: Please find my source code that use SetEndpoint() for further insights.
[dummy_ldpb_board.cpp.zip](https://github.com/ipbus/ipbus-software/files/3600075/dummy_ldpb_board.cpp.zip)

tswilliams, 2019-09-22:
Hi,

Beyond using the `SetEndpoint` method there isn't any other way to accomplish what you want to do with the current API of the dummy hardware classes. Given that, I'll make relax that method to being `protected` again, and create a new patch release with that change in the next few weeks.

In the longer run, it might be wise for us to restructure the dummy hardware classes to enhance the support for this kind of use case. However, given the demands of other projects on my time over the next month, I won't be able to do such restructuring before the next patch release.

Cheers,
Tom

sly74fr, 2019-09-25:
Hi
First thank you for the relaxing interim solution.
On the long term side of things, I remain at your disposal to further discuss it.
Cheers,

closed 2019-10-15 by tswilliams

tswilliams, 2019-10-22:
Hi @sly74fr 

This fix has now been included in the version 2.6.5 release.

Cheers,
Tom

sly74fr, 2019-11-14:
Think you very much Tom,
Will keep you updated once deployed at CERN !

commit e0169e46 (2019-10-15): uHAL : Make dummy hardware 'SetEndpoint' method non-private again; fixes #149
PR #156 (2019-10-15): Make dummy hardware 'SetEndpoint' method non-private again
Fixes #149
commit 860d7176 (2019-10-15): uHAL : Make dummy hardware 'SetEndpoint' method non-private again; fixes #149