ipbus/ipbus-software #150: test suite to exercize ipbus clients?

sbinet, 2019-09-30:
hi,

I am the current maintainer of [go-daq/ipbus](https://github.com/go-daq/ipbus), a Go-based implementation of IPbus.

I was wondering whether there was an official testsuite to make sure the protocol was correctly implemented and testing cross-language interoperability.
(so I could test things against the C++ implementation).

tswilliams, 2019-10-07:
Hi,

Were you thinking of any kind of test suite tool / functionality in particular? We have a set of executables that respond to the IPbus protocol. Is that the kind of thing you were looking for, or something else?

Cheers,
Tom

sbinet, 2019-10-07:
yes, a set of executables written in LangA that communicate with (and thus test the implementation of) an executable written in LangB.
that would fit the bill.

tswilliams, 2019-10-07:
OK. The uHAL 'dummy hardware' executables [1] respond to IPbus transactions, mimicking a simple endpoint just containin single-port RAM slaves (i.e. the values returned by a read at any address is the same as the last value written to that address - or zero if that address hasn't been written to before). In particular, you can emulate a UDP-based endpoint on `localhost`, port 50001 with the following command (for an RPM-based installation):
```
/opt/cactus/bin/uhal/tests/DummyHardwareUdp.exe -p 50001 -v 2
```

Let me know if you need any other info.

Cheers,
Tom

[1] http://ipbus.web.cern.ch/ipbus/doc/user/html/software/uhalQuickTutorial.html#working-with-dummy-hardware

tswilliams, 2019-12-05:
Closing ticket (since no response in a few weeks). Please re-open if you have any other questions on this topic

closed 2019-12-05 by tswilliams