ipbus/ipbus-software #151: open connections per ipbus target ?

juskoa, 2019-10-04:
Dear developers/users of ipbus-suite,
Is there a way how to find out about open connections
for an ipbus target, i.e. to get a following table:
```
IP1 pid1@host1 pid2@host2...
IP2 pid1@host1 pid2@host2...
...
```
or at least:
```
IP1 number_of_opened_connections
IP2 number_of_opened_connections
...
```
?
/opt/cactus/bin/controlhub_stats
returns only `active` connections, where `active` means: a packet exchange ongoing.
We re running uhal version 2.6.4.
Cheers, Anton

tswilliams, 2019-10-07:
Hi,

The first section of the output from the `controlhub_stats` command - titled `CLIENT` - lists the number of requests & responses for every open TCP connection to a uHAL client - e.g. when there is only one client connected:
```
CLIENT  All-time connections: 2
          Active connections: 1 (peak: 1)
  @ 127.0.0.1:42295
           Requests received: 51484 (of which 0 were malformed)
              Responses sent: 51430
  Old clients
           Requests received: 101 (of which 0 were malformed)
              Responses sent: 101
```

A uHAL `chtcp-2.0` client will connect to the ControlHub the first time that the client sends a request packet to the board, and the connection will be closed once all responses have been received, and no further transactions have been sent over the TCP connection for 60 seconds.

So, I believe that with current releases you should be able to either manually or programmatically determine the number of currently open TCP connections to each IP address  - i.e:
```
IP1 number_of_opened_connections
IP2 number_of_opened_connections
...
```
... by reading all of the entries from that part of the `controlhub_stats` output. However, we could possibly add that information explicitly as a summary at the start of the command's output, if that would be useful.

The ControlHub is not aware of the PID of the process at the other end of the TCP connection, so could not provide the PID for each connection, but does know the port number being used on the the remote end of the TCP connection (and that port number is already included as part of the `controlhub_stats` command's output).

Cheers,
Tom

juskoa, 2019-10-08:
Hi,
I see, the TCP connection to a client is closed if there is not active packet exchange ongoing with a timeout of 60s.
Even if I polled controlhub every minute (which I do not like), I would not be sure if client is still alive.
So the idea of having an info about 'who is using which board' using controlhub_stats is not the way to go.
I will use different mechanism (an explicit user request, or wrapper for manager.getDevice(boardname) ).
Thank you, I am closing this issue.
Cheers, Anton

closed 2019-10-08 by juskoa