ipbus/ipbus-software #158: RPMs for CentOS 8?

jhegeman, 2019-10-22:
Dear IPBus experts,

Are there any plans, or would it be possible, to create RPMs for CentOS 8? The current plan for CMS is not to wait for CERN CentOS 8 for deployment at P5, but to deploy 'raw' CentOS 8. Of course that would require the IPBus RPMs as well. Does that sound feasible?

Cheers,
Jeroen

tswilliams, 2019-10-22:
Hi,

My default plan was to wait until CERN CentOS 8 had been created, but that was in part because I assumed that would be the CMS plan for Point 5 as well :)

As I've already built the software and run the unit tests on higher versions of gcc and boost than the system-provided versions for CentOS8, I think that I should be able to create CentOS 8 RPMs quite quickly - I should only need to create the relevant docker containers for the automated build jobs. I'll give it a go later this week, and let you know how I get on.

Cheers,
Tom

tswilliams, 2019-10-29:
Hi @jhegeman 

Quick update on this: I've now managed to build the RPMs on CentOS8 - the software compiled without any changes, but I had to make a few updates on the packaging side.

I may want to refine these packaging changes over the next week or two, but we now have a preliminary version of CentOS8 RPMs. If you'd like to try out these bleeding-edge RPMs over the next couple of weeks, let me know, and I'll post the details.

Cheers,
Tom

tswilliams, 2020-01-20:
Hi again,

The changes required for CentOS8 were merged into the `master` branch before Christmas, and have now been included in a new release: version 2.7.0

The resulting CentOS8 RPMs are available through a new YUM repo, as explained at http://ipbus.web.cern.ch/ipbus/doc/user/html/software/install/yum.html . Please let me know if you have any problems in using this repo (or in using the software on CentOS8 in general).

Cheers,
Tom

closed 2020-01-20 by tswilliams

BothoP, 2020-01-28:
Hi @tswilliams,

the URL for the CentOS8 yum repo given on the above website gives a 404 error when I try to open it (http://ipbus.web.cern.ch/ipbus/doc/user/html/_downloads/ipbus-sw.centos8.x86_64.repo)

Cheers,
Botho

tswilliams, 2020-01-28:
Hi,

Sorry about that. I've now added the file. Let me know if you encounter any other problems.

Cheers,
Tom