ipbus/ipbus-software #159: Compile IPbus as static library

dinardo, 2019-10-24:
Dear experts,
would it be possible to compile IPbus as a static library?
I'm not an expert of make files, how shall I do that?

Many thanks,
- Mauro.

tswilliams, 2019-11-06:
Hi,

The makefiles did not include the commands required for static libraries, but I've now added those commands to the `task/static-linkage` branch.

If you're in urgent need to produce statically-linked applications that are based on uHAL then checking out that branch (commit 60c2c44bb35fbef7fa2dacd241152262b6ec2f69), and running `make` with `BUILD_STATIC=1` should produce statically-linked libraries successfully. I.e:
```
git clone --depth=1 -b task/static-linkage  https://github.com/ipbus/ipbus-software.git
cd ipbus-software/uhal
make BUILD_STATIC=1
```

I still want to make one or two changes to this branch (related to unit tests) before merging it, so it won't be merged until the end of the week, but based on my tests so far, the the main library should already work correctly.

Cheers,
Tom

dinardo, 2019-11-06:
Hi Tom,
ok git it.
Many thanks,
- Mauro.

tswilliams, 2019-11-07:
Brief update: There was actually a makefile bug in the previous commit (that only affected the `uhal/tests` package. I've just pushed a new commit to the `task/static-linkage` branch (commit 9c31975da4a328944dc60b5ae40cd7cd420e7dfd) that resolves that problem.

tswilliams, 2020-01-17:
Brief update: The changes from the branch mentioned above have now been merged into the `master` branch, and will be included in a new release - version 2.7.0 - will be created in the next week.

closed 2020-01-17 by tswilliams