ipbus/ipbus-software #163: uHAL Python 3 bindings

larks, 2019-11-07:
Any plans to add bindings for Python 3?
As Python 2 will be dead in under 2 months (https://pythonclock.org/), most have moved on to develop stuff for Python 3.

tswilliams, 2019-11-07:
Hi,

Yes, in fact the code base was updated to support Python 3 in the v2.6.3 release (February).

When the library is compiled, the Python bindings are automatically built against whichever installation of Python the `python` command corresponds to. The RPMs in the SLC6 and CentOS7 YUM repositories are built for Python 2 since that is the default Python version on those operating systems, but the software can always be rebuilt for other Python versions on those operating systems if needed.

Let me know if you have any other questions, or need any other info.

Cheers,
Tom

tswilliams, 2019-12-05:
Closing ticket (no response in a few weeks). Please re-open if you have any other questions, or need any other info, beyond that included in my comment above.

bg3mdo, 2020-08-06:
Is there a walkthrough for python3 custom building on CentOS 7?

tswilliams, 2020-08-11:
Hi @bg3mdo

I updated makefiles yesterday to make this process a bit easier; these changes are included in a new release, version 2.7.6.

I'm on holiday this week, so haven't written a walkthrough yet to match these changes, but the main points are as follows: By default, uHAL will build the Python bindings using whichever Python installation is accessed through the `python` command/alias. The uHAL Python bindings can be built against a different Python installation by setting the `PYTHON` makefile variable to point to the corresponding `python` command when running `make` - i.e. `make PYTHON=/path/to/pythonXY`

For example, if you've installed Python 3.6 (along with a corresponding build of the boost Python library) on CentOS7 using YUM - `sudo yum install python3-3.6.8 python3-devel-3.6.8 python3-libs-3.6.8 boost-python36-devel` - then the Python command is installed at `/usr/bin/python3.6` and you can build the uHAL Python bindings against that Python installation using the following command: `make PYTHON=/usr/bin/python3.6`

bg3mdo, 2020-08-11:
Hi @tswilliams 

Many thanks for the great work, I will try to build it and see if I can get it worked.

bg3mdo, 2020-08-11:
Hi @tswilliams 

I can successfully build uHAL from the source code and using "PYTHON" to point to a particular Python interpreter.  I installed these binaries using "make rpm" and "yum" for our CentOS7.  I found there were no "uhal" site-packages automatically added to the Python interpreter lib path that I want to use. I tried to copy an old one, in the Python, I can import the "uhal", but cannot use any "uhal" methods.

tswilliams, 2020-08-12:
Hi,

Did you run just `make rpm` or `make rpm PYTHON=/usr/bin/python3.6` to build the RPMs? The Python interpreter is also used for creating the RPMs, so you need to set the `PYTHON` variable with `make rpm` as well in order to make sure that the Python bindings library is installed to the correct path. (Sorry, forgot to mention that in my last post)

If you previously build the RPMs without setting the `PYTHON` variable, you should run `make cleanrpm` before running `make rpm PYTHON=/usr/bin/python3.6` in order to ensure a clean build of the RPMs.

Cheers,
Tom

bg3mdo, 2020-08-13:
Hi @tswilliam,

It works now!!, Thanks a lot. 

Cheers,
Yuan

bg3mdo, 2020-08-14:
Another silly question, can I have both 2.x and 3.x uHAL at the same time.

Cheers,
Yuan

tswilliams, 2020-08-17:
Yes, that is technically possible.

tswilliams, 2021-06-20:
Bindings for Python 3 have been included in the YUM repositories since the v2.8.0 release earlier this year.

closed 2021-06-20 by tswilliams