ipbus/ipbus-software #169: Feature Request: SmartDummyHardware ?

sly74fr, 2019-12-12:
Dear IPbus team,

Back to the topic following issue #149...

My use-case of such a 'SmartDummyHardware' class is:
1. Load a connection file;
1. For a given connection ID, load its address_table and hash-map it in RAM (NodeName -> NodeAddress -> NodeValue);
1. Provide read() and write() APIs to the mapped address_table from node names (same API as p_hw->getNode(node).read() and p_hw->getNode(node).write()) for the end-user to define its DummyHardware custom memory state);
1. Start some kind of DummyHardware presenting this mapped address_table and values on a given UDP or TCP port number;
1. The icing on the cake would be to propose an event-based mechanism for the end-user to provide some sort of callback for a given node name to compute its value dynamically.

Once again thank you very much for all your great work on IPbus.
I would also happily contribute to this enhancement...

Best regards,
Sly

dmnewbold, 2020-05-06:
Hi Sly: sorry it's taken to long to reply to this. Can you outline a use case for what you suggest? What would be the performance requirements? People are doing similar things to debug their firmware, but typically using a VHDL simulator (modelsim, etc) rather than software.

sly74fr, 2020-05-19:
No problem !
My use-case is

dmnewbold, 2020-05-19:
Hi Sly. Can you comment further on your original proposal? I can see how we could provide a registration / callback mechanism for user-written slave emulators to receive ipbus transactions. It's harder to see how that can be automatically done from the address table, as the behaviour of slaves is not really well defined there. Not everything is a register or a ram, right?

sly74fr, 2020-05-20:
Dear Dave,

On my side everything is either a register or a ram indeed. And I do not forsee an easy  way to define the stub behaviour from the address table, apart from hypothetic simple, static, default values. I think the dynamic behaviour should be of the entire responsability of the stub code. This would already represents a big step forward from my point of view.

Did I respond to your interrogations ?
Sly

dmnewbold, 2020-05-21:
Right, I think we agree. So the address table decoding could happen on the framework side, but everything else is up to the user. Note that in hardware world, slaves do not know their base address, and probably that should also be true in an emulation world.

Tom and I are thinking about how this could be implemented easily, and what language bindings might the most useful. Do you have any time to help in implementation?

sly74fr, 2020-05-25:
I would really enjoy getting my hands on but cannot make any promesses as my schedule is already awful !
What timeline is on your mind ?

dmnewbold, 2020-06-02:
Hm, when do you want it? 8-)

Prof David Newbold | Director Particle Physics Department
STFC Rutherford Appleton Laboratory, Harwell Campus,  Didcot OX11 0QX United Kingdom
Tel | Office: +44 (0) 1235 446760       https://ukri.zoom.us/my/dave.newbold
Email | dave.newbold@stfc.ac.uk<mailto:dave.newbold@stfc.ac.uk>       http://www.stfc.ukri.org<http://www.stfc.ukri.org/>

This email and any attachments are intended solely for the use of the named recipients. If you are not the intended recipient you must not use, disclose, copy or distribute this email or any of its attachments and should notify the sender immediately and delete this email from your system. UK Research and Innovation (UKRI) has taken every reasonable precaution to minimise risk of this email or any attachments containing viruses or malware but the recipient should carry out its own virus and malware checks before opening the attachments. UKRI does not accept any liability for any losses or damages which the recipient may sustain due to presence of any viruses. Opinions, conclusions or other information in this message and attachments that are not related directly to UKRI business are solely those of the author and do not represent the views of UKRI.

sly74fr, 2020-06-07:
No pressure here as I am not able myself to contribute effectivly...