ipbus/ipbus-software #175: Release 2.7.0 seems noticeably slower than 2.6.5

jhegeman, 2020-02-06:
Hi,

I just updated from release 2.6.5 to 2.7.0, and IPbus accesses have become noticeably slower. I went through a step-by-step downgrade, verify, upgrade-in-steps, and it looks as if the difference does not come from the controlhub but from something in the uhal libraries.

I'm running on CC7 (with the default GCC) and using IPbus over Ethernet via the controlhub.

Has anyone else noticed this? Any ideas on where to look for this?

Cheers,
Jeroen

jhegeman, 2020-02-06:
I just checked with 2.6.6, and I see no clear difference in speed between 2.6.5 and 2.6.6. So the difference came in with release 2.7.0.

tswilliams, 2020-02-06:
Hi,

Thanks for reporting this - and isolating the effect to a pair of versions of uHAL. I haven't heard this from anyone else yet, but that may just be because many other people have yet to migrate.

The performance degredation that you report: Is this based on the overall runtime of an applications that issues both reads and writes of a variety of sizes? Or does the application mostly issue single-word, or mostly issue large block reads/writes?

Also, when you say "noticeably slower", could you clarify the order of magnitude (e.g. 10% slower, 50% slower vs 90% slower)? I appreciate that you may not have an exact answer, but knowing a rough order of magnitude estimate (with whatever uncertainties you currently have) would help us confirm whether we see behaviour that's compatible with what your observations. 

Cheers,
Tom

jhegeman, 2020-02-07:
Hi Tom,

Thanks for thinking along.

> The performance degredation that you report: Is this based on the overall runtime of an applications that issues both reads and writes of a variety of sizes? Or does the application mostly issue single-word, or mostly issue large block reads/writes?

One should take this with a grain of salt, but I don't think I see a real slow-down in the monitoring (which is dominated by single-word reads). I do see a significant slow-down during my configuration steps (which are dominated by single-word writes).
 
> Also, when you say "noticeably slower", could you clarify the order of magnitude (e.g. 10% slower, 50% slower vs 90% slower)?

I noticed this while running some of the scripted tests in the TCDS lab. These tests use a Python-based type of run-control that configures applications, enables them, and controls them in order to (re-)create certain test conditions during something that looks like a CMS run. The controlling scripts issue periodic status updates about what applications are doing (e.g., Configuring). Based on this output I'd say that the board configuration takes about twice as long as it used to.

Some more anecdotes/observations:
- I believe I see the same level of slow-down (i.e., ~ 2 x) for applications that use wildly different address tables. For example the APVEController is affected as much as the CPMController, while the APVE has an almost trivial address table compared to the CPM. This would point to the hardware accesses, and absolve the address table parsing and handling. (The latter should only happen once anyway.)
- The lab control host is quite heavily loaded. I do see timeouts in the controlhub journal, but this does not seem to differ (visibly) between releases 2.6.x and 2.7.0.

Cheers,
Jeroen

jhegeman, 2020-02-07:
By the way: this may very well be an artefact/interplay with something in my setup or the TCDS software. Not necessarily an issue with 2.7.0 itself.

jhegeman, 2020-02-07:
Ok, I have narrowed down the origin of the slow-down. And I was wrong about the location, of course ;-)

In release 2.7.0 a lot more time is spent in `uhal::NodeTreeBuilder::build()`, and I believe all the extra time is spent in `uhal::NodeTreeBuilder::checkForAddressCollisions()`.

I made a quick-n-dirty piece of Python to investigate, that times how long it takes to load the CPM T1 address table (which is indeed huge):
```
#!/usr/bin/env python

import datetime

import uhal

if __name__ == '__main__':

    uhal.setLogLevelTo(uhal.LogLevel.WARNING)

    time_start = datetime.datetime.utcnow()
    manager = uhal.ConnectionManager("file:///opt/xdaq/share/tcdslab/etc/tcds_connections.xml")
    hw = manager.getDevice("cpm-t1")
    time_stop = datetime.datetime.utcnow()
    time_diff = time_stop - time_start
    print "Loading the LPM address tables took {0:s}".format(time_stop - time_start)

    print "Done"
```

The parsing and checking of the address table is clearly CPU-limited, as can be seen from the fact that I get quite different results when I load the host on which this runs with lots of XDAQ processes from when the machine is more or less idle.

- Release 2.6.6:
  - Idling: Loading the LPM address tables took 0:00:03.283630.
  - Loaded with XDAQs: Loading the LPM address tables took 0:00:04.794305.
- Release 2.7.0:
  - Idling: Loading the LPM address tables took 0:00:23.698415.
  - Loaded with XDAQs: Loading the LPM address tables took 0:01:07.427727.

So depending on how loaded the host is, the slow-down is about 7 to 17 times.

I have not yet had the courage to look into the changes in `checkForAddressCollisions()`.

tswilliams, 2020-02-07:
Thanks narrowing down the problem. I'll try to reproduce + fix ASAP. Can you point me where the LPM address tables are stored (so that I can verify that the problem's fully resolved when creating a fix)?

jhegeman, 2020-02-07:
You can use the attached RPM (once I manage to attach it), if you like. This will install the TCDS address tables in `/opt/xdaq/etc/tcds/addresstables/`.

tswilliams, 2020-02-10:
I think I've resolved the issue.

@jhegeman :  Can you try out this RPM, and let me know if it fixes the problem for you?
[cactuscore-uhal-uhal-2.7.0-1.autobuild_d39976f.centos7.gcc4_8_5.x86_64.rpm.zip](https://github.com/ipbus/ipbus-software/files/4182110/cactuscore-uhal-uhal-2.7.0-1.autobuild_d39976f.centos7.gcc4_8_5.x86_64.rpm.zip)

It may even slightly reduce the address table loading time (it does on my laptop).

Cheers,
Tom

jhegeman, 2020-02-12:
Thanks @tswilliams.

This indeed slightly reduces the loading/parsing time when the host is idling. And more importantly: it brings the 'loaded' results way down.

- Idling: Loading the LPM address tables took 0:00:02.941544.
- Loaded with XDAQs: Loading the LPM address tables took 0:00:04.935973.

All this just from using a dedicated iterator? I'm impressed!

Cheers,
Jeroen

tswilliams, 2020-02-12:
Thanks! I have a few more ideas on how to speed up the address-overlap-checking code, but it may take some time to implement them.

Since the slowdown has been resolved, I'll create a patch release later this evening based on the changes I've made so far.

Cheers,
Tom

commit 2e07954d (2020-02-12): uHAL: Fix accidental slowdown in function that checks for address collisions. Fixes #175
closed 2020-02-12 by tswilliams