ipbus/ipbus-software #176: Problems with python bindings

olagrottvik, 2020-02-09:
Hi,

I have some problems with the python bindings. For both python2 and python3 the following error occurs while loading uhal:

ImportError: /usr/lib/python2.7/site-packages/uhal/_core.so: undefined symbol: _ZN5boost6python6detail11init_moduleER11PyModuleDefPFvvE
N.B. ImportError (or derived exception) raised when uHAL's __init__.py tries to load python bindings library
     Maybe you need to add "/opt/cactus/lib", or some other path, to the "LD_LIBRARY_PATH" environment variable?

However, /opt/cactus/lib is correctly in LD_LIBRARY_PATH. Also the boost library used for building are added to the variable.

Also, building with python3 support only seems to happen if /usr/bin/python points to python3, not if the main python cmd points to python3. Having /usr/bin/python point to python3 will break yum on CentOS for instance. It would be nice if this could work by having /usr/bin/local/python as the main python command point to python3 and still have a functional yum.

tswilliams, 2020-02-09:
Hi,

Sorry to hear that you've encountered problems. A few questions to clarify the situation:
 * Which operating system are you using?
 * Which version of the IPbus software are you using?
 * How was the software installed, when you encountered the import error? Did you recompile the code from source, or install from one of the YUM repositories?

Cheers,
Tom

olagrottvik, 2020-02-10:
Hi Tom!

Thanks for the quick response. I'm building the code from source on CentOS 7.

I was building the software with external boost installed specified at make time with EXTERN_BOOST_INCLUDE_PREFIX and EXTERN_BOOST_LIB_PREFIX set.

However, I managed to fix the problem by rebuilding my boost library after I did a yum update. So I guess the problem was somewhere there, and not at your side.

So now everything works as expected :)

tswilliams, 2020-02-10:
I'm glad to hear that the Python bindings are working now. As a result I'll close the ticket, but please reopen if there are any other/remaining problems.

closed 2020-02-10 by tswilliams