ipbus/ipbus-software #177: Instructions on GUI

m-kru, 2020-02-12:
Are there any instructions on GUI? I mean how to run it and what is the purpose of GUI? I can't find anything on [IPbus 2.0 firmware: user guide](https://ipbus.web.cern.ch/ipbus/doc/user/html/firmware/index.html).

I have tried running GUI on Linux with following result:
```bash
[user@arch uhal]$ pwd
/tmp/ipbus-software/uhal
[user@arch uhal]$ python gui/pkg/uhal/gui/guiloader.py
Traceback (most recent call last):
  File "gui/pkg/uhal/gui/guiloader.py", line 3, in <module>
    from uhal.gui.utilities.utilities import dynamic_loader
ModuleNotFoundError: No module named 'uhal'
```
Of course python doesn't know where uhal is located so I have exported the path to `PYTHONPATH` and it gives following result:
```bash
[user@arch uhal]$ export PYTHONPATH=/tmp/ipbus-software/uhal
[user@arch uhal]$ python gui/pkg/uhal/gui/guiloader.py
Traceback (most recent call last):
  File "gui/pkg/uhal/gui/guiloader.py", line 3, in <module>
    from uhal.gui.utilities.utilities import dynamic_loader
ModuleNotFoundError: No module named 'uhal.gui'
```

tswilliams, 2020-02-18:
Hi,

There are not any instructions on how to use the GUI on the website at the moment. The GUI uses the uHAL Python bindings to display the values of the registers listed in an address table. It's not intended as a core tool that should be used in production environments scenarios; instead, it's an example GUI that illustrates how the uHAL Python API can be used to display simple tables of values read from hardware.

For the path-related problem, I guess you build uHAL from source, correct? If so, you need to source the `setup.sh` script under `uhal/tests` - i.e:
```
source uhal/tests/setup.sh
```
That should set up your environment variables correctly. 

Cheers,
Tom

tswilliams, 2020-05-06:
Closing. Please reopen if you have further questions on this.

closed 2020-05-06 by tswilliams