ipbus/ipbus-software #180: rpmbuild does not work for ARM architecture

DavidMonk00, 2020-02-19:
I am trying to build uhal within an ARM container and I am coming up against a number errors when creating the rpm.

The spec template throws errors based around the source.debug file and the centos8 if statement.
Furthermore the pycohal makefile seems to have references to x86 which are not architecture agnostic:
https://github.com/ipbus/ipbus-software/blob/1f8bf9b53579e195e4ab27f5d274e44301edbbf5/uhal/config/mfPythonRPMRules.mk#L42

Can you provide any experience with this?

tswilliams, 2020-02-19:
Hi,

What's the exact error message that you see when running `make rpm`?

Cheers,
Tom

tswilliams, 2021-01-22:
At the time, we confirmed that the problem was due to a non-CentOS OS being used, rather than being specific to a particular architecture; I have since confirmed that `make rpm` works on both arm/v7 and arm64 versions of CentOS8. So, I'll close this issue.

closed 2021-01-22 by tswilliams