ipbus/ipbus-software #181: Address table checking produces false positives

jhegeman, 2020-02-20:
This is not new, but it's now bugging me enough to open a ticket for discussion.

Let's consider the example address table file `example.xml`:
```
<?xml version="1.0" encoding="ISO-8859-1"?>

<node id="top">

  <node id="ctrl" address="0x00000000"  permission="rw" fwinfo="endpoint; width=2">
    <node id="clkgen1">
      <node id="reset" address="0x00000000" mask="0x00000001"/>
      <node id="in_sel" address="0x00000001" mask="0x00000003"/>
    </node>

    <node id="clkgen2">
      <node id="reset" address="0x00000000" mask="0x00000002"/>
      <node id="in_sel" address="0x00000001" mask="0x0000000c"/>
     </node>
  </node>

</node>
``` 

When this table is parsed by uhal, for example when creating an address decoder (`/opt/cactus/bin/uhal/tools/gen_ipbus_addr_decode example.xml`), the following overlaps are reported:
```
Branch 'top.ctrl.clkgen1' has address 0x00000000 and mask 0xffffffff which overlaps with branch 'top.ctrl.clkgen2' which has address 0x00000000 and mask 0xffffffff.
Branch 'top.ctrl.clkgen1' has address 0x00000000 and mask 0xffffffff which overlaps with branch 'top.ctrl.clkgen2.reset' which has address 0x00000000 and mask 0x00000002.
Branch 'top.ctrl.clkgen1.reset' has address 0x00000000 and mask 0x00000001 which overlaps with branch 'top.ctrl.clkgen2' which has address 0x00000000 and mask 0xffffffff.
```

In my opinion these are not really overlaps, but more a difference in understanding of the address table hierarchy. The uhal expectation seems to be that the resets are grouped together, and the input selections as well, with a structure like:
- reset
  - clkgen1
  - clkgen2
- in_sel
  - clkgen1
  - clkgen2
whereas the address table above tries to follow the physical hardware hierarchy.

It would be nice to have a way to suppress these 'philosophical warnings' and to only check for overlaps between leaf-level nodes.

alessandrothea, 2020-02-20:
Hi,
The address table above seems to try to use nodes to group bits from the same register in different subnodes. This is not how the address table is supposed to be used.

In fact it's not a matter of how uhal interprets the grouping. The address table above defines 2 registers, `ctrl.clkgen1` and `ctrl.clkgen2` at the same address (`0x0`), which is why uhal complains for an overlap.

jhegeman, 2020-02-20:
So this means that the way the hierarchy 'is supposed to be used' is _not_ to model the physical hardware structure?

In my point of view, the node `ctrl.clkgen1` is a logical grouping of control bits related to clock generator 1. I don't care where these bits physically live.

alessandrothea, 2020-02-20:
Well, the address table is supposed to model the firmware structure, which may or may not map the hardware one. So, in that sense yes, the address table cares about where the bits live.
And address table above represents 2 different registers (`clkgen1` and `clkgen2`) with overlapping addresses, so there is nothing very philosophical about it...

jhegeman, 2020-02-21:
So let me rephrase the question then, and make it philosophical: why is it considered bad if multiple address table nodes exist that resolve to the same word/bits?

The address table is simply a mapping from a human-readable string to an address plus a mask. There are good use cases for such overlaps. For example memory access as page-wise blocks and as random single-word accesses inside that same page. Similarly the address table could provide 'dual access' to device control lines. All resets for N devices could be physically wired to the same word so that they can all be strobed together by writing to that word, and the address table could provide an additional view by grouping the all functionality (including the reset) device by device, like the in my example table.

Why would one want to block such functionality?

dmnewbold, 2020-05-06:
Hi Jeroen. I can imagine many situations in which permitting multiple names for the same physical register would make generation of address decoders impossible. I mean, *you* know that these overlaps are all in the same physical firmware slave, but there is no way for uHAL to detect that. Most times things overlap, it's an error.

jhegeman, 2020-05-06:
Hi Dave,

Why would this make it impossible to generate address decoders? I can understand that the resulting decoders would be less efficient (since they can't take into account the overlaps) but not impossible to generate.

Jeroen

dmnewbold, 2020-05-06:
It depends what we are talking about. I can see (perhaps) a use case for aliases within one endpoint, but syntactically if one relaxes checks there is nothing to stop you specifying the same register in two completely different endpoints, which makes no sense.

In this case, we are either talking about the same resource at two different addresses (not *impossible* in firmware of course, but surely that is not what you intend) or overlapping address spaces for different decoders? In the latter case, how would the decoder at the level above know which path to use? How could we sanity check the address table?

I think if there is a real use case for aliases within one endpoint, then it would not be hard to implement a flag to override the normal address overlap check (e.g. an 'alias' flag). I think we would never want to do that by default, as this would prevent catching many common errors.

tswilliams, 2020-05-06:
Dave: I'm not sure why the discussion involves address decoders here. In the example at the top of this ticket, both the `clkgen1` and `clkgen2` nodes are descendants of the same 'address decoder endpoint' node (i.e. `ctrl`). So, whether or not child nodes of `clkgen1` and `clkgen2` have the same address doesn't affect the requirements on the auto-generated address decode logic.

dmnewbold, 2020-05-06:
That's right. My answer was to the implied question 'why is uHAL annoyingly stopping me doing this', and that involves some consideration of what would be syntactically possible if it didn't. As I said, I can't think of anything that will break if we DID allow (under some condition) aliasing of addresses within one endpoint, since I don't believe anything relies on a one-to-one mapping of registers to addresses... am I wrong?

jhegeman, 2020-05-07:
It looks as if the aim of the ticket is working out: discussion :-)

I think I agree with Dave that 'Most times things overlap, it's an error.'. If we document this as the reason behind this (somewhat-arbitrary-but-minor) limitation, I will stop complaining.

To be honest the functionality I described above, with a dedicated reset register grouping all resets together, can also be achieved by creating a true, dedicated reset register in the firmware. Design-wise I prefer complexity in software over duplication in firmware because in software the resources tend to be less constrained. On the other hand the software has more (and more diverse) users, so robustness there is a Good Thing too.

jhegeman, 2020-06-03:
Hi again,

It appears we have converged. Or at least run out of discussion. I interpret this as agreement with Dave's 'Most times things overlap, it's an error', and the choice was made to provide protection for users rather than increase the flexibility/potential for unnoticed problems.

Cheers,
Jeroen

closed 2020-06-03 by jhegeman