ipbus/ipbus-software #182: Systemd EnvironmentFile

m-kru, 2020-03-02:
What should be placed in the systemd EnvironmentFile (`EnvironmentFile=/etc/sysconfig/controlhub`) for controlhub? Today I have upgraded my controlhub installation and I have problems starting it with systemctl.

```
● controlhub.service - IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
     Loaded: loaded (/etc/systemd/system/controlhub.service; disabled; vendor preset: disabled)
     Active: failed (Result: exit-code) since Mon 2020-03-02 15:50:13 CET; 1s ago
       Docs: http://ipbus.web.cern.ch/ipbus/doc/index.html
    Process: 2285 ExecStart=/opt/cactus/bin/controlhub_start (code=exited, status=4)

Mar 02 15:50:13 arch systemd[1]: Starting IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)...
Mar 02 15:50:13 arch controlhub_start[2309]: erlexec: HOME must be set
Mar 02 15:50:13 arch controlhub_start[2286]: Using config file: /etc/controlhub/default/sys.config
Mar 02 15:50:13 arch controlhub_start[2286]: ERROR in ControlHub configuration files! Details below.
Mar 02 15:50:13 arch controlhub_start[2286]: erlexec: HOME must be set
Mar 02 15:50:13 arch systemd[1]: controlhub.service: Control process exited, code=exited, status=4/NOPERMISSION
Mar 02 15:50:13 arch systemd[1]: controlhub.service: Failed with result 'exit-code'.
Mar 02 15:50:13 arch systemd[1]: Failed to start IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
```

tswilliams, 2020-03-02:
Hi 

In that file, `HOME` needs to be set, and we typically set it to `/opt/cactus`. Which release of the software and which operating system are you using? If CentOS7/8, are you using the RPMs from the YUM repo?

Cheers,
Tom

m-kru, 2020-03-02:
ipbus-software v2.7.1, Arch Linux. Can you provide me with an example of /etc/sysconfig/controlhub? It looks like it is not in the repository, while `controlhub.service` for systemd is provided.

tswilliams, 2020-03-02:
Hi,

In a standard installation, the contents of `/etc/sysconfig/controlhub` would be:
```
CACTUSROOT=/opt/cactus
HOME=$CACTUSROOT
```
The template for this file is `controlhub/pkg/systemd/controlhub.env`, though the value of `CACTUSROOT` is changed during RPM creation using `sed`

Cheers,
Tom

m-kru, 2020-03-02:
Ok works, thanks.

closed 2020-03-02 by m-kru