ipbus/ipbus-software #187: PCIe protocol driver locks up after one failed access

jhegeman, 2020-03-17:
After one failed access, the PCIe protocol implementation fails on all further accesses because it mistakenly 'remembers' the original problem.

The issue is easily demonstrated using a pair of registers: one that functions normally, and one that does not exist and hence triggers a timeout. For example:

```
import uhal

if __name__ == "__main__":

    uhal.setLogLevelTo(uhal.LogLevel.WARNING)

    hw = uhal.getDevice("test_device",
                        "ipbuspcie-2.0:///dev/xdma0_h2c_0,/dev/xdma0_c2h_0",
                        "file:///home/FEROL/HAL/TCDS/pytcds2/tcds2_dth_address_tables/"
                        "dth_tcds_top.xml")

    node_name_good = "system_main.id.board_id"
    node_name_bad = "user_main.dummy"

    # Read a working register.
    tmp = hw.getNode(node_name_good).read()
    hw.dispatch()
    print "Before: reading node '{0:s}' works just fine: 0x{1:08x}".format(node_name_good, tmp.value())

    # Try to read a non-working register.
    try:
        tmp = hw.getNode(node_name_bad).read()
        hw.dispatch()
    except Exception as err:
        print "Now reading node '{0:s}' does not work: {1:s}".format(node_name_bad, str(err))

    # Try the working register again.
    try:
        tmp = hw.getNode(node_name_good).read()
        hw.dispatch()
        print "After: reading node '{0:s}' still works: {1:08x}".format(node_name_good, tmp.value())
    except Exception as err:
        print "After: reading node '{0:s}' no longer works either: {1:s}".format(node_name_good, str(err))
```

will produce output like:

```
$ python test_pcie_lockup_after_timeout.py
Before: reading node 'system_main.id.board_id' works just fine: 0x4454485f
17-03-20 16:49:30.764895 [7f3fa9f46740] WARNING - Expected reply packet to contain 3 words, but it actually contains 2 words
17-03-20 16:49:30.766165 [7f3fa9f46740] ERROR - Bad response code (0x = 'bus timeout on read') received for Incrementing read at base address 0x10000 (node "dth_tcds_top.user_main.dummy").  URI: "ipbuspcie-2.0:///dev/xdma0_h2c_0,/dev/xdma0_c2h_0". Sent/received headers: 0x2001010f / 0x20010006 (1/1 bytes into IPbus payload)
Now reading node 'user_main.dummy' does not work: Bad response code (0x = 'bus timeout on read') received for Incrementing read at base address 0x10000 (node "dth_tcds_top.user_main.dummy").  URI: "ipbuspcie-2.0:///dev/xdma0_h2c_0,/dev/xdma0_c2h_0". Sent/received headers: 0x2001010f / 0x20010006 (1/1 bytes into IPbus payload)
After: reading node 'system_main.id.board_id' no longer works either: Bad response code (0x = 'bus timeout on read') received for Incrementing read at base address 0x10000 (node "dth_tcds_top.user_main.dummy").  URI: "ipbuspcie-2.0:///dev/xdma0_h2c_0,/dev/xdma0_c2h_0". Sent/received headers: 0x2001010f / 0x20010006 (1/1 bytes into IPbus payload)
```

tswilliams, 2020-03-18:
This has been fixed in pull request #188 (@jhegeman : That's what I understand you see in your tests - correct me if I'm wrong). Closing.

closed 2020-03-18 by tswilliams

tswilliams, 2020-03-22:
@jhegeman : The fix for this issue has now been included in the v2.7.3 release (created yesterday)