ipbus/ipbus-software #193: missing checks in gen_ipbus_addr_decode script

mkrivda, 2020-04-30:
Hello.

We had 2 issues with inconsistency inside XML file, but gen_ipbus_addr_decode script has returned 0 (success) in both cases. We have found report only in \tmp folder.
I suggest to return 1 if XML file has such errors.

A) Two registers inside one slave had the same address, so there is missing address overlap check.
Here is the example:
<node id="ctpemu" address="0x120" description="ctp emulator" fwinfo="endpoint;width=5">
  <node id=  "SOC_tf_rate" address = "0xf" description = "Cal Trigger rate, TF downscaled[15:0]" />
  <node id=  "EOC_tf_rate" address = "0xf" description = "Cal Trigger rate, TF downscaled[15:0]" />

We have found report only in /tmp folder file OverlapReport-home-trigger-WORK-vhd-ltu_logic.xml:
Overlap report for "/home/trigger/WORK/vhd/ltu_logic.xml".
Branch 'ctpemu.SOC_tf_rate' has address 0x0000012f and mask 0xffffffff which overlaps with branch 'ctpemu.EOC_tf_rate' which has address 0x0000012f and mask 0xffffffff.

B) Endpoint width was different as block size, so there is missing check for block size in respect to endpoint width.
Here is the example:
1. unspotted error when the last node in an .xml file:
   <node id="ttcitcnts" address="0x2600" description="Counters for outgoing signals" fwinfo="endpoint;width=6">
     <node id="TTCitcnts" address="0x0" mode="block" size="0x440" 
description="48 TTCit counters" />
   </node>

i.e. for a 'width=6' max. size is 0x40, 'size=0x440' should be reported as an error.

2. The error is spotted only, when it causes an overlap, i.e. not last one in an .xml file:
   <node id="ttcitcnts" address="0x2600" description="Counters for outgoing signals" fwinfo="endpoint;width=6">
     <node id="TTCitcnts" address="0x0" mode="block" size="0x440" 
description="48 TTCit counters" />
   </node>
   <node id="ttcitcntsX" address="0x2700" description="Counters for outgoing signals" fwinfo="endpoint;width=6">
     <node id="TTCitcntsX" address="0x0" mode="block" size="0x40" 
description="48 TTCit counters" />
   </node>

Now we see at least warning in stdout of gen_ipbus_addr_decode (ret. code is still 0), and corresponding report in /tmp:
WARNING - Address overlaps observed - report file written at "/tmp/trigger/uhal/OverlapReport-home-trigger-WORK-vhd-ttcit_logic.xml

Overlap report for "/home/trigger/WORK/vhd/ttcit_logic.xml".
Branch 'ttcitcntsX.TTCitcntsX' has address range [0x00002700 - 0x0000273f] which overlaps with branch 'ttcitcnts.TTCitcnts' which has address range [0x00002600 - 0x00002a3f].

dmnewbold, 2020-05-06:
Hi. Both these issues have been addressed by some improvements in the script, which is now on a feature branch for testing. Could you maybe check out the updated script, try it on your design, and make sure it fixes the two issues you mention?

juskoa, 2020-05-07:
With a new version of the script gen_ipbus_addr_decode (from branch feature/193-addr-table-checks) I get following error:

> ERROR	: Endpoint node <<TOP>> has a mixture of endpoint and non-endpoint descendants
> ERROR	: Node errors detected, exiting early before writing output

The same .xml file is processed without error with a script from master branch.
There is an _endpoint_ keyword in each direct descendant of the TOP node. 
Do we have to declare _endpoint_ with all, i.e. also not direct descendants of the TOP node?
Here is an .xml:
[ttcit_logicxml.txt](https://github.com/ipbus/ipbus-software/files/4593393/ttcit_logicxml.txt)

dmnewbold, 2020-05-07:
Hi - if I run the script on that file, I don't see any error? The endpoint flag should be used to signify a block that has a concrete implementation in firmware as an ipbus slave or an address decoder. That means that you cannot mix 'physical' descendant nodes with 'logical' ones...

dmnewbold, 2020-05-07:
This is my output:

-bash-4.2$ ./gen_ipbus_addr_decode -v test.xml 
07-05-20 16:35:11.382983 [7f89849e5740] DEBUG - Parsed "file://test.xml" to:
07-05-20 16:35:11.383023 [7f89849e5740] DEBUG -     > [file] "test.xml"
07-05-20 16:35:11.383057 [7f89849e5740] DEBUG - lPath was "test.xml"
07-05-20 16:35:11.383066 [7f89849e5740] DEBUG - aParentPath is "/net/home/ppd/dmn/gen"
07-05-20 16:35:11.383078 [7f89849e5740] DEBUG - lPath now "/net/home/ppd/dmn/gen/test.xml"
07-05-20 16:35:11.383102 [7f89849e5740] DEBUG - Shell expansion of "test.xml" returned:
07-05-20 16:35:11.383115 [7f89849e5740] DEBUG -  > [file] /net/home/ppd/dmn/gen/test.xml
07-05-20 16:35:11.383219 [7f89849e5740] INFO - Reading XML address file "/net/home/ppd/dmn/gen/test.xml"
07-05-20 16:35:11.383333 [7f89849e5740] DEBUG - boardid built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383350 [7f89849e5740] DEBUG - fwinfo built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383376 [7f89849e5740] DEBUG - status built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383387 [7f89849e5740] DEBUG - fpga_temperature built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383398 [7f89849e5740] DEBUG - fpga_vccaux built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383415 [7f89849e5740] DEBUG - fpga_vccint built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383426 [7f89849e5740] DEBUG - fpga_vccbram built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383437 [7f89849e5740] DEBUG - fpga_alarms built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383449 [7f89849e5740] DEBUG - ctrl built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383459 [7f89849e5740] DEBUG - scopea built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383469 [7f89849e5740] DEBUG - scopeb built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383482 [7f89849e5740] DEBUG - fmcsfpsel built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383492 [7f89849e5740] DEBUG - ledenable built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383502 [7f89849e5740] DEBUG - sfpledenable built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383514 [7f89849e5740] DEBUG - adc_test_data built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383524 [7f89849e5740] DEBUG - c_spare2 built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383538 [7f89849e5740] DEBUG - ctrl built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383555 [7f89849e5740] DEBUG - iprog_icap built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383580 [7f89849e5740] DEBUG - icap_status built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383601 [7f89849e5740] DEBUG - icap built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383616 [7f89849e5740] DEBUG - ram built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383630 [7f89849e5740] DEBUG - reg built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383646 [7f89849e5740] DEBUG - addr built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383660 [7f89849e5740] DEBUG - data built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383670 [7f89849e5740] DEBUG - pram built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383688 [7f89849e5740] DEBUG - userregs built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383699 [7f89849e5740] DEBUG - i2cmain built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383715 [7f89849e5740] DEBUG - cfg built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383726 [7f89849e5740] DEBUG - errorcount built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
07-05-20 16:35:11.383737 [7f89849e5740] DEBUG - startcount built by uhal::Node* uhal::NodeTreeBuilder::plainNodeCreator(const bool&, const pugi::xml_node&)
[... cut ...]
INFO	: uHAL slave 13,FLASH_SPI_Ram_0,0b00000000000000000010001000000000,0b00000000000000000000000111111111
INFO	: uHAL slave 14,FLASH_SPI_Ram_1,0b00000000000000000010010000000000,0b00000000000000000000000111111111
INFO	: uHAL slave 15,ttcitcnts,0b00000000000000000010011000000000,0b00000000000000000000000011111111
INFO	: uHAL slave 16,ttcitcnts_channelA_BCRST,0b00000000000000000010011100000000,0b00000000000000000000000000001111
INFO	: Significant address bits 0b00000000000000000011011101111000
VHDL decode file saved:  ipbus_decode_test.vhd

juskoa, 2020-05-07:
I just took the script alone from github and replaced it in our ipbus installation, i.e. using our uhal (2.7) in /opt/cactus/bin/uhal/tools.

Should I instead checkout branch feature/193-addr-table-checks, compile and install it in a separate directory and arrange the usage of it when starting the script?

tswilliams, 2020-05-07:
Hi,

While I wouldn't normally recommend this, copying the script under the installed area should work in this case as all changes are contained within this script. However, if doing this you should update your installation to the latest release - version 2.7.3.

Can you retry after updating the installed software, and using the version of `gen_ipbus_addr_decode` from commit `f796b8e14e29f4ba5b1cbc9aee8933c65841440b`?

If you still see problems after doing that, can you tell us exactly what command you are running?

Thanks,
Tom

tswilliams, 2020-05-12:
Hi @juskoa 

Did you manage to retry the script as outlined in my last comment? If not, would you have time to try it out in the next couple of days? I'm asking as in principle this branch is ready to merge, and it's passed our tests so far. Since you initially reported the bugs, it would be good to verify that the script works well for you before we merge it.

If you do have time to retry this script in the next few days, please could you use commit `9c826b4c2f47f93d0beb531c10e4c8bb305fc379`?

Thanks,
Tom

juskoa, 2020-05-13:
Hi Tom,
I will install 2.7.3, copy the script from
branch feature/193-addr-table-checks
and check and report on gitlab  hopefully before
tomorrow evening,
Cheers, Anton

juskoa, 2020-05-13:
I installed uhal 2.7.3 and copied the script from feature/193-addr-table-checks.

When endpoint length < size, at the end of .xml file i.e.:
```
  ...
  <node id="leco1" address="0x3700" description="Link error counters for OLT_1 upstream" fwinfo="endpoint;width=6">
    <node id="data" address="0x0" mode="block" size="0x50" permission="r" description="64 Link Error counters" />
  </node>
</node>
```
the error is detected:
```
...
ERROR   : Endpoint node <<leco1>> (base address 0x3700, width 6) has a sub-node <<leco1.data>> (base address 0x3700, size 0x50) that overflows its address space
ERROR   : Node errors detected, exiting early before writing output
[trigger@acsl3 vhd]$ echo $?
2
```
For 'too many items' inside the node it is not detected:
```
...
  <node id="cdc_phase_scan" address="0x3740" fwinfo="endpoint;width=1">
    <node id="cdc_ctrl" address="0x0" /> 
    <node id="cdc_status" address="0x1" permission="r" /> 
    <node id="test" address="0x2" description="bad addr" /> 
  </node>
</node>
```
i.e. $? is 0 and /tmp/trigger/uhal/OverlapReport-... is not produced.

dmnewbold, 2020-05-13:
Interesting, I’ll look into it.

Prof David Newbold | Director Particle Physics Department
STFC Rutherford Appleton Laboratory, Harwell Campus,  Didcot OX11 0QX United Kingdom
Tel | Office: +44 (0) 1235 446760       https://ukri.zoom.us/my/dave.newbold
Email | dave.newbold@stfc.ac.uk<mailto:dave.newbold@stfc.ac.uk>       http://www.stfc.ukri.org<http://www.stfc.ukri.org/>

tswilliams, 2020-05-14:
@juskoa : I tried adding the register block you mentioned to a simple test address table, but when I did this `gen_ipbus_addr_decode` correctly detected the problem. To make sure that we're using the same files, can you send us the full address table(s) you used when `gen_ipbus_addr_decode` failed to report that there were 'too many items' inside the `cdc_phase_scan` node?

Also, when you run this test, what's the output of running `md5sum $(which gen_ipbus_addr_decode)`?

juskoa, 2020-05-14:
Hi Tom,
as said above, i copied it from feature/193-addr-table-checks, with '...new' name:
```
[root@acsl3 tools]# pwd
/opt/cactus/bin/uhal/tools
[root@acsl3 tools]# md5sum gen_ipbus_addr_decodenew 
5ac955444928391d10f3a3efaf9dd711  gen_ipbus_addr_decodenew
```
Here is the xml file:
```
<node id="TOP">
  <node id="ctrl" address="0x0" description="stat ctrl regs" fwinfo="endpoint;width=4" >
    <node id="boardid" address="0x0" permission="r" description="7-0: xx in MAC:08-00-30-00-2A-xx" />
    <node id="fwinfo" address="0x1" permission="r" description="31-24:type 23-8:version 7-0:subversion" />
    <node id="status" address="0x2" permission="r" description="1:SI5345_INTR 0:SI5345_LOL" />
    <node id="fpga_temperature" address="0x3" permission="r" description="bits 15..0" />
	<node id="fpga_vccaux" address="0x4" permission="r" description="bits 15..0" />
	<node id="fpga_vccint" address="0x5" permission="r" description="bits 15..0" />
	<node id="fpga_vccbram" address="0x6" permission="r" description="bits 15..0" />
	<node id="fpga_alarms" address="0x7" permission="r" description="bits 15..0" />
    <node id="ctrl" address="0x8" description="2:link error counters reset,1:soft ipbus reset,0:ipbus reset" />
    <node id="scopea" address="0x9" />
    <node id="scopeb" address="0xa" />
    <node id="fmcsfpsel" address="0xb" />
    <node id="ledenable" address="0xc" description="24..0"/>
    <node id="sfpledenable" address="0xd" description="23..0" />
    <node id="tdo_mux" address="0xe" description="onu_data[3],tdg_data[2],ctpemu_data_trdcpv[1],ctpemu_data[0]" /> <!-- ttcpon_data is default i.e."000"-->
    <node id="tdi_mux" address="0xf" description="ssm_data[3],onu_data_det3[2],onu_data_det2[1],onu_data_det1[0]" /> <!-- onu_data is default i.e."000"-->
  </node>
  <node id="icap" address="0x100" description="N_SLV_ICAP" fwinfo="endpoint;width=1">
    <node id="iprog_icap" address="0x0"  description="data" permission="w" />
    <node id="icap_status" address="0x1" description="avail[31],prdone[30],prerror[29]" permission="r" />
  </node>
  <node id="gbt1" address="0x108" description="GBT1 (SFP3)" fwinfo="endpoint;width=3">
    <node id="resets" address="0x0"  description="resets" />
    <node id="txctrl" address="0x1" description="tx cfg" />
    <node id="rxctrl" address="0x2" description="rx cfg" />
	<node id="c_spare" address="0x3" description="spare" />
    <node id="status" address="0x4" description="status" permission="r" />
	<node id="Word_count" address="0x5" description="Number of word Received" permission="r" />
    <node id="Error_count" address="0x6" description="Number of errors detected" permission="r" />
    <node id="Correct_count" address="0x7" description="Number of errors corrected" permission="r" />
  </node>
  <node id="gbt2" address="0x110" description="GBT2 (SFP4)" fwinfo="endpoint;width=3">
    <node id="resets" address="0x0"  description="resets" />
    <node id="txctrl" address="0x1" description="tx cfg" />
    <node id="rxctrl" address="0x2" description="rx cfg" />
	<node id="c_spare" address="0x3" description="spare" />
    <node id="status" address="0x4" description="status" permission="r" />
	<node id="Word_count" address="0x5" description="Number of word Received" permission="r" />
    <node id="Error_count" address="0x6" description="Number of errors detected" permission="r" />
    <node id="Correct_count" address="0x7" description="Number of errors corrected" permission="r" />
  </node>
  <node id="ctpemu" address="0x120" description="ctp emulator" fwinfo="endpoint;width=5">
    <node id=         "ctrl" address = "0x0" description =
      "PULSER_EDGE[13],PULSER_ena_for_TPC_RST[12], PULSER_ena_for_TPC_SYNC[11],
      PULSER_ena_for_CAL[10], PULSER_ena_for_PP[9], PULSER_ena_for_PH[8], 
      rst_out_cnts[4],
      LTU_mode[3],
      EMU_TorC[2],
      EMU_run[1],
	  EMU_reset+reset_scalers+reset_HB_counter[0]"	/>
	<node id=        "BCmask_ena" address = "0x1" description =
	  "BCMASK_ena_for_TOF[3], BCMASK_ena_for_CAL[2], BCMASK_ena_for_PP[1], BCMASK_ena_for_PH[0]" />
    <node id=        "TTena" address = "0x2" description =
      "TOF_ena[31],TPC_RST_ena[30],TPC_SYNC_ena[29],EOC_ena[10],SOC_ena[9],EOT_ena[8],SOT_ena[7],
      CAL_ena[6], PP_ena[5],PH_ena[4],HC_enable[3],HBr_enable[2]" />
    <node id=   "rst_N_cnts" address = "0x3" description =
      "rst_TOF_cnt[31],rst_TPC_RST_cnt[30],rst_TPC_SYNC_cnt[29],rst_EOC[10],rst_SOC[9],rst_EOT[8],rst_SOT[7],
      rst_CAL_cnt[6],rst_PP_cnt[5],rst_PH_bc_cnt[4],rst_PH_rnd_cnt[3],rst_HBr_cnt[2]" />
[... 170 lines cut ...]
    <node id="L1reject" address="0x3"  description="L1 reject rate" />
    <node id="CAL_rate" address="0x4"  description="TRD cal rate" />
    <node id="write_spare1" address="0x5"  description="write spare 1" />
    <node id="write_spare2" address="0x6"  description="write spare 2" />
    <node id="write_spare3" address="0x7"  description="write spare 3" />
    <node id="status" address="0x8" permission="r" description="status bits" />
    <node id="L0pre_count" address="0x9" permission="r" description="L0 pre-trigger counter" />
    <node id="L0_count" address="0xa" permission="r" description="L0 trigger counter" />
    <node id="L1_count" address="0xb" permission="r" description="L1 trigger counter" />
    <node id="Orbit_count" address="0xc" permission="r" description="Orbit trigger counter" />
    <node id="PP_count" address="0xd" permission="r" description="PP trigger counter" />
[... cut ...]
  <node id="leco1" address="0x3700" description="Link error counters for OLT_1 upstream" fwinfo="endpoint;width=6">
    <node id="data" address="0x0" mode="block" size="0x40" permission="r" description="64 Link Error counters" />
  </node>
  <node id="cdc_phase_scan" address="0x3740" fwinfo="endpoint;width=1">
    <node id="cdc_ctrl" address="0x0" /> 
    <node id="cdc_status" address="0x1" permission="r" /> 
    <node id="test" address="0x2" description="bad addr" /> 
  </node>
</node>
```
Cheers, Anton

tswilliams, 2020-05-14:
Hi,

Based on that MD5 hash, I think you have an old version of the script (commit `8a92791286e7fc97efe00e7d063a3b5332b374df`); since that version of the script, we've fixed some bugs in it. The latest version of the script on the `feature/193-addr-table-checks` branch reports the following error messages for the quoted address table:
```
ERROR	: Endpoint node <<cdc_phase_scan>> (base address 0x3740, width 1) has a sub-node <<cdc_phase_scan.test>> (base address 0x3742, size 0x1) that overflows its address space
ERROR	: Node errors detected, exiting early before writing output
```
... and its exit code is 2.

To try the latest version of the script, can you run the following commands?
```
curl https://raw.githubusercontent.com/ipbus/ipbus-software/9c826b4c2f47f93d0beb531c10e4c8bb305fc379/uhal/tools/scripts/gen_ipbus_addr_decode -o /tmp/gen_ipbus_addr_decode
chmod u+x /tmp/gen_ipbus_addr_decode
/tmp/gen_ipbus_addr_decode /path/to/addr_table.xml
```

Cheers,
Tom

juskoa, 2020-05-14:
Yes, works fine:
```$ /tmp/gen_ipbus_addr_decode ltu_logic.xml
ERROR	: Endpoint node <<cdc_phase_scan>> (base address 0x3740, width 1) has a sub-node <<cdc_phase_scan.test>> (base address 0x3742, size 0x1) that overflows its address space
ERROR	: Node errors detected, exiting early before writing output
[trigger@acsl3 vhd]$ echo $?
2
```
Thank you for patience, Cheers, Anton

closed 2020-05-14 by tswilliams

tswilliams, 2020-05-15:
Hi @mkrivda @juskoa 

The fixes for this ticket have now been included in a new software tag - v2.7.4

Cheers,
Tom

PR #195 (2020-05-14): Add missing checks to gen_ipbus_addr_decode
This branch fixes #193

Written by @dmnewbold