ipbus/ipbus-software #199: Make a transaction based on address, not the node.

m-kru, 2020-05-23:
In IPbus sfotware to make a transaction you need to get a node, and then call a method of this node such as `read` or `write`. However, sometimes it might be useful to make `write`/`read` on a specific address. Is it possible? Is there any function which I would be able to call like this: `write(address, value)`?

tswilliams, 2020-05-23:
Hi,

This is possible by accessing the 'client' class associated with the `Node` tree / `HwInterface` object, via the `getClient()` method, and then calling the `read`/`readBlock`/`write`/`writeBlock` methods of the returned `ClientInterface` reference.

For example, if `myHw` is a `HwInterface` instance/reference:
 * To write 0x10101010 at address 0x1000:
```
myHw.getClient().write(0x1000, 0x10101010);
```
 * To do a single-word read from the same address:
```
uhal::ValWord<uint32_t> value = mHw.getClient().read(0x1000);
```

Let me know if you have further questions.

tswilliams, 2020-05-23:
BTW, you can find the exact signature of all of the `ClientInterface` read/write methods in the doxygen pages at https://ipbus.web.cern.ch/ipbus/sw/release/2.7/api/html/db/d2b/classuhal_1_1_client_interface.html

m-kru, 2020-05-23:
These signatures are from C++, is it also available in Python? Do I also need to call `dispatch` in such case?

tswilliams, 2020-05-25:
Hi,

The same signatures are also available in Python.

You do need to call the `dispatch` method when issuing reads & writes through the `ClientInterface` methods.

Cheers,
Tom

m-kru, 2020-05-26:
Ok, I have tested it, it works.

closed 2020-05-26 by m-kru