ipbus/ipbus-software #208: Questions Regarding "connections.xml" and "address.xml" files

Desariirn, 2020-08-12:
Hello,

I had a couple questions about the "connections.xml" and "address.xml" files mentioned in the "uHAL Quick Tutorial" seen here (http://ipbus.web.cern.ch/ipbus/doc/user/html/software/uhalQuickTutorial.html). I know these two files are supposed to be created, though I'm not sure what directory they're supposed to be in. I've noticed in the default folder that IPbus keeps most of its files under, it places the "dummy_connections.xml" and "dummy_address.xml" files in "/opt/cactus/etc/uhal/tests". Are both "connections.xml" and "address.xml" supposed to be located here too?

Also, in the pre-requisites section for the link featured above, it mentions describing where the IPbus endpoints are located in the connections file and what the address layout is in the address file. I'm confused about where to find this information from. If it helps, I'm currently trying to setup uHAL so I can communicate with an FC7.

-Jonny

Desariirn, 2020-08-17:
Also: If it helps, I figured I could also supply the specific error I was getting originally and why I thought uHAL and IPbus were the issue. At the bottom of this link that I found (https://nam02.safelinks.protection.outlook.com/?url=https%3A%2F%2Flibraries.io%2Fgithub%2Fj1990grossmann%2Fph2_ACF&amp;data=02%7C01%7C%7Cf94e73d68cd74ab6032d08d83a1bbdf5%7Cb92d2b234d35447093ff69aca6632ffe%7C1%7C0%7C637323238106510482&amp;sdata=V72JzFQp7tGtPbmoM4SIKqAOYjVi%2FxAD%2FfNNei0maVo%3D&amp;reserved=0) there is a very similar issue that seems to suggest that a change to the "connections.xml" file could solve the issue I'm seeing below. However, it doesn't seem like I have a "connections.xml" file to edit.

[root@hexpc8 myCBC3Directory]# CMSITminiDAQ -f CMSIT.xml -r
||I| Resetting the backend board... it may take a while
terminate called after throwing an instance of 'uhal::exception::TransactionLevelError'
  what():  Bad response code (0x6 = 'bus timeout on read') received for Read-Modify-Write bits at base address 0x40000040 (11 nodes under "RD53.user.ctrl_regs" match).  URI: "chtcp-2.0://localhost:10203?target=192.168.1.80:50001". Sent/received headers: 0x2006014f / 0x20060046 (1/1 bytes into IPbus payload)
Aborted (core dumped)

tswilliams, 2020-08-17:
Hi,

> I had a couple questions about the "connections.xml" and "address.xml" files mentioned in the "uHAL Quick Tutorial" seen here [...]

Connections and address table files can be located in any directory - whichever directory is most suitable for your applications.

> Also, in the pre-requisites section for the link featured above, it mentions describing where the IPbus endpoints are located in the connections file and what the address layout is in the address [...]

The connections files are used to describe each hardware device (referred to as "IPbus endpoints" in the quick tutorial page): each line of a connections file specifies the protocol to use to communicate with a device (e.g. UDP, PCIe, ...) and what address table file should be used with the device. The address table file is a configuration file that allows users to specify a user-friendly string-based label for each register, RAM, FIFO etc that's connected to the the control bus in their firmware design, so that when writing software users can perform reads and writes via to these string labels, rather than having to hardcode the control bus addresses in software source code. Does that help, or were you looking for other information?

Cheers,
Tom

P.S. Apologies for the delay in replying - I was off work last week.

tswilliams, 2020-08-18:
The error message that you quoted ...
> [root@hexpc8 myCBC3Directory]# CMSITminiDAQ -f CMSIT.xml -r
||I| Resetting the backend board... it may take a while
terminate called after throwing an instance of 'uhal::exception::TransactionLevelError'
what(): Bad response code (0x6 = 'bus timeout on read') received for Read-Modify-Write bits at base address 0x40000040 (11 nodes under "RD53.user.ctrl_regs" match). URI: "chtcp-2.0://localhost:10203?target=192.168.1.80:50001". Sent/received headers: 0x2006014f / 0x20060046 (1/1 bytes into IPbus payload)

... indicates that there was no response on the control bus when the software tried to read/write to a subnode of `RD53.user.ctrl_regs`. I'm not an expert on CBC related software, but I typically encounter this kind of error when I'm accidentally using an incorrect address table that does not match the firmware that I'm trying to control. So in case you haven't done this already, my first suggestion would be to get in touch with the people responsible for the software and firmware that you're using, and check that you are using the correct address table & bitfile combination.

Cheers,
Tom

tswilliams, 2021-01-22:
Closing since no response received in a while; please re-open if you have further questions.

closed 2021-01-22 by tswilliams