ipbus/ipbus-software #217: AttributeError: module 'uhal' has no attribute 'ConnectionManager'

carmijo-az, 2020-09-29:
I recently changed the bindings to use python3 as per 
uHAL Python 3 bindings
#163 opened on Nov 7, 2019 by larks

This has been working fine.  There was a yum update last night.  Now I get the following error.  (See below)  Clearly, uhal does have a ConnectionManager.  'import uhal' did not raise any warnings or errors.  My OS is Centos 7.8  Any ideas on what happened?

  File "/home/larctest/git/git-090820/LATOME/LATOME/projects/firmware_control/firmware_control/ipbus.py", line 108, in OpenConnection
    manager = uhal.ConnectionManager(xml_file_url) 
AttributeError: module 'uhal' has no attribute 'ConnectionManager'

yum uhal update:
[root@eepp-xilinx2 pkg_resources]# rpm -qa --last
cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64 Tue 29 Sep 2020 10:35:29 AM MST
cactuscore-uhal-grammars-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:29 AM MST
cactuscore-uhal-uhal-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:28 AM MST
cactuscore-uhal-tests-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:28 AM MST
cactuscore-uhal-log-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:28 AM MST
cactuscore-uhal-uhal-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:27 AM MST
cactuscore-uhal-tools-2.7.9-1.centos7.gcc4_8_5.noarch Tue 29 Sep 2020 10:35:27 AM MST
cactuscore-uhal-tests-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:27 AM MST
cactuscore-uhal-pycohal-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64 Tue 29 Sep 2020 10:35:27 AM MST
cactuscore-uhal-grammars-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:27 AM MST
cactuscore-uhal-log-2.7.9-1.centos7.gcc4_8_5.x86_64 Tue 29 Sep 2020 10:35:26 AM MST

tswilliams, 2020-09-29:
Hi,

There haven't been any changes to the `ConnectionManager` class or the source code that implements the Python bindings in the last few releases, so I'm not sure on exactly what happened.

A few immediate thoughts/questions: I assume that you installed the locally-built Python 3 buildings as an RPM, right? First thing I should check: Is that RPM still installed? (The RPM list above only includes one `pycohal` RPM, which was built for Python 2.7.)

Cheers,
Tom

carmijo-az, 2020-09-29:
Hi,
Thanks.  
Yes, one of the first things I did was check to make sure that uhal was still in python3.6/site-packages.  

I tried to reinstall the rpm.   I get the following:

[root@eepp-xilinx2 ipbus-software-py3]# rpm -i uhal/gui/rpm/cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64.rpm
	package cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64 (which is newer than cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64) is already installed
	file /opt/cactus/bin/uhal/uhal_gui.exe from install of cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64 conflicts with file from package cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64

-Charlie

carmijo-az, 2020-09-30:
/opt/cactus/bin/uhal/uhal_gui.exe from install of cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64 conflicts with file from package cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64

So, is there a python3.6 rpm for cactuscore-uhal-gui-2.7.8-1.centos7?

carmijo-az, 2020-09-30:
sorry, that should have been, is there a python3.6 rpm for cactuscore-uhal-gui-2.7.9-1.centos7.

tswilliams, 2020-09-30:
Hi,

Right now there are not any Python 3 uHAL RPMs in the YUM repository. If you wanted to use that package from Python 3, you would need to rebuild that RPM locally with Python 3. To clarify exactly what RPMs are installed, can you send me the output of `rpm -qa | grep -i cactuscore-uhal`?

Cheers,
Tom

carmijo-az, 2020-10-01:
Hi,

[root@eepp-xilinx2 ~]# rpm -qa | grep -i cactuscore-uhal
cactuscore-uhal-grammars-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-pycohal-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64
cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64
cactuscore-uhal-tests-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-tools-2.7.9-1.centos7.gcc4_8_5.noarch
cactuscore-uhal-uhal-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-grammars-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-log-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-tests-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-uhal-debuginfo-2.7.9-1.centos7.gcc4_8_5.x86_64
cactuscore-uhal-log-2.7.9-1.centos7.gcc4_8_5.x86_64

Best,
Charlie

tswilliams, 2020-10-01:
Hi,

OK, thanks. I can only see a Python 2.7 `uhal-pycohal` RPM there, so I think your local Python 3.6 build of that RPM must have been removed during the YUM update. In that case, you should be able to solve the problem by just reinstalling your local Python 3 build of the `uhal-pycohal` RPM.

Cheers,
Tom

carmijo-az, 2020-10-02:
Hi,

When I try to reinstall the python3.6 build, I get this message.

[root@eepp-xilinx2 ipbus-software-py3]# rpm -i uhal/gui/rpm/cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64.rpm

	package cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64 (which is newer than cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64) is already installed
	file /opt/cactus/bin/uhal/uhal_gui.exe from install of cactuscore-uhal-gui-2.7.8-1.centos7.python3.6.x86_64 conflicts with file from package cactuscore-uhal-gui-2.7.9-1.centos7.gcc4_8_5.python2.7.x86_64

-Charlie

tswilliams, 2020-10-02:
Hi,

OK. It's not currently possible to install multiple builds of the `cactuscore-uhal-gui` RPM that correspond to different Python versions. If you're not using the GUI, I suggest that you just install the Python 3.6 build of the `cactuscore-uhal-pycohal` RPM. Otherwise, if you are using the GUI in Python 3, I suggest that you uninstall the Python 2.7 build of the GUI package (`yum remove cactuscore-uhal-gui`) and then try to install the Python 3 version of the packages after doing that.

Let me know if that doesn't work

Cheers,
Tom

closed 2021-01-30 by tswilliams