ipbus/ipbus-software #221: How to enable Jumbo Frame

habrade, 2020-11-27:


tswilliams, 2020-11-30:
Hi,

The 1400-byte limit is currently hardcoded in the source code, and cannot be changed from the client URI. We could add the ability to override this default from the client URI in future software releases. But if you want to increase this now you can do so by compiling the code from source after increasing the return values of the `getMaxSendSize()` and `getMaxReplySize()` methods of the `UDP` and `TCP` classes from 1400 to 8900 (for example) on lines 135 and 142 of `uhal/uhal/src/common/ProtocolUDP.cpp` as well as on lines 136 and 143 of `uhal/uhal/src/common/ProtocolTCP.cpp`.

That said, in previous performance tests (a few years ago) we never observed any packet loss at 0.5 Gbps using standard Ethernet frames, and so if the problem is that some packets are dropped I'm not sure whether switching to Jumbo frames will help.

Cheers,
Tom

habrade, 2020-12-02:
Hi Tom,

Thank you very much for the detailed instruction.
I changed the lines you mentioned and did some tests.
Unfortunately, I got some strange results...

For the parameter of the function `readBlock(num)`
- **num <= 375** 
Everything OK.
- **num = 389**
> 02-12-20 16:21:21.099188 [7f17aca75740] DEBUG - Checking buffer space
> 02-12-20 16:21:21.099197 [7f17aca75740] DEBUG - Sending 20 bytes
> 02-12-20 16:21:21.099214 [7f1799739700] DEBUG - Expecting 1568 bytes in reply.
> 02-12-20 16:21:21.099294 [7f1799739700] DEBUG - Validation Complete!
> 2020-12-02 16:21:21 sdong-ubuntu __main__[249525] INFO Transfer speed: 145.985059 Mbps
> 2020-12-02 16:21:21 sdong-ubuntu __main__[249525] INFO Write data to .txt ...
> 2020-12-02 16:21:21 sdong-ubuntu __main__[249525] INFO Write file speed: 52.980528 Mbps
> 2020-12-02 16:21:21 sdong-ubuntu __main__[249525] INFO Write to .txt end.
> free(): invalid next size (normal)
>  *** Break *** abort

no log printed like when num = 390

-  **num=390**:
> 02-12-20 16:12:25.345999 [7f79e3f9d740] DEBUG - Checking buffer space
> 02-12-20 16:12:25.346011 [7f79e3f9d740] DEBUG - Sending 20 bytes
> 02-12-20 16:12:25.346029 [7f79d0c61700] DEBUG - Expecting 1516 bytes in reply.
> 02-12-20 16:12:25.346100 [7f79d0c61700] DEBUG - Validation Complete!
> 2020-12-02 16:12:25 sdong-ubuntu __main__[246821] INFO Transfer speed: 115.376132 Mbps
> 2020-12-02 16:12:25 sdong-ubuntu __main__[246821] INFO Write data to .txt ...
> 2020-12-02 16:12:25 sdong-ubuntu __main__[246821] INFO Write file speed: 52.611320 Mbps
> 2020-12-02 16:12:25 sdong-ubuntu __main__[246821] INFO Write to .txt end.
>  *** Break *** segmentation violation

[crash.log](https://github.com/ipbus/ipbus-software/files/5627844/crash.log)

- **num=400**
> 02-12-20 16:19:08.460391 [7fb39b6d0740] DEBUG - Read unsigned block of size 400 from address 0x60000024
> 02-12-20 16:19:08.460403 [7fb39b6d0740] DEBUG - Checking buffer space
> 02-12-20 16:19:08.460409 [7fb39b6d0740] DEBUG - Checking buffer space
> 02-12-20 16:19:08.460423 [7fb39b6d0740] DEBUG - Sending 20 bytes
> 02-12-20 16:19:08.460446 [7fb388394700] DEBUG - Expecting 1612 bytes in reply.
>  *** Break *** [1]    248758 segmentation fault (core dumped)  ./run.py

no log printed like when num = 390

- **num=131072**
Timeout....
```
02-12-20 16:36:51.923768 [7fab6fa50700] DEBUG - Expecting 28 bytes in reply.
02-12-20 16:36:51.923837 [7fab6fa50700] DEBUG - Validation Complete!
02-12-20 16:36:51.923876 [7fab82d8c740] DEBUG - Read unsigned block of size 131072 from address 0x60000024
02-12-20 16:36:51.924346 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924358 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924365 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924371 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924377 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924383 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924388 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924393 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924398 [7fab82d8c740] DEBUG - Checking buffer space
02-12-20 16:36:51.924404 [7fab82d8c740] DEBUG - Triggering automated dispatch
```
.....
> 02-12-20 16:36:51.937420 [7fab82d8c740] DEBUG - Checking buffer space
> 02-12-20 16:36:51.937425 [7fab82d8c740] DEBUG - Triggering automated dispatch
> 02-12-20 16:36:51.937432 [7fab82d8c740] DEBUG - Checking buffer space
> 02-12-20 16:36:51.937438 [7fab82d8c740] DEBUG - Checking buffer space
> 02-12-20 16:36:52.925572 [7fab6fa50700] WARNING - Closing UDP socket for URI "ipbusudp-2.0://192.168.3.17:50001" since deadline has passed
> 02-12-20 16:36:52.925756 [7fab6fa50700] ERROR - Timeout (1000 milliseconds) occurred for UDP receive from target with URI: ipbusudp-2.0://192.168.3.17:50001
> 02-12-20 16:36:52.925988 [7fab82d8c740] WARNING - Closing Socket since exception detected.
> Traceback (most recent call last):
>   File "./run.py", line 111, in <module>
>     mem = jadepix_dev.read_ipb_data_fifo(slice_size, safe_style=False)
>   File "/home/sdong/work/JadePix3/software/lib/jadepix_device.py", line 301, in read_ipb_data_fifo
>     return self._ipbus_link.read_ipb_data_fifo(self.reg_name_base, "DATA_FIFO", num, safe_style)
>   File "/home/sdong/work/JadePix3/software/lib/ipbus_link.py", line 85, in read_ipb_data_fifo
>     self._hw.dispatch()
> uhal._core.UdpTimeout: Timeout (1000 milliseconds) occurred for UDP receive from target with URI: ipbusudp-2.0://192.168.3.17:50001

I will do more tests...

cheers,
Sheng

tswilliams, 2020-12-03:
Hi,

Looking at the code again, I think another change might be needed - in addition to those I outlined above. Can you change line 66 of `uhal/uhal/src/common/ProtocolUDP.cpp` from:
```
  mReplyMemory ( 1500 , 0x00000000 ),
``` 
... to:
```
  mReplyMemory ( 9000 , 0x00000000 ),
```

Cheers,
Tom

habrade, 2020-12-03:
Hi Tom,

I just tested the new modification.
The good news is I don't get various strange results this time.
But I found a new limitation of the parameter for `readBlock(num)`:

Here are the logs:

When num <= 2028, everything OK.
> 3-12-20 21:30:04.007504 [7f56b00fe740] DEBUG - Read unsigned block of size 2028 from address 0x60000024
> 03-12-20 21:30:04.007520 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007526 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007531 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007536 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007541 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007546 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007551 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007555 [7f56b00fe740] DEBUG - Checking buffer space
> 03-12-20 21:30:04.007567 [7f56b00fe740] DEBUG - Sending 68 bytes
> 03-12-20 21:30:04.007591 [7f569cdc2700] DEBUG - Expecting 8148 bytes in reply.
> 03-12-20 21:30:04.007797 [7f569cdc2700] DEBUG - Validation Complete!
> 2020-12-03 21:30:04 sdong-ubuntu __main__[368085] INFO Transfer speed: 43.910717 Mbps
> 2020-12-03 21:30:04 sdong-ubuntu __main__[368085] INFO Write data to .txt ...
> 2020-12-03 21:30:04 sdong-ubuntu __main__[368085] INFO Write file speed: 183.325648 Mbps
> 2020-12-03 21:30:04 sdong-ubuntu __main__[368085] INFO Write to .txt end.
 
When num > 2028, reply timeout.
> 03-12-20 21:30:33.795831 [7f43b2289740] DEBUG - Read unsigned block of size 2029 from address 0x60000024
> 03-12-20 21:30:33.795846 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795852 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795855 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795859 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795863 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795866 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795870 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795873 [7f43b2289740] DEBUG - Checking buffer space
> 03-12-20 21:30:33.795883 [7f43b2289740] DEBUG - Sending 68 bytes
> 03-12-20 21:30:33.795899 [7f439ef4d700] DEBUG - Expecting 8152 bytes in reply.
> 03-12-20 21:30:34.795925 [7f439ef4d700] WARNING - Closing UDP socket for URI "ipbusudp-2.0://192.168.3.18:50001" since deadline has passed
> 03-12-20 21:30:34.795985 [7f439ef4d700] ERROR - Timeout (1000 milliseconds) occurred for UDP receive from target with URI: ipbusudp-2.0://192.168.3.18:50001
> 03-12-20 21:30:34.796059 [7f43b2289740] WARNING - Closing Socket since exception detected.
> Traceback (most recent call last):
>   File "./run.py", line 113, in <module>
>     mem = jadepix_dev.read_ipb_data_fifo(slice_size, safe_style=True)
>   File "/home/sdong/work/JadePix3/software/lib/jadepix_device.py", line 301, in read_ipb_data_fifo
>     return self._ipbus_link.read_ipb_data_fifo(self.reg_name_base, "DATA_FIFO", num, safe_style)
>   File "/home/sdong/work/JadePix3/software/lib/ipbus_link.py", line 79, in read_ipb_data_fifo
>     self._hw.dispatch()
> uhal._core.UdpTimeout: Timeout (1000 milliseconds) occurred for UDP receive from target with URI: ipbusudp-2.0://192.168.3.18:50001

By using Wireshark to monitor the transmission, when num = 2028, PC receive a package which data size is 8190, when num = 2029, then it's 8194.
The rx/tx buffer depth in FPGA is 8192, I am not sure whether there is a connetion.

Cheers,
Sheng

tswilliams, 2020-12-06:
Hi,

>  The rx/tx buffer depth in FPGA is 8192, I am not sure whether there is a connetion.

Good point. Can you rerun your tests after reducing the return value of the `getMaxSendSize()` and `getMaxReplySize()` methods in `uhal/uhal/src/common/ProtocolUDP.cpp` to 8140 (and then recompiling uHAL)?

Cheers,
Tom

habrade, 2020-12-07:
Hi Tom,

> Can you rerun your tests after reducing the return value of the getMaxSendSize() and getMaxReplySize() methods in uhal/uhal/src/common/ProtocolUDP.cpp to 8140 (and then recompiling uHAL)?

Yes, set them to 8140 will work.

If suppose the depth of rx/tx buffer is the MTU, for my case, the max data(rx/tx) size (UDP data size) shoule be
` 8192(MTU) - 20(IP) - 8(UDP) = 8164` bytes.

And I did some tests, set the return value less than **8151** will work. (8151 also work)

Do you have any idea why is this number(8151)?

cheers,
Sheng

habrade, 2020-12-07:
Hi,

I checked the ethernet package via Wireshark, I found the 8192 should be the whole ethernet package length, which means the 
MTU = 8192 - 14(length of Ethernet II header, dst mac(6) + src mac(6) + type(2)) = 8178.

then the return value of the getMaxSendSize() and getMaxReplySize() methods in uhal/uhal/src/common/ProtocolUDP.cpp can be maxium set to **8150** = 8178(MTU) - 20(IP) - 8(UDP)

But still can not explain why 8151 works...

Cheers,
Sheng

commit 79095e43 (2021-01-26): uHAL: Add URI attribute for specifying maximum size of IPbus packets. Fixes #221
closed 2021-01-26 by tswilliams