ipbus/ipbus-software #222: Problem with building on Debian/testing

wzab, 2020-12-02:
When I try to build ipbus-software on a Debian/testing machine, I get the following error:
OS Detected: unknown.os
escript ./rebar get-deps compile
Uncaught error in rebar_core: {'EXIT',
                               {undef,
                                [{rebar_utils,get_cwd,[],[]},
                                 {rebar_config,new,0,[]},
                                 {rebar,init_config,1,[]},
                                 {rebar,run,1,[]},
                                 {rebar,main,1,[]},
                                 {escript,run,2,
                                  [{file,"escript.erl"},{line,758}]},
                                 {escript,start,1,
                                  [{file,"escript.erl"},{line,277}]},
                                 {init,start_em,1,[]}]}}
=ERROR REPORT==== 2-Dec-2020::15:37:43.426041 ===
beam/beam_load.c(1621): Error loading module rebar_utils:
  please re-compile this module with an 23 compiler (old-style fun with indices: 3/6)

tswilliams, 2020-12-03:
Hi,

Which version of Erlang is installed on that machine? So far we've tested the software up to Erlang v22, and I'm wondering if your machine has a more recent version installed.

Cheers,
Tom

tswilliams, 2021-01-22:
Hi,

I just merged a fix for this into the master branch. This will be included in the next release.

Cheers,
Tom

closed 2021-01-22 by tswilliams