ipbus/ipbus-software #223: Error when building

m-kru, 2020-12-13:
I am trying to build current master, but I get following error:
```
escript ./rebar get-deps compile
Uncaught error in rebar_core: {'EXIT',
                               {undef,
                                [{rebar_utils,get_cwd,[],[]},
                                 {rebar_config,new,0,[]},
                                 {rebar,init_config,1,[]},
                                 {rebar,run,1,[]},
                                 {rebar,main,1,[]},
                                 {escript,run,2,
                                  [{file,"escript.erl"},{line,758}]},
                                 {escript,start,1,
                                  [{file,"escript.erl"},{line,277}]},
                                 {init,start_em,1,[]}]}}
=ERROR REPORT==== 13-Dec-2020::19:09:16.197609 ===
beam/beam_load.c(1621): Error loading module rebar_utils:
  please re-compile this module with an 23 compiler (old-style fun with indices: 3/6)

=ERROR REPORT==== 13-Dec-2020::19:09:16.197631 ===
Loading of /home/mkru/workspace/ipbus-software/controlhub/rebar/rebar/ebin/rebar_utils.beam failed: badfile

=ERROR REPORT==== 13-Dec-2020::19:09:16.214943 ===
Loading of /home/mkru/workspace/ipbus-software/controlhub/rebar/rebar/ebin/rebar_utils.beam failed: badfile

=ERROR REPORT==== 13-Dec-2020::19:09:16.214912 ===
beam/beam_load.c(1621): Error loading module rebar_utils:
  please re-compile this module with an 23 compiler (old-style fun with indices: 3/6)

escript: exception error: undefined function rebar_utils:delayed_halt/1
  in function  escript:run/2 (escript.erl, line 758)
  in call from escript:start/1 (escript.erl, line 277)
  in call from init:start_em/1 
  in call from init:do_boot/3 
make[1]: *** [Makefile:65: _all] Error 127
make[1]: Leaving directory '/home/mkru/workspace/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2
```
Does anyone know what might be wrong?

tswilliams, 2020-12-14:
Hi,

Which operating system are you building the software on, and which version of Erlang is installed? So far we've tested the software up to Erlang v22, and I'm wondering if your machine has a more recent version installed.

Cheers,
Tom

m-kru, 2020-12-14:
I am using Linux arch, but the issue is probably cause by the Erlang.
```
Erlang/OTP 23 [erts-11.1.3] [source] [64-bit] [smp:12:12] [ds:12:12:10] [async-threads:1] [hipe]
```

tswilliams, 2021-01-22:
Hi,

I just merged a fix for this into the `master` branch. This will be included in the next release.

Cheers,
Tom

commit 4137e33b (2021-01-22): ControlHub: Add newer release of rebar (from Fedora 33 erlang-rebar-2.6.4-16.fc33.noarch RPM) to support Erlang 23. Fixes #223
closed 2021-01-22 by tswilliams