ipbus/ipbus-software #232: Unused parameters in uhal/log/log_inserters.type.hxx

strilov, 2021-01-12:
Hi,

There appear to be some unused parameters in uhal/log/log_inserters.type.hxx resulting in the following compiler warnings below. Is this something which you could help with?

/cvmfs/dune.opensciencegrid.org/dunedaq/DUNE/products_dev/uhal/v2_6_4/include/uhal/log/log_inserters.type.hxx:45:70: warning: unused parameter ‘aType’ [-Wunused-parameter]
   std::ostream& operator<< ( std::ostream& aStr ,  const _Type< T >& aType )

/cvmfs/dune.opensciencegrid.org/dunedaq/DUNE/products_dev/uhal/v2_6_4/include/uhal/log/log_inserters.type.hxx:66:30: warning: unused parameter ‘aT’ [-Wunused-parameter]
   _Type< T > Type ( const T& aT )

Regards,
Stoyan

tswilliams, 2021-01-13:
Hi,

I've removed the name of that unused parameter. I've merged this changed into the `master` branch, so it will be included in the next release.

Cheers,
Tom

strilov, 2021-01-13:
Hi Tom,

Thanks very much for looking into this. There we two unused parameters, one at line 45 (now fixed), and another at line 66. Could you take a look at the one on line 66?

Regards,
Stoyan

closed 2021-01-22 by tswilliams

tswilliams, 2021-01-22:
Hi Stoyan,

I've just merged a fix for this into the master branch (along with fixes some other compiler warnings).

Cheers,
Tom

PR #233 (2021-01-13): Remove name of unused argument
Very minor change - fixes #232
commit 78414d62 (2021-01-13): uHAL: Remove name of unused argument. Fixes #232
commit c4f4b372 (2021-01-21): uHAL: Resolve compiler warnings. Fixes #232