ipbus/ipbus-software #237: Error in buildindg the code (make command)

tonaspiuck, 2021-01-28:
Hi, 

I have problem in the make step:

--First installing via YUM and build the code I got an error:

/usr/bin/ld: can't find -lcactus_extern_pugixml

--Second, I just couldn't overcome the problem so I try compiling and installing from source, and I got the next error:
ERROR: 'get-deps' failed while processing .../pbus-software/controlhub: rebar_abort

OS is Centos7

Does anyone can help me with this issue please?

tswilliams, 2021-01-28:
Hi,

> --First installing via YUM and build the code I got an error:
> 
> /usr/bin/ld: can't find -lcactus_extern_pugixml

I assume you encountered this error when using the example makefile linked from https://ipbus.web.cern.ch/doc/user/html/software/uhalQuickTutorial.html    Is that correct? Sorry, that line of the makefile is out-of-date; it should now be `-lpugixml` rather than `-lcactus_extern_pugixml` - I've updated the file on the website accordingly. Can you see if making that change solves your problem?

Cheers,
Tom

tonaspiuck, 2021-01-28:
Perfect, thank you very much!

Have a nice day!

closed 2021-01-28 by tonaspiuck