ipbus/ipbus-software #239: making controlhub work with LArC board which is handled in software.

carmijo-az, 2021-03-01:
Hi,

The Liquid Argon Carrier (LArC) board  responds to ipbus commands using a software interface.  There are no problems with handling udp packets and ipbus commands.  The board now needs to respond to control hub packets.  It is necessary to know what commands are to be expected from control hub packets and what responses are expected by control hub.  Does the control hub communicate with the boards via udp or is it necessary to use tcp on port 10203?

Thanks for your assistance,

Charlie Armijo
University of Arizona
carmijo@email.arizona.edu

carmijo-az, 2021-03-23:
Let me try this again.  

This is from a README.md in ipbus firmware.
* **ControlHub** : A software application that mediates simultaneous hardware access
    from multiple uHAL clients, and implements the IPbus reliability
    mechanism over UDP

I am getting complaints about our boards not working correctly with control hub.  In order to fix this I would like to know the following.
Does control hub send packets to the destination hardware?  
I don't see anything in the ipbus firmware that suggests that it is receiving or sending tcp packets, but the documentation suggests that control hub uses tcp packets.  

Does control hub expect some response from our hardware?  If so, I will reconfigure our code to respond correctly.  

Thanks for answering these questions.
Charlie Armijo

tswilliams, 2021-03-23:
Hi,

The ControlHub communicates with uHAL over TCP, and with IPbus devices using UDP.

More specifically, uHAL sends sequences of IPbus request packets to the ControlHub in a TCP stream; the ControlHub forwards each of these IPbus request packets to the boards as UDP packets, and waits for the corresponding IPbus response packets (again UDP) to be sent back, then forwards those to the relevant uHAL client in a TCP stream.

When first communicating with any specific IPbus device, in order to determine how many packets can be in flight at any given time the ControlHub sends an IPbus status request (which the IPbus device must reply to). Additionally, if the ControlHub does not receive the next IPbus response packet after a certain timeout period, it sends a status request to determine whether the request or reply packet was lost, and sends to the IPbus device an IPbus resend request if the reply packet got lost. (More details on these packet types can be found in the protocol spec linked from https://ipbus.web.cern.ch/doc/user/html/FAQ.html)

Overall, compared to 'direct UDP' communication from uHAL, the main difference when a ControlHub routes packets to an IPbus device is that IPbus status and resend request packets are sent.

Cheers,
Tom

carmijo-az, 2021-03-23:
Thanks for your answers.  

I recently provided a response to the status request.  It sends a correct response without errors when going through ipbus-2.0, but apparently not when using chtcp-2.0.     

The ControlHub did not receive any response from the target with URI "chtcp-2.0://pc-lar-ptm-03:10203?target=aca-lar-larc-025.cern.ch:50001"
ControlHub returned error code 3 = 'no reply to status packet'

I'm going to close this issue, but I may have more questions later.

Cheers,
Charlie Armijo

closed 2021-03-23 by carmijo-az

tswilliams, 2021-03-23:
One quick comment to clarify the difference between the packets seen by the IPbus device in the two scenarios: when using an `ipbusudp-2.0` URI, uHAL does not send any status requests to the IPbus device; however, the ControlHub will send status requests to the device