ipbus/ipbus-software #242: Udptimeout

Zeks-lzhang, 2021-03-17:
Dear IPBus experts.
  I want to use IPBus to read streaming data or registers, so I have some codes like this:
```
while(true)
{
  hw.getNode ( "REG" ).read();
}
```
And there is a signal called dummy_ack so that timeout won't happen.
But I got the  (uhal::exception::UdpTimeout), and it will happen at an uncertain time, also depends on the computer I used.
To find something, I keep the data fifo empty and run the code upon, it may cost 5-60 sec to get the Udptimeout error.
I think my dummy_ack signal was working otherwise it would error immediately.
Btw, if I change the code to this:
```
while(true)
{
  usleep(1000);
  hw.getNode ( "REG" ).read();
}
```
 it may cost much more time(several minutes and even hours) to get the Udptimeout error.
Could you tell me what this is all about? Or IPBus should not be used like this?
Many thanks！

tswilliams, 2021-03-22:
Hi,

If a `UdpTimeout` exception was thrown, that indicates that uHAL did not receive any reply packet within the timeout period (1 second by default, the exact value though in your case should be printed in the exception message). This could happen for example if some component in your network (e.g. a switch or NIC) occasionally drops either the UDP request/reply packet (due to an internal buffer overflow), or if your computer takes longer than the timeout period to process the network traffice.

If you need reliable communication with your boards, we recommend that rather than communicating with them directly over UDP from uHAL, you route your traffic via a ControlHub instance (see [this section](https://ipbus.web.cern.ch/doc/user/html/software/uhalQuickTutorial.html#working-with-the-controlhub) at the end of the uHAL quick tutorial). Communication between uHAL and the ControlHub will occur over TCP streams, and the ControlHub implements a retransmission mechinism built into the IPbus protocol in order to recover any lost UDP packets.

I hope that answers your question and solves your problem. Let us know if not.

Cheers,
Tom

Zeks-lzhang, 2021-03-22:
I use chtcp instead and it seems to be solved now! 
Thanks a lot!

closed 2021-03-22 by Zeks-lzhang