ipbus/ipbus-software #243: Option to switch off PCIE support while building uhal

habrade, 2021-03-22:
Dear developers,
I am trying to build uhal on MacOS (BigSur, Apple M1).
However, there is a problem I can not fix.

The info on the terminal:
> /opt/ipbus-software/uhal/uhal/src/common/ProtocolPCIe.cpp:362:43: error: use of undeclared identifier 'PTHREAD_MUTEX_ROBUST'
>   s = pthread_mutexattr_setrobust(&lAttr, PTHREAD_MUTEX_ROBUST);
>                                           ^
> /opt/ipbus-software/uhal/uhal/src/common/ProtocolPCIe.cpp:388:9: error: use of undeclared identifier 'pthread_mutex_consistent'; did you mean 'pthread_mutex_destroy'?
>     s = pthread_mutex_consistent(&mMutex);
>         ^~~~~~~~~~~~~~~~~~~~~~~~
>         pthread_mutex_destroy
> /Library/Developer/CommandLineTools/SDKs/MacOSX.sdk/usr/include/pthread.h:369:5: note: 'pthread_mutex_destroy' declared here
> int pthread_mutex_destroy(pthread_mutex_t *);
>     ^
> 2 errors generated.

I tried " -DMDB_USE_ROBUST=0", but doesn't work.
I found "PTHREAD_MUTEX_ROBUST" is not supported on macOS,  so I am wondering is there an option to switch off PCI-E  support in IPbus, or is it possible to modify some Makefiles?

Many thanks,
Sheng

tswilliams, 2021-03-23:
Hi

As `PTHREAD_MUTEX_ROBUST` is not supported on macOS, for the short term I suggest that you locally edit `uhal/uhal/src/common/ProtocolPCIe.cpp` to remove the lines that cause the compilation errors. (As long as you don't comment out entire functions, if you're not using the PCIe client, this won't have any adverse effects.)

In the meantime, I'll work on a more maintainable solution to avoid needing to modify code in the long run.

Cheers,
Tom

habrade, 2021-03-27:
Hi Tom,

I modified the source file which caused the error, and the "PCIE" related error were gone.
However, I got an error about the system architecture.

> g++-10 -shared -Wl,-install_name,libcactus_uhal_uhal.so.2.8 -Wall -g -O3 -fPIC /opt/ipbus-software/uhal/uhal/obj/Buffers.o /opt/ipbus-software/uhal/uhal/obj/ClientFactory.o /opt/ipbus-software/uhal/uhal/obj/ClientInterface.o /opt/ipbus-software/uhal/uhal/obj/ConnectionManager.o /opt/ipbus-software/uhal/uhal/obj/DerivedNodeFactory.o /opt/ipbus-software/uhal/uhal/obj/HwInterface.o /opt/ipbus-software/uhal/uhal/obj/IPbusInspector.o /opt/ipbus-software/uhal/uhal/obj/Node.o /opt/ipbus-software/uhal/uhal/obj/NodeTreeBuilder.o /opt/ipbus-software/uhal/uhal/obj/ProtocolControlHub.o /opt/ipbus-software/uhal/uhal/obj/ProtocolIPbus.o /opt/ipbus-software/uhal/uhal/obj/ProtocolIPbusCore.o /opt/ipbus-software/uhal/uhal/obj/ProtocolMmap.o /opt/ipbus-software/uhal/uhal/obj/ProtocolPCIe.o /opt/ipbus-software/uhal/uhal/obj/ProtocolTCP.o /opt/ipbus-software/uhal/uhal/obj/ProtocolUDP.o /opt/ipbus-software/uhal/uhal/obj/ValMem.o /opt/ipbus-software/uhal/uhal/obj/detail/utilities.o /opt/ipbus-software/uhal/uhal/obj/utilities/TimeIntervalStats.o /opt/ipbus-software/uhal/uhal/obj/utilities/bits.o /opt/ipbus-software/uhal/uhal/obj/utilities/files.o /opt/ipbus-software/uhal/uhal/obj/utilities/xml.o -L/opt/homebrew/Cellar/boost/1.75.0_2/lib -L/opt/homebrew/Cellar/pugixml/1.11.4_2/lib -L/opt/ipbus-software/uhal/log/lib -L/opt/ipbus-software/uhal/grammars/lib -lpthread -lpugixml -lboost_filesystem -lboost_regex -lboost_system -lboost_chrono -lcactus_uhal_grammars -lcactus_uhal_log -o lib/libcactus_uhal_uhal.so.2.8.0
> Undefined symbols for architecture arm64:
>   "__ZN5boost11basic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE9do_assignEPKcS7_j", referenced from:
>       __ZN5boost11basic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE6assignEPKcS7_j in ConnectionManager.o
>       __ZN5boost11basic_regexIcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE6assignEPKcS7_j in Node.o
>   "__ZN5boost16re_detail_10750012perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE14construct_initERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsE", referenced from:
>       __ZN5boost16re_detail_10750012perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEC1ES3_S3_RNS_13match_resultsIS3_S6_EERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsES3_ in ConnectionManager.o
>       __ZN5boost16re_detail_10750012perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEC1ES3_S3_RNS_13match_resultsIS3_S6_EERKNS_11basic_regexIcSA_EENS_15regex_constants12_match_flagsES3_ in Node.o
>   "__ZN5boost16re_detail_10750012perl_matcherIPKcSaINS_9sub_matchIS3_EEENS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEE5matchEv", referenced from:
>       __ZN5boost11regex_matchIPKcSaINS_9sub_matchIS2_EEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEEbT_SA_RNS_13match_resultsISA_T0_EERKNS_11basic_regexIT1_T2_EENS_15regex_constants12_match_flagsE in ConnectionManager.o
>       __ZN5boost11regex_matchIPKcSaINS_9sub_matchIS2_EEEcNS_12regex_traitsIcNS_16cpp_regex_traitsIcEEEEEEbT_SA_RNS_13match_resultsISA_T0_EERKNS_11basic_regexIT1_T2_EENS_15regex_constants12_match_flagsE in Node.o
> ld: symbol(s) not found for architecture arm64

It seems pretty hard to build uHal on the macOS (Apple M1) for now...
Anyway, thanks very much for your reply and suggestion.

Cheers,
Sheng