ipbus/ipbus-software #245: controlhub closing before response.

carmijo-az, 2021-04-13:
If I send an ipbus request, the tcp socket is closed before the packet is received at the destination.  The 4 subsequent packets sent by controlhub are not seen in wireshark until many seconds after the socket has been closed.  Adding time to the timeout in sys.config has no effect on this.  Ipbus queries work as usual.  We are using ipbus/controlhub 2.8.0.  We were using 2.7.9, but that did not work either.

Charlie Armijo
University of Arizona - Physics

 [larctest@eepp-xilinx2]$ **python testReg.py 6**
atca.udp.larc
13-04-21 12:53:12.349170 [140565606647552] WARNING - Closing TCP socket for device with URI "chtcp-2.0://localhost:10203?target=192.168.0.172:50001" since deadline has passed
13-04-21 12:53:12.349596 [140565606647552] ERROR - Timeout (1000 ms) occurred for receive (header) from ControlHub with URI 'chtcp-2.0://localhost:10203?target=192.168.0.172:50001'. 1 packets in flight, 1 in this chunk (20/24 bytes sent/expected). Last send / receive queued 1000.42 / 1000.39 ms ago.
13-04-21 12:53:12.349660 [140565606647552] ERROR - Extra timeout-related info - round-trip times: no values recorded
13-04-21 12:53:12.349711 [140565606647552] ERROR - Extra timeout-related info - send-recv  times: no values recorded
13-04-21 12:53:12.349757 [140565606647552] ERROR - Extra timeout-related info - inter-send times: no values recorded
13-04-21 12:53:12.349800 [140565606647552] ERROR - Extra timeout-related info - inter-recv times: no values recorded
Traceback (most recent call last):
  File "testReg.py", line 28, in <module>
    hw.dispatch()    
uhal._core.TcpTimeout

**This works!**
[larctest@eepp-xilinx2]$ **python testReg.py 7**
atca.udp.larc
sanity =  0xc0ffee00

[larctest@eepp-xilinx2$ **/opt/cactus/bin/controlhub_info**
The ControlHub is running. General info ...
    started at 2021-04-13 19:45:25 UTC
    uptime:    0 days, 0 hours, 8 minutes, 30 seconds
    version:   2.8.0
    config file: /etc/controlhub/sys.config

Device client parameters ...
    max packets in flight to each board = 16
    timeout for replies from board = 15000ms
    device client lifetime when inactive = 15000ms

TCP parameters ...
    Listening on port 10203

[larctest@eepp-xilinx2]$ **/opt/cactus/bin/controlhub_stats**
Control Hub Stats Report
------------------------

CLIENT  All-time connections: 3
          Active connections: 0 (peak: 1)
  Old clients
           Requests received: 3 (of which 0 were malformed)
              Responses sent: 0

Old UDP
              Packets Out: 15
               Packets In: 0
                 Timeouts: 0 in normal operation (of which 0 were recovered)
                             (0 request packets got lost, 0 reply packet)
           Other timeouts: 15 (status), 0 (in resends)

[root@eepp-xilinx2]# **systemctl status controlhub -l**
● controlhub.service - IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
   Loaded: loaded (/usr/lib/systemd/system/controlhub.service; enabled; vendor preset: disabled)
   Active: active (running) since Tue 2021-04-13 12:45:27 MST; 8min ago
     Docs: http://ipbus.web.cern.ch/ipbus/doc/index.html
  Process: 5501 ExecStop=/opt/cactus/bin/controlhub_stop (code=exited, status=1/FAILURE)
  Process: 5761 ExecStartPost=/bin/sh -c /opt/cactus/lib/controlhub/bin/controlhub getpid > /var/run/controlhub.pid (code=exited, status=0/SUCCESS)
  Process: 5556 ExecStart=/opt/cactus/bin/controlhub_start (code=exited, status=0/SUCCESS)
 Main PID: 5695 (beam.smp)
    Tasks: 24
   CGroup: /system.slice/controlhub.service
           ├─5605 /opt/cactus/lib/controlhub/erts-5.10.4/bin/epmd -daemon
           └─5695 /opt/cactus/lib/controlhub/erts-5.10.4/bin/beam.smp -zdbbl 2097151 -sbt db -scl false -sbwt very_long -swt low -- -root /opt/cactus/lib/controlhub -progname controlhub -- -home $CACTUSROOT -- -boot /opt/cactus/lib/controlhub/releases/2.8.0/controlhub -mode embedded -config /etc/controlhub/sys.config -controlhub config_file "/etc/controlhub/sys.config" -sname controlhub -setcookie ch_cookie -- console -noshell -noinput ''

Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:45:25.522 [notice] <0.1072.0> - Starting the ControlHub application (version 2.8.0).
Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:45:25.523 [notice] <0.1073.0> - ControlHub root supervisor starting now.
Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:45:25.523 [notice] <0.1074.0> - Initialising the stats server.
Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:45:25.523 [notice] <0.1075.0> - Initialising the device client registry.
Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:45:25.523 [notice] <0.1076.0> - Initialising the TCP listener (port 10203).
Apr 13 12:45:25 eepp-xilinx2.physics.arizona.edu controlhub_start[5556]: Starting ControlHub ... ok
Apr 13 12:45:27 eepp-xilinx2.physics.arizona.edu systemd[1]: Started IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications).
Apr 13 12:46:08 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:46:08.828 [warning] <0.1077.0> - TCP socket was closed early - there is still 1 replies pending for that socket. Presumably the socket was closed by the TCP client.
Apr 13 12:48:46 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:48:46.972 [warning] <0.1089.0> - TCP socket was closed early - there is still 1 replies pending for that socket. Presumably the socket was closed by the TCP client.
Apr 13 12:53:12 eepp-xilinx2.physics.arizona.edu controlhub[5695]: 2021-04-13 12:53:12.349 [warning] <0.1093.0> - TCP socket was closed early - there is still 1 replies pending for that socket. Presumably the socket was closed by the TCP client.

carmijo-az, 2021-04-13:
I forgot to add that we are using the latest version of centos7.  The rack containing the destination board and the centos computer are adjacent.  So, distance is not an issue.

Cheers,
Charlie Armijo

tswilliams, 2021-04-13:
Hi,

In this case, the TCP socket is being closed by uHAL, due to the uHAL client timeout being reached (it looks like the default 1-second timeout is being used here). This can be changed using the [HwInterface::setTimeoutPeriod method](https://ipbus.web.cern.ch/sw/release/2.7/api/html/d9/d7b/classuhal_1_1_hw_interface.html#afc94d0e45ec5a0e4b99804c7922fe670)

That said, based on the output of the `controlhub_stats` command, it looks like the ControlHub may not be receiving any response to the status requests that it sends to the boards. In wireshark, do you see any packets being sent back from the boards? Are you able to talk to those boards directly with uHAL from the same computer as the ControlHub is running on?

Cheers,
Tom

carmijo-az, 2021-04-14:
Hi, Tom.

Thanks for pointing to the HwInterface::setTimeoutPeriod method.

The destination board is the LArC (Liquid Argon Carrier).  The ipbus code on that board is in software, not firmware.  (At the time there was no ipbus firmware for the virtex 7.)  This code has worked with ipbus for years.  It seems to correctly respond to a status request via ipbus.  The same code running at EMF at CERN gives "error 3" with controlhub,  but controlhub definitely sees the response.  In wireshark here  the response appears immediately.  I'm in the process of trying to get controlhub to work at Arizona so that I can duplicate the error and fix the problem.  Ipbus and controlhub are running on the same computer.  Ipbus has been working fine here for years.  

Thanks for your assistance,
Charlie

tswilliams, 2021-04-14:
Hi,

>  The same code running at EMF at CERN gives "error 3" with controlhub, but controlhub definitely sees the response.
This error code means that the ControlHub did not receive a reply to a status packet request (I believe that this translation is included in the corresponding uHAL error message - but if not let me know).

>  In wireshark here the response appears immediately.
An encoding error in creating the status packet response could cause the symptoms you're reporting; if you send me the contents of the status packet response that you see in wireshark (or attach a saved copy of the captured packets), I can take a look. 

Cheers,
Tom

carmijo-az, 2021-04-27:
Hi, Tom.

Okay after getting rid of some other issues, he is a screen shot of the our status request response.

[image]

I'm still getting the timeout error.  Any comments about the structure of the packet would be much appreciated.

Cheers,
Charlie

tswilliams, 2021-04-27:
Hi,

Thanks for the screenshot. If I'm reading it correctly, the first four 32-bit words in the UDP payload are:
```
20 00 00 f1
00 00 05 c0
04 00 00 00 
00 00 00 01
```
The value of the 4th 32-bit word in the payload appears to be incorrect. It should be the next expected control packet header - e.g. `20 00 01 f0` if the next expected packet ID is `01`.

Cheers,
Tom

carmijo-az, 2021-04-28:
Thanks, Tom!

That seems to have been the last fix needed.  Control hub is responding appropriately.

Cheers,
-Charlie

closed 2021-04-28 by carmijo-az