ipbus/ipbus-software #261: make rpm fails in RHEL8

aylons, 2022-03-29:
I was able to install with traditional make install, however.

The command was make PYTHON=python3 rpm

```

error: parse error in expression
error: /home/xxxx/ipbus-software/uhal/log/rpm/cactuscore-uhal-log.spec:136: bad %if condition
make[2]: *** [/home/xxx/ipbus-software/uhal/config/mfRPMRules.mk:24: _rpmbuild] Error 1
```

Relevant part of mentioned file is:

```
%if %{defined _build_debuginfo_package}
# Temporary workaround for subpackages not being built on CentOS8
%if %{_os} != centos8
%files debuginfo
%defattr(-,root,root,-)
%endif

/usr/lib/debug
/usr/src/debug
%endif

```

The system is a RHEL 8 running python 3.6.8.

tswilliams, 2022-03-30:
Hi @aylons

I've implemented a fix in branch `bug/rhel8-rpmbuild-error`. This resolves the problem locally for me. Could you try again using that branch? If that works for you, I'll merge and create a new tag.

Cheers,
Tom

aylons, 2022-03-31:
Seems to have worked. As a double check, this is the result of  yum localinstall `find . -iname "*.rpm"`
```

===================================================================================================================================================================================================================
 Package                                                    Architecture                             Version                                                  Repository                                      Size
===================================================================================================================================================================================================================
Installing:
 cactuscore-uhal-grammars                                   x86_64                                   2.8.3-1.el8.gcc8_5_0                                     @commandline                                   3.7 M
 cactuscore-uhal-log                                        x86_64                                   2.8.3-1.el8.gcc8_5_0                                     @commandline                                   136 k
 cactuscore-uhal-tests                                      x86_64                                   2.8.3-1.el8.gcc8_5_0                                     @commandline                                    16 M
 cactuscore-uhal-uhal                                       x86_64                                   2.8.3-1.el8.gcc8_5_0                                     @commandline                                    12 M

```

closed 2022-04-01 by tswilliams

tswilliams, 2022-04-03:
Thanks for the feedback. The `yum localinstall` output that you quoted looks good. I've now included this change in the `v2.8.3` tag.

PR #262 (2022-04-01): uHAL: Fix RHEL8 RPM build error
Fixes #261

At the same time, I changed the `.centosN` release field suffix to the standard `.elN`, taken from the RPM `dist` variable
commit 90dd380c (2022-04-01): uHAL: Fix RHEL8 RPM build error. Change '.centosN' release field suffix to standard '.el8' from RPM dist variable. Fixes #261