ipbus/ipbus-software #264: ControlHub resend error

carmijo-az, 2022-04-18:
Hi,  we are running a software version of ipbus on a custom board for the atlas liquid argon calorimeter.  While running a stress test in the emf lab at cern,  We received the following error from ControlHub.   
ERROR - The ControlHub did not receive any response from the target with URI
"chtcp-2.0://pcemf-pm-02.cern.ch:10203?target=172.26.80.148:50001" [0m
 [0;31m15-04-22 13:47:42.723826 [140696819611392] ERROR - ControlHub
returned error code 4 = 'no reply to resend request' [0m
 [0;33m15-04-22 13:47:42.724090 [140697012963136] WARNING - Closing TCP
socket for device with URI
"chtcp-2.0://pcemf-pm-02.cern.ch:10203?target=172.26.80.148:50001" since
exception detected. [0m

The ipbus protocol says that if the packet is not found, our code should return nothing.  Currently, we return the header and pad the remainder of the 64 byte payload with zero.  Is this not correct?  This is a bit difficult for us since we cannot create this problem here at Arizona and wireshark is not allowed to run at CERN.  If we use a script to send a resend packet here, the result is successful.  What exactly is ContorlHub expecting here?  Particularly, what is ControlHub expecting if the requested packet history cannot be found?

Thanks for your assistance,
Charlie Armijo
University of Arizona, Physics Department.
carmijo@email.arizona.edu

tswilliams, 2022-04-19:
Hi,

If an IPbus target receives a resend request packet, then it should resend the specified reply control packet if the corresponding request packet was one of the last `nResponseBuffers` packets received (as the target should keep a copy of the last `nResponseBuffers` reply packets in a buffer); otherwise, it should not respond to the resend request packet.

The ControlHub will only send `nResponseBuffers` request packets to a target before receiving the replies to those packets (this is referred to as the maximum number of requests in flight); this is to ensure that if a response packet is lost and the the ControlHub needs to ask for the target to resend that response, the target will still have a copy of that packet. Therefore for a correctly functioning IPbus target, in principle the error message you quoted should never be seen.

I hope that's clear. Cheers,
Tom

charles-armijo, 2022-04-21:
Thanks, Tom.

In our case, the nResponseBuffer value is 4.   max packets in flight to each board = 16.  timeout for replies from board = 20ms.  The tester had over twenty terminals open, each sending continuous loops of read or write commands to this one board,  Is it possible that, under these conditions, by the time ipbus on our end got the resend message the data was no longer in our history buffer?   In which case, there should be no reply to the resend request and we should not see error code 4 = 'no reply to resend request.'  

There is an additional question.  If the resend request comes from contolhub and we do not reply because the information is no longer in our history buffer, is this really an error or merely an information notice?

Sorry if I'm a bit vague here, but neither we at Arizona nor colleagues SMU have seen this error running similar tests at our institutions.  So, I'm not sure whether this was simply a dropped packet somewhere or an actual fault in our system response.  We also don't know the traffic level on the 1G network at CERN nor how many switches are involved between the board under test and the computer running ipbus/controhub and the computer running the terminals.

Cheers,

Charlie Armijo
Charlie Armijo
University of Arizona, Physics Department.
[carmijo@email.arizona.edu]

tswilliams, 2022-04-21:
Hi,

> In our case, the nResponseBuffer value is 4. max packets in flight to each board = 16. timeout for replies from board = 20ms. The tester had over twenty terminals open, each sending continuous loops [...]

In this scenario with multiple clients, all of those clients should be routing communication with the board through the same ControlHub instance, and so the ControlHub would make sure that no more than 4 packets were in flight with the board at any one time. As a result, even if a reply packet is dropped in the network path, it must still be present in the board's history buffer when the ControlHub requests that the reply be sent again.

> There is an additional question. If the resend request comes from contolhub and we do not reply because the information is no longer in our history buffer, is this really an error or merely an information notice?
In practice, this does indicate a genuine error, as by the design of the protocol, the history buffer should always contain the reply packets that the ControlHub may need to request be resent (since the ControlHub limits the number of packets in flight to the board to be less than nResponseBuffers)

Cheers,
Tom

tswilliams, 2022-05-16:
Closing since no response in a while. Please re-open if you have any further questions on this.