ipbus/ipbus-software #265: Support for c++17

dinardo, 2022-04-27:
Dear experts,
is it possible to compile IPbus with `--std=c++1z` flag?

Many thanks,
Mauro.

tswilliams, 2022-05-04:
Hi,

This should be possible by ovrerriding `CXXFLAGS` when running `make` - i.e. `make CXXFLAGS="-g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z"`

Cheers,
Tom

dinardo, 2022-05-10:
Many thanks. I used this make command:

`make EXTERN_BOOST_INCLUDE_PREFIX=/home/testbeam/boost_v1_77_0/ EXTERN_BOOST_LIB_PREFIX=/home/testbeam/boost_v1_77_0/stage/lib/ CXXFLAGS="-g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z"`

I'm getting this error:
```
mkdir -p /home/testbeam/ipbus-software/uhal/tests/lib
g++ -shared -Wl,-soname,libcactus_uhal_tests.so.2.8 -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/testbeam/ipbus-software/uhal/tests/obj/DummyClient.o /home/testbeam/ipbus-software/uhal/tests/obj/DummyDerivedNode.o /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/PCIeDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/PerfTester_static.o /home/testbeam/ipbus-software/uhal/tests/obj/TCPDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/UDPDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/fixtures.o /home/testbeam/ipbus-software/uhal/tests/obj/test_block.o /home/testbeam/ipbus-software/uhal/tests/obj/test_check_permissions.o /home/testbeam/ipbus-software/uhal/tests/obj/test_client_factory.o /home/testbeam/ipbus-software/uhal/tests/obj/test_config_space.o /home/testbeam/ipbus-software/uhal/tests/obj/test_empty_dispatch.o /home/testbeam/ipbus-software/uhal/tests/obj/test_hierarchy.o /home/testbeam/ipbus-software/uhal/tests/obj/test_masking.o /home/testbeam/ipbus-software/uhal/tests/obj/test_multithreaded.o /home/testbeam/ipbus-software/uhal/tests/obj/test_nodes.o /home/testbeam/ipbus-software/uhal/tests/obj/test_nonreachable.o /home/testbeam/ipbus-software/uhal/tests/obj/test_rawclient.o /home/testbeam/ipbus-software/uhal/tests/obj/test_single.o /home/testbeam/ipbus-software/uhal/tests/obj/test_soak.o /home/testbeam/ipbus-software/uhal/tests/obj/test_timeout.o /home/testbeam/ipbus-software/uhal/tests/obj/test_uri.o /home/testbeam/ipbus-software/uhal/tests/obj/tools.o -L/home/testbeam/boost_v1_77_0/stage/lib/ -L/home/testbeam/ipbus-software/uhal/log/lib -L/home/testbeam/ipbus-software/uhal/grammars/lib -L/home/testbeam/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -o lib/libcactus_uhal_tests.so.2.8.1
ln -s -f /home/testbeam/ipbus-software/uhal/tests/lib/libcactus_uhal_tests.so.2.8.1 lib/libcactus_uhal_tests.so.2.8
ln -s -f /home/testbeam/ipbus-software/uhal/tests/lib/libcactus_uhal_tests.so.2.8 lib/libcactus_uhal_tests.so
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z  -Iinclude -I/home/testbeam/boost_v1_77_0/ -I/home/testbeam/ipbus-software/uhal/log/include -I/home/testbeam/ipbus-software/uhal/grammars/include -I/home/testbeam/ipbus-software/uhal/uhal/include /home/testbeam/ipbus-software/uhal/tests/src/common/DummyHardwareTcp.cxx -o /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardwareTcp.o
mkdir -p /home/testbeam/ipbus-software/uhal/tests/bin
g++ -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardwareTcp.o -L/home/testbeam/boost_v1_77_0/stage/lib/ -L/home/testbeam/ipbus-software/uhal/log/lib -L/home/testbeam/ipbus-software/uhal/grammars/lib -L/home/testbeam/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_tests -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -lstdc++ -o /home/testbeam/ipbus-software/uhal/tests/bin/DummyHardwareTcp.exe
/opt/rh/devtoolset-10/root/usr/libexec/gcc/x86_64-redhat-linux/10/ld: warning: libboost_chrono.so.1.77.0, needed by /home/testbeam/ipbus-software/uhal/uhal/lib/libcactus_uhal_uhal.so, not found (try using -rpath or -rpath-link)
/opt/rh/devtoolset-10/root/usr/libexec/gcc/x86_64-redhat-linux/10/ld: warning: libboost_atomic.so.1.77.0, needed by /home/testbeam/boost_v1_77_0/stage/lib//libboost_filesystem.so, not found (try using -rpath or -rpath-link)
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z  -Iinclude -I/home/testbeam/boost_v1_77_0/ -I/home/testbeam/ipbus-software/uhal/log/include -I/home/testbeam/ipbus-software/uhal/grammars/include -I/home/testbeam/ipbus-software/uhal/uhal/include /home/testbeam/ipbus-software/uhal/tests/src/common/DummyHardwareUdp.cxx -o /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardwareUdp.o
g++ -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardwareUdp.o -L/home/testbeam/boost_v1_77_0/stage/lib/ -L/home/testbeam/ipbus-software/uhal/log/lib -L/home/testbeam/ipbus-software/uhal/grammars/lib -L/home/testbeam/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_tests -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -lstdc++ -o /home/testbeam/ipbus-software/uhal/tests/bin/DummyHardwareUdp.exe
/opt/rh/devtoolset-10/root/usr/libexec/gcc/x86_64-redhat-linux/10/ld: warning: libboost_chrono.so.1.77.0, needed by /home/testbeam/ipbus-software/uhal/uhal/lib/libcactus_uhal_uhal.so, not found (try using -rpath or -rpath-link)
/opt/rh/devtoolset-10/root/usr/libexec/gcc/x86_64-redhat-linux/10/ld: warning: libboost_atomic.so.1.77.0, needed by /home/testbeam/boost_v1_77_0/stage/lib//libboost_filesystem.so, not found (try using -rpath or -rpath-link)
[... cut ...]
make[2]: Leaving directory '/home/testbeam/ipbus-software/uhal/tests'
make[1]: *** [Makefile:49: tests/.virtual.Makefile] Error 2
make[1]: Leaving directory '/home/testbeam/ipbus-software/uhal'
make: *** [Makefile:58: uhal/.virtual.Makefile] Error 2
```
Both these two libraries are present precisely in the directory `/home/testbeam/boost_v1_77_0/stage/lib/`:
- libboost_chrono.so.1.77.0
- libboost_atomic.so.1.77.0

Instead, if don't use `CXXFLAGS="-g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z"` everything seems to be compiled properly

dinardo, 2022-05-10:
How can I force make to use ONLY my boost personal installation?

tswilliams, 2022-05-12:
Hi,

> How can I force make to use ONLY my boost personal installation?
The use of `EXTERN_BOOST_INCLUDE_PREFIX` and `EXTERN_BOOST_LIB_PREFIX` should already be including your personal boost installation

> Instead, if don't use CXXFLAGS="-g -Wall -pedantic -O3 -MMD -MP -fPIC -std=c++1z" everything seems to be compiled properly
Was boost also compiled with the same `-std=c++1z` flag?

dinardo, 2022-05-12:
I see for example that one of the make command lines is:
```
g++ -shared -Wl,-soname,libcactus_uhal_tests.so.2.8 -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/testbeam/ipbus-software/uhal/tests/obj/DummyClient.o /home/testbeam/ipbus-software/uhal/tests/obj/DummyDerivedNode.o /home/testbeam/ipbus-software/uhal/tests/obj/DummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/PCIeDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/PerfTester_static.o /home/testbeam/ipbus-software/uhal/tests/obj/TCPDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/UDPDummyHardware.o /home/testbeam/ipbus-software/uhal/tests/obj/fixtures.o /home/testbeam/ipbus-software/uhal/tests/obj/test_block.o /home/testbeam/ipbus-software/uhal/tests/obj/test_check_permissions.o /home/testbeam/ipbus-software/uhal/tests/obj/test_client_factory.o /home/testbeam/ipbus-software/uhal/tests/obj/test_config_space.o /home/testbeam/ipbus-software/uhal/tests/obj/test_empty_dispatch.o /home/testbeam/ipbus-software/uhal/tests/obj/test_hierarchy.o /home/testbeam/ipbus-software/uhal/tests/obj/test_masking.o /home/testbeam/ipbus-software/uhal/tests/obj/test_multithreaded.o /home/testbeam/ipbus-software/uhal/tests/obj/test_nodes.o /home/testbeam/ipbus-software/uhal/tests/obj/test_nonreachable.o /home/testbeam/ipbus-software/uhal/tests/obj/test_rawclient.o /home/testbeam/ipbus-software/uhal/tests/obj/test_single.o /home/testbeam/ipbus-software/uhal/tests/obj/test_soak.o /home/testbeam/ipbus-software/uhal/tests/obj/test_timeout.o /home/testbeam/ipbus-software/uhal/tests/obj/test_uri.o /home/testbeam/ipbus-software/uhal/tests/obj/tools.o -L/home/testbeam/boost_v1_77_0/stage/lib -L/home/testbeam/ipbus-software/uhal/log/lib -L/home/testbeam/ipbus-software/uhal/grammars/lib -L/home/testbeam/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -o lib/libcactus_uhal_tests.so.2.8.1
```
Is for example `-lboost_system` taken from my local BOOST or from system BOOST?

Yes indesd, I compiled my local BOOST with: `-std=c++1z`, this is the sequence of commands:
```
./bootstrap.sh
./b2 cxxflags=-std=c++1z --prefix=/home/testbeam/boost_v1_77_0/
```
Did you try yourself to compile `v2.8.1` with `c++17` ?

Many thanks for your support

tswilliams, 2022-05-12:
Hi,

To understand the situation: Which OS & g++ version are you using?

Cheers,
Tom