ipbus/ipbus-software #272: software installation problem

YangToTyz, 2022-11-21:
I follow Compiling and installing from source in Ubuntu 22.04.1，when Checkout from git and compile ，i use sudo make PYTHON=python3.6 ,have error.
Is it my version? I don't know if you have encountered similar problems? Looking forward to your help.
`yzy@yzy-virtual-machine:~/ipbus-software$ sudo make PYTHON=python3.6
OS Detected: unknown.os
PACKAGES=uhal controlhub
make  -C uhal 
make[1]: Entering directory '/home/yzy/ipbus-software/uhal'
make  -C log 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/log'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/log'
make  -C grammars 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/grammars'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/grammars'
make  -C uhal 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/uhal'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/uhal'
make  -C tests 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/tests'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/tests'
make  -C python 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/python'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/python'
make  -C gui 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/gui'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'default'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/gui'
make  -C tools 
make[2]: Entering directory '/home/yzy/ipbus-software/uhal/tools'
OS Detected: unknown.os
Using BUILD_HOME=/home/yzy/ipbus-software
make[2]: Nothing to be done for 'clean'.
make[2]: Leaving directory '/home/yzy/ipbus-software/uhal/tools'
make[1]: Leaving directory '/home/yzy/ipbus-software/uhal'
make  -C controlhub 
make[1]: Entering directory '/home/yzy/ipbus-software/controlhub'
OS Detected: unknown.os
escript /home/yzy/ipbus-software/controlhub/rebar get-deps compile
==> goldrush (get-deps)
==> lager (get-deps)
==> syslog (get-deps)
==> lager_syslog (get-deps)
==> rel (get-deps)
==> controlhub (get-deps)
==> goldrush (compile)
==> lager (compile)
==> syslog (compile)
==> lager_syslog (compile)
==> rel (compile)
==> controlhub (compile)
src/ch_utils.erl:none: error in parse transform 'lager_transform':
exception error: bad argument
  in function  integer_to_list/1
     called as integer_to_list({65,14})
     *** argument 1: not an integer
  in call from lager_transform:make_varname/2 (src/lager_transform.erl, line 241)
  in call from lager_transform:do_transform/5 (src/lager_transform.erl, line 180)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:transform_statement/2 (src/lager_transform.erl, line 107)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
Compiling src/ch_utils.erl failed:
ERROR: compile failed while processing /home/yzy/ipbus-software/controlhub: rebar_abort
make[1]: *** [Makefile:65: _all] Error 1
make[1]: Leaving directory '/home/yzy/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2`

dejardin, 2023-02-13:
Dear all
I got the same problem with Fedora 36 (gcc12.2.1) after a system upgrade.
Do you have any idea on what to do with issue ? 
I can run my test bench any more...

Best regards
Marc

dacieri, 2023-04-26:
I have the same problem when trying to compile with ubuntu 22.04.2

alindote, 2024-10-31:
Has anybody managed to solve this? I have the same problem...
Ubuntu 22.04, gcc 11.4.0, python3.10, Erlang/OTP 24 [erts-12.2.1]