ipbus/ipbus-software #273: Error during compilation

ikeshel, 2022-12-02:
Dear All,

I'm trying to compile ipbus-software from git 2.8.4 on Kubuntu 22.04. I have experience in compiling 2.8.1 on 20.04 which I'm currently using.
Here is the error:

Compiled src/lager_default_formatter.erl
Compiled src/lager_msg.erl
src/lager.erl:55:10: Warning: variable '_App' is bound multiple times in this pattern. If you mean to ignore this value, use '_' or a different underscore-prefixed name
%   55| start_ok(_App, {error, {already_started, _App}}) -> ok;
%     |          ^

src/lager.erl:55:42: Warning: variable '_App' is bound multiple times in this pattern. If you mean to ignore this value, use '_' or a different underscore-prefixed name
%   55| start_ok(_App, {error, {already_started, _App}}) -> ok;
%     |                                          ^

/home/irakli/ipbus-software/controlhub/deps/lager/src/lager.erl:55:10: Warning: variable '_App' is bound multiple times in this pattern. If you mean to ignore this value, use '_' or a different underscore-prefixed name
/home/irakli/ipbus-software/controlhub/deps/lager/src/lager.erl:55:42: Warning: variable '_App' is bound multiple times in this pattern. If you mean to ignore this value, use '_' or a different underscore-prefixed name
Compiled src/lager.erl
Compiled src/lager_trunc_io.erl
Compiled src/lager_handler_watcher.erl
Compiled src/error_logger_lager_h.erl
==> syslog (compile)
Compiled src/syslog_app.erl
Compiled src/syslog_sup.erl
Compiled src/syslog.erl
Compiling c_src/syslog_drv.c
==> lager_syslog (compile)
Compiled src/lager_syslog_backend.erl
==> rel (compile)
==> controlhub (compile)
Compiled src/ch_tcp_listener.erl
src/ch_utils.erl:none: error in parse transform 'lager_transform':
exception error: bad argument
  in function  integer_to_list/1
     called as integer_to_list({65,14})
     *** argument 1: not an integer
  in call from lager_transform:make_varname/2 (src/lager_transform.erl, line 241)
  in call from lager_transform:do_transform/5 (src/lager_transform.erl, line 180)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:transform_statement/2 (src/lager_transform.erl, line 107)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
Compiling src/ch_utils.erl failed:
ERROR: compile failed while processing /home/irakli/ipbus-software/controlhub: rebar_abort
make[1]: *** [Makefile:65: _all] Error 1
make[1]: Leaving directory '/home/irakli/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2

real    1m0,390s
user    4m38,630s
sys     0m13,650s
irakli@dtlpc019:~/ipbus-software$ 

Please instruct me on how to proceed.
Best regards...

tswilliams, 2022-12-07:
Hi,

I'm wondering whether this could be this OS has a more recent version of Erlang than the version that we've tested the ControlHub against in the past. Can you check which version of Erlang is installed on that machine?

BTW, if you don't need to use the ControlHub on this machine, you can also simply just compile uHAL by adding `Set=uhal` to `make` commands (i.e. `make Set=uhal` and `make Set=uhal rpm`)

Cheers,
Tom

virtmax, 2023-01-17:
Hi Irakli,

you could try to update the erlang-lager to a newer version in the _rebar.config_ file.

```
%%%%%%%%%%%%%%%%
% DEPENDENCIES

{deps, [
  %{lager, "3\.2\.2", {git, "https://github.com/basho/lager.git", {tag, "3.2.4"}}},
  {lager, "3\.9\.2", {git, "https://github.com/erlang-lager/lager.git", {tag, "3.9.2"}}},
  {syslog, "1\.0\.5", {git, "https://github.com/Vagabond/erlang-syslog.git", {tag, "1.0.5"}}},
  {lager_syslog, "2\.1\.2", {git, "https://github.com/basho/lager_syslog.git", {tag, "3.0.3"}}}
]}.
```

And then recompile. 

Cheers,
Maxim

UweBonnes, 2023-01-20:
Using 3.9.2 from erlang-lager results in a different error:

cd rel && escript /home/bon/ipbus-software/controlhub/rebar generate
==> rel (generate)
ERROR: generate failed while processing /home/bon/ipbus-software/controlhub/rel: {'EXIT',{{badmatch,{error,"Release \"controlhub\" uses non existing application observer"}},
         [{rebar_reltool,generate,2,
                         [{file,"src/rebar_reltool.erl"},{line,53}]},
          {rebar_core,run_modules,4,[{file,"src/rebar_core.erl"},{line,493}]},
          {rebar_core,execute,6,[{file,"src/rebar_core.erl"},{line,418}]},
          {rebar_core,maybe_execute,8,
                      [{file,"src/rebar_core.erl"},{line,302}]},
          {rebar_core,process_dir1,7,[{file,"src/rebar_core.erl"},{line,261}]},
          {rebar_core,process_commands,2,
                      [{file,"src/rebar_core.erl"},{line,93}]},
          {rebar,main,1,[{file,"src/rebar.erl"},{line,58}]},
          {escript,run,2,[{file,"escript.erl"},{line,750}]}]}}
make: *** [Makefile:66: _all] Error 1

It seem that cd rel && escript /home/bon/ipbus-software/controlhub/rebar generate
==> rel (generate)
ERROR: generate failed while processing /home/bon/ipbus-software/controlhub/rel: {'EXIT',{{badmatch,{error,"Release \"controlhub\" uses non existing application observer"}},
         [{rebar_reltool,generate,2,
                         [{file,"src/rebar_reltool.erl"},{line,53}]},
          {rebar_core,run_modules,4,[{file,"src/rebar_core.erl"},{line,493}]},
          {rebar_core,execute,6,[{file,"src/rebar_core.erl"},{line,418}]},
          {rebar_core,maybe_execute,8,
                      [{file,"src/rebar_core.erl"},{line,302}]},
          {rebar_core,process_dir1,7,[{file,"src/rebar_core.erl"},{line,261}]},
          {rebar_core,process_commands,2,
                      [{file,"src/rebar_core.erl"},{line,93}]},
          {rebar,main,1,[{file,"src/rebar.erl"},{line,58}]},
          {escript,run,2,[{file,"escript.erl"},{line,750}]}]}}
make: *** [Makefile:66: _all] Error 1

There is no 3.9.2 from basho.

It seems that Erlang/OTP 24 causes the error. I can compile on OpenSuse where Erlang/OTP 22 is used. But the resulting file does not start:
scripts> ./controlhub_start 
Using config file: xxx/controlhub/sys.config
Starting ControlHub ................................FAILED!

tswilliams, 2023-01-22:
Hi,

One of the OSs that the software is built and tested on in the CI is Fedora 33, which has Erlang 23, and so it should compile fine on Erlang 22 as well. If you are running the ControlHub from the source directories without installing it, you could try installing it using `make install`, and then running it using the scripts under `/opt/cactus`.

Otherwise, if that doesn't work, I'd suggest running the ControlHub on Fedora/RedHat/CentOS/Alma/Rocky, as the CI tests the software on a wide range of Fedora/RedHat/RedHat-derivative operating systems, and so we can most readily provide support for those operating systems.

virtmax, 2023-01-23:
@UweBonnes 
I'm using Linux Mint 21.1 that is based on Ubuntu 22.04. For me, updating the erlang-lager version in rebar.config to 3.9.2 fixed all compiler errors. ControlHub works fine too.
Erlang version: _Erlang/OTP 24 [erts-12.2.1]_

ikeshel, 2023-02-01:
> Hi Irakli,
>
> you could try to update the erlang-lager to a newer version in the _rebar.config_ file.
>
> ```
> %%%%%%%%%%%%%%%%
> % DEPENDENCIES
>
> {deps, [
>   %{lager, "3\.2\.2", {git, [...]

Dear Maxim,
I just tried this, but it doesn't help. My lager version and is 3.9.2 but it still gives me an error :(

This time I'm trying ipbus 2.8.5 , on kubuntu 22.04 LTS 

<<<<<<
g++ -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/irakli/ipbus-software/uhal/tests/obj/PerfTester.o -L/home/irakli/ipbus-software/uhal/log/lib -L/home/irakli/ipbus-software/uhal/grammars/lib -L/home/irakli/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_tests -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -lstdc++ -o /home/irakli/ipbus-software/uhal/tests/bin/PerfTester.exe
g++ -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/irakli/ipbus-software/uhal/tests/obj/run_uhal_tests.o -L/home/irakli/ipbus-software/uhal/log/lib -L/home/irakli/ipbus-software/uhal/grammars/lib -L/home/irakli/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_tests -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -lstdc++ -o /home/irakli/ipbus-software/uhal/tests/bin/run_uhal_tests.exe
g++ -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt /home/irakli/ipbus-software/uhal/tests/obj/test_log.o -L/home/irakli/ipbus-software/uhal/log/lib -L/home/irakli/ipbus-software/uhal/grammars/lib -L/home/irakli/ipbus-software/uhal/uhal/lib -Llib -lcactus_uhal_tests -lcactus_uhal_uhal -lcactus_uhal_grammars -lcactus_uhal_log -lboost_filesystem -lboost_program_options -lboost_unit_test_framework -lboost_regex -lboost_system -lpugixml -lrt -lpthread -lstdc++ -o /home/irakli/ipbus-software/uhal/tests/bin/test_log.exe
make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/tests'
make  -C python 
make[2]: Entering directory '/home/irakli/ipbus-software/uhal/python'
cat: /etc/system-release: No such file or directory
OS Detected: unknown.os
Using BUILD_HOME=/home/irakli/ipbus-software
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
mkdir -p /home/irakli/ipbus-software/uhal/python/obj/
mkdir -p /home/irakli/ipbus-software/uhal/python/lib
make[2]: python-config: No such file or directory
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -ftemplate-depth-128 -O0 -rdynamic -finline-functions -Wno-inline -DNDEBUG  -DBOOST_TEST_DYN_LINK -Iinclude -Ipybind11/include -I/usr/include/python3.10 -I/home/irakli/ipbus-software/uhal/log/include -I/home/irakli/ipbus-software/uhal/grammars/include -I/home/irakli/ipbus-software/uhal/uhal/include -I/home/irakli/ipbus-software/uhal/tests/include -I/home/irakli/ipbus-software/uhal/uhal/include/_static /home/irakli/ipbus-software/uhal/python/src/common/cactus_pycohal.cpp -o /home/irakli/ipbus-software/uhal/python/obj/cactus_pycohal.o
make[2]: python-config: No such file or directory
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -ftemplate-depth-128 -O0 -rdynamic -finline-functions -Wno-inline -DNDEBUG  -DBOOST_TEST_DYN_LINK -Iinclude -Ipybind11/include -I/usr/include/python3.10 -I/home/irakli/ipbus-software/uhal/log/include -I/home/irakli/ipbus-software/uhal/grammars/include -I/home/irakli/ipbus-software/uhal/uhal/include -I/home/irakli/ipbus-software/uhal/tests/include -I/home/irakli/ipbus-software/uhal/uhal/include/_static /home/irakli/ipbus-software/uhal/python/src/common/enums_logging.cpp -o /home/irakli/ipbus-software/uhal/python/obj/enums_logging.o
make[2]: python-config: No such file or directory
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -ftemplate-depth-128 -O0 -rdynamic -finline-functions -Wno-inline -DNDEBUG  -DBOOST_TEST_DYN_LINK -Iinclude -Ipybind11/include -I/usr/include/python3.10 -I/home/irakli/ipbus-software/uhal/log/include -I/home/irakli/ipbus-software/uhal/grammars/include -I/home/irakli/ipbus-software/uhal/uhal/include -I/home/irakli/ipbus-software/uhal/tests/include -I/home/irakli/ipbus-software/uhal/uhal/include/_static /home/irakli/ipbus-software/uhal/python/src/common/exceptions.cpp -o /home/irakli/ipbus-software/uhal/python/obj/exceptions.o
make[2]: python-config: No such file or directory
<string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
<string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
g++ -shared  -Wall -g -O3 -fPIC -Wl,--no-as-needed -lrt  -Wl,-h -Wl,-Bstatic -Wl,-Bdynamic /home/irakli/ipbus-software/uhal/python/obj/cactus_pycohal.o /home/irakli/ipbus-software/uhal/python/obj/enums_logging.o /home/irakli/ipbus-software/uhal/python/obj/exceptions.o -L/usr/lib -L/home/irakli/ipbus-software/uhal/log/lib -L/home/irakli/ipbus-software/uhal/grammars/lib -L/home/irakli/ipbus-software/uhal/uhal/lib -L/home/irakli/ipbus [... cut ...]
Cloning into 'lager'...
error: pathspec '3.9.2' did not match any file(s) known to git
ERROR: sh(git checkout -q 3.9.2)
failed with return code 1 and the following output:
error: pathspec '3.9.2' did not match any file(s) known to git

ERROR: 'get-deps' failed while processing /home/irakli/ipbus-software/controlhub: rebar_abort
make[1]: *** [Makefile:65: _all] Error 1
make[1]: Leaving directory '/home/irakli/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2

UweBonnes, 2023-02-01:
You look in basho but you should use erlang-lager/lager.git.

B.t.w, in the meantime I got controlhub compiling and running on OpenSuse Leap 15.4. Install all needed packages and compile with the environment variable PYTHON set as python3. Install as root.

dejardin, 2023-02-13:
Dear all,
I got the same problem during compilation on FC36 (gcc 12.2.1, erlang 24.3.4.8)
For the moment, I don't use ControlHub, so I turned around with "make Set=uhal".
But it would be good to have a full working version for recent distros.

Best regards
Marc

ikeshel, 2023-02-21:
Dear All,
In order to eliminate the error with an unknown OS, I have tried to do the following:
`sudo ln -s /etc/lsb-release /etc/system-release`
Which is in my case:
DISTRIB_ID=Ubuntu
DISTRIB_RELEASE=22.04
DISTRIB_CODENAME=jammy
DISTRIB_DESCRIPTION="Ubuntu 22.04.2 LTS"

and it still has a problem with erlang

> OS Detected: unknown.os
> PACKAGES=uhal controlhub
> make  -C uhal 
> make[1]: Entering directory '/home/irakli/ipbus-software/uhal'
> Using BUILD_HOME=/home/irakli
> make  -C log 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/log'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/log'
> make  -C grammars 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/grammars'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/grammars'
> make  -C uhal 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/uhal'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/uhal'
> make  -C tests 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/tests'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/tests'
> make  -C python 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/python'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
> <string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
> <string>:1: DeprecationWarning: The distutils package is deprecated and slated for removal in Python 3.12. Use setuptools or check PEP 632 for potential alternatives
> <string>:1: DeprecationWarning: The distutils.sysconfig module is deprecated, use sysconfig instead
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/python'
> make  -C gui 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/gui'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'default'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/gui'
> make  -C tools 
> make[2]: Entering directory '/home/irakli/ipbus-software/uhal/tools'
> OS Detected: unknown.os
> Using BUILD_HOME=/home/irakli/ipbus-software
> make[2]: Nothing to be done for 'clean'.
> make[2]: Leaving directory '/home/irakli/ipbus-software/uhal/tools'
> make[1]: Leaving directory '/home/irakli/ipbus-software/uhal'
> make  -C controlhub 
> make[1]: Entering directory '/home/irakli/ipbus-software/controlhub'
> OS Detected: unknown.os
> escript /home/irakli/ipbus-software/controlhub/rebar get-deps compile
> ==> goldrush (get-deps)
> ==> lager (get-deps)
> ==> syslog (get-deps)
> ==> lager_syslog (get-deps)
> ==> rel (get-deps)
> ==> controlhub (get-deps)
> ==> goldrush (compile)
> ==> lager (compile)
> ==> syslog (compile)
> ==> lager_syslog (compile)
> ==> rel (compile)
> ==> controlhub (compile)
> src/ch_utils.erl:none: error in parse transform 'lager_transform':
> exception error: bad argument
>   in function  integer_to_list/1
>      called as integer_to_list({65,14})
>      *** argument 1: not an integer
>   in call from lager_transform:make_varname/2 (src/lager_transform.erl, line 241)
>   in call from lager_transform:do_transform/5 (src/lager_transform.erl, line 180)
>   in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
>   in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
>   in call from lager_transform:transform_statement/2 (src/lager_transform.erl, line 107)
>   in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
>   in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
> Compiling src/ch_utils.erl failed:
> ERROR: compile failed while processing /home/irakli/ipbus-software/controlhub: rebar_abort
> make[1]: *** [Makefile:65: _all] Error 1
> make[1]: Leaving directory '/home/irakli/ipbus-software/controlhub'
> make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2

How can I/we contribute to fixing this problem? I'm still using Kubuntu 20.04 LTE to have running ipbus software.

Best regards,
Irakli

ikeshel, 2024-02-12:
@dhymers 
Dear Maxim, 
I did the following on mu Kubuntu 22.04:

1. git clone --depth=1 -b v2.8.12 --recurse-submodules https://github.com/ipbus/ipbus-software.git
2. cd  ipbus-software/
3. vim controlhub/rebar.config

%%%%%%%%%%%%%%%%
% DEPENDENCIES

{deps, [
  %{lager, "3\.2\.2", {git, "https://github.com/basho/lager.git", {tag, "3.2.4"}}},
  {lager, "3\.9\.2", {git, "https://github.com/erlang-lager/lager.git", {tag, "3.9.2"}}},
  {syslog, "1\.0\.5", {git, "https://github.com/Vagabond/erlang-syslog.git", {tag, "1.0.5"}}},
  {lager_syslog, "2\.1\.2", {git, "https://github.com/basho/lager_syslog.git", {tag, "3.0.3"}}}
]}.

exactly as you suggest. My lager is indeed 3.9.2

Compiled src/lager_syslog_backend.erl
==> rel (compile)
==> controlhub (compile)
Compiled src/ch_tcp_listener.erl
Compiled src/ch_device_client_registry.erl
Compiled src/controlhub_app.erl
src/ch_utils.erl:22:2: Warning: type log_level() is unused
src/ch_utils.erl:120:1: Warning: function timestamp_string/1 is unused
src/ch_utils.erl:219:1: Warning: function mini_state_string/1 is unused
Compiled src/ch_utils.erl
Compiled src/ch_stats.erl
Compiled src/ch_config.erl
Compiled src/ch_sup.erl
src/ch_transaction_manager.erl:260:12: Warning: variable 'S' is unused
Compiled src/ch_transaction_manager.erl
src/ch_device_client.erl:36:2: Warning: type timestamp() is unused
src/ch_device_client.erl:38:2: Warning: type in_flight_info() is unused
src/ch_device_client.erl:372:11: Warning: variable 'RcvdHdr' is unused
src/ch_device_client.erl:538:1: Warning: function reset_header/2 is unused
Compiled src/ch_device_client.erl
cd rel && escript /home/irakli/ipbus-software/controlhub/rebar generate
==> rel (generate)
excl_archive_filters is no longer supported in reltoolERROR: generate failed while processing /home/irakli/ipbus-software/controlhub/rel: {'EXIT',{{badmatch,{error,"Illegal option: [{excl_sys_filters,[\"^bin/(?!start_clean.boot)\",\n                                    \"^erts.*/bin/(dialyzer|typer)\",\n                                    \"^erts.*/(doc|info|include|lib|man|src)\"]},\n                 {excl_app_filters,[\".gitignore\"]},\n                 {app,crypto,[{incl_cond,exclude}]},\n                 {app,hipe,[{incl_cond,exclude}]},\n                 {app,syslog,\n                      [{mod_cond,app},\n                       {incl_cond,include},\n                       {lib_dir,\"../deps/syslog\"}]},\n                 {app,lager_syslog,\n                      [{mod_cond,app},\n                       {incl_cond,include},\n                       {lib_dir,\"../deps/lager_syslog\"}]},\n                 {app,goldrush,\n                      [{mod_cond,app},\n                       {incl_cond,include},\n                       {lib_dir,\"../deps/goldrush\"}]},\n                 {app,lager,\n                      [{mod_cond,app},\n                       {incl_cond,include},\n                       {lib_dir,\"../deps/lager\"}]},\n                 {app,controlhub,\n                      [{mod_cond,app},{incl_cond,include},{lib_dir,\"..\"}]}]"}},
         [{rebar_reltool,generate,2,
                         [{file,"src/rebar_reltool.erl"},{line,53}]},
          {rebar_core,run_modules,4,[{file,"src/rebar_core.erl"},{line,493}]},
          {rebar_core,execute,6,[{file,"src/rebar_core.erl"},{line,418}]},
          {rebar_core,maybe_execute,8,
                      [{file,"src/rebar_core.erl"},{line,302}]},
          {rebar_core,process_dir1,7,[{file,"src/rebar_core.erl"},{line,261}]},
          {rebar_core,process_commands,2,
                      [{file,"src/rebar_core.erl"},{line,93}]},
          {rebar,main,1,[{file,"src/rebar.erl"},{line,58}]},
          {escript,run,2,[{file,"escript.erl"},{line,750}]}]}}
make[1]: *** [Makefile:66: _all] Error 1
make[1]: Leaving directory '/home/irakli/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2

:(( unfortunately it didn't help. 
Which version of ipbus are you using?
Regards, Irakli

dhymers, 2024-02-13:
@ikeshel 
I am using v2.8.12 also. I have just tested again to re-clone and recompile in a new directory on the same PC, and I do not experience this issue, although I do get a lot of the same warnings.

```
Compiled src/lager_syslog_backend.erl
==> rel (compile)
==> controlhub (compile)
Compiled src/ch_stats.erl
src/ch_transaction_manager.erl:260:12: Warning: variable 'S' is unused
Compiled src/ch_transaction_manager.erl
Compiled src/ch_sup.erl
Compiled src/ch_config.erl
Compiled src/controlhub_app.erl
Compiled src/ch_device_client_registry.erl
Compiled src/ch_tcp_listener.erl
src/ch_utils.erl:22:2: Warning: type log_level() is unused
src/ch_utils.erl:120:1: Warning: function timestamp_string/1 is unused
src/ch_utils.erl:219:1: Warning: function mini_state_string/1 is unused
Compiled src/ch_utils.erl
src/ch_device_client.erl:38:2: Warning: type in_flight_info() is unused
src/ch_device_client.erl:372:11: Warning: variable 'RcvdHdr' is unused
src/ch_device_client.erl:538:1: Warning: function reset_header/2 is unused
Compiled src/ch_device_client.erl
cd rel && escript /home/daq/Documents/ipbus-software/controlhub/rebar generate
==> rel (generate)
make[1]: Leaving directory '/home/daq/Documents/ipbus-software/controlhub'
```