ipbus/ipbus-software #288: "Undefined reference to  uhal::" error occurred when buiding the program

DeathAndLife, 2023-03-27:
I hava biilt ipbus successfully on Debian9, but when I try to build my porgram with gcc and g++ version 10.2.1, I get an error. it compiles without any issues on CentOS
How can I resolve this issue?

[image]

[image]
[image]

DeathAndLife, 2023-03-27:
Sorry, GitHub network exception submitted twice

tswilliams, 2023-03-27:
Which version of gcc did you use for building uHAL itself?

DeathAndLife, 2023-03-27:
I just re built using Set=uhal alone and succeeded

> Which version of gcc did you use for building uHAL itself?

DeathAndLife, 2023-03-27:
> Which version of gcc did you use for building uHAL itself?

 it's still not possible
gcc 10.2.1

DeathAndLife, 2023-03-29:
I set version of GCC&G++ to 4.8 and recompiled Uhal,  but that did not solve the issue

tswilliams, 2023-03-29:
Can you send me the compilation & linker commands that your build tool is running (depending on the current config for your build tool, you may need to increase its verbosity for this).

DeathAndLife, 2023-03-30:
Thank you for your replay,but I still have not resovled this issue. if you have found any ways to solve it, Please let me konw as soon as possible.

DeathAndLife, 2023-04-05:
Hi, I have solved this problem
My program framework provides some parameters for Makeffile to introduce external libraries
I saw from the actual printed command that it only put uhal at the end
And then the problem was solved
[image]

closed 2023-04-05 by DeathAndLife