ipbus/ipbus-software #290: 1 packe<<TRUNCATED>>

DeathAndLife, 2023-04-20:
Timeout (1000ms)occurred for receive (header)from ControlHub with URI 'xxx'. 1 packe<<TRUNCATED>>
Hello,may I ask what is the cause of this issue and how can it be resolved?
The situation in which this issue occured was when the number of connections reached 460 and the program had been running for about half an hour. After restarting ControlHub, the issue was resolved , but the program did not run for such a long time afterwards.

tswilliams, 2023-04-20:
The only circumstance I might expect this to occur in is if the ControlHub is overloaded - or if the PC that it's running on is overloaded. How many uHAL clients do you have communicating with the ControlHub, how many boards are you controlling, and how large are the reads/writes that you're performing? Also, which version of the software are you using, and on what OS?

DeathAndLife, 2023-04-21:
> 
232 boards => 462 uHal clients (Two hosts use the same ControlHub)
90% CPU usage
reads/writes: average 7500±/s  highest 9000±/s
Version: 2.8.2
OS: CentOs 7
[image]

tswilliams, 2023-04-21:
For that many boards, I'd suggest splitting the ControlHub load over a few machines - the largest system number of boards that I've routinely controlled via a single ControlHub instance is about 60 boards.

DeathAndLife, 2023-04-21:
Thank your reply
Our system ultimately need to run 10,000+ boards
So, if there is any way to reduce the usage CPU of ControlHub, it will be of great help to us
Thanks again

DeathAndLife, 2023-04-21:
> For that many boards, I'd suggest splitting the ControlHub load over a few machines - the largest system number of boards that I've routinely controlled via a single ControlHub instance is about 60 boards.

So is 60 boards related to the number of connections?
Two hosts using the same ControlHub to connect to the same batch of Boards
for example: 60b
[image]
oards with 120 active connections