ipbus/ipbus-software #310: error at compiling newest version on Kubuntu 22.04

ikeshel, 2023-12-08:
make  -C controlhub 
make[1]: Entering directory '/home/irakli/ipbus-software/controlhub'
OS Detected: unknown.os
escript /home/irakli/ipbus-software/controlhub/rebar get-deps compile
==> goldrush (get-deps)
==> lager (get-deps)
==> syslog (get-deps)
==> lager_syslog (get-deps)
==> rel (get-deps)
==> controlhub (get-deps)
==> goldrush (compile)
==> lager (compile)
==> syslog (compile)
==> lager_syslog (compile)
==> rel (compile)
==> controlhub (compile)
src/ch_utils.erl:none: error in parse transform 'lager_transform':
exception error: bad argument
  in function  integer_to_list/1
     called as integer_to_list({65,14})
     *** argument 1: not an integer
  in call from lager_transform:make_varname/2 (src/lager_transform.erl, line 241)
  in call from lager_transform:do_transform/5 (src/lager_transform.erl, line 180)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:transform_statement/2 (src/lager_transform.erl, line 107)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
  in call from lager_transform:'-transform_statement/2-lc$^0/1-0-'/2 (src/lager_transform.erl, line 109)
Compiling src/ch_utils.erl failed:
ERROR: compile failed while processing /home/irakli/ipbus-software/controlhub: rebar_abort
make[1]: *** [Makefile:65: _all] Error 1
make[1]: Leaving directory '/home/irakli/ipbus-software/controlhub'
make: *** [Makefile:58: controlhub/.virtual.Makefile] Error 2
irakli@dtlpc019:~/ipbus-software$ 

Cold you please advice how to proceed?

prisundind, 2024-03-08:
Is there a workaround for this error?

ikeshel, 2024-03-10:
> Is there a workaround for this error?

I don't think so, unfortunately. I tried with latest Debian too and also there the same error...

ikeshel, 2024-03-10:
src/ch_utils.erl:none: error in parse transform 'lager_transform':
exception error: bad argument
  in function  integer_to_list/1
     called as integer_to_list({65,14})
     *** argument 1: not an integer
  in call from lager_transform:make_varname/2 (src/lager_transform.erl, line 241)
  in call from lager_transform:do_transform/5 (src/lager_transform.erl, line 180)

ChetanMandloi, 2024-05-31:
Any possible workaround or update for this issue?

cbespin, 2024-07-02:
I tried fixing this for Debian 12 and the following seems to work. Update the erlang dependencies in `controlhub/rebar.config`
```
{deps, [
  {lager, "3\.9\.2", {git, "https://github.com/erlang-lager/lager.git", {tag, "3.9.2"}}},
  {syslog, "1\.0\.5", {git, "https://github.com/Vagabond/erlang-syslog.git", {tag, "1.0.5"}}},
  {lager_syslog, "2\.1\.2", {git, "https://github.com/erlang-lager/lager_syslog.git", {tag, "3.0.3"}}}
]}.
```
Apart from a later problem with the python bindings, this works for me on multiple devices. I have no experience with erlang, this was just a brute-force fix.

ikeshel, 2025-01-21:
Dear All,

After a few hours of heavy attampts, I finally found the way to install controlhub on Kubuntu 24.04 LTE.

Thanks to @cbespin 
one should definitely fix the:

> controlhub/rebar.config
`
{deps, [
  {lager, "3\.9\.2", {git, "https://github.com/erlang-lager/lager.git", {tag, "3.9.2"}}},
  {syslog, "1\.0\.5", {git, "https://github.com/Vagabond/erlang-syslog.git", {tag, "1.0.5"}}},
  {lager_syslog, "2\.1\.2", {git, "https://github.com/erlang-lager/lager_syslog.git", {tag, "3.0.3"}}}
]}.
`

Then I was forced to do bit nasty things like:

`
ll /usr/bin/x86_64-linux-gnu-cpp-13 
file /usr/bin/x86_64-linux-gnu-cpp-13
locate x86_64-linux-gnu-cpp-13.1.gz 
sudo ln -sf /usr/share/man/man1/x86_64-linux-gnu-cpp-13.1.gz cpp.1.gz 

ll gfortran.1.gz
locate gfortran.1.gz
sudo ln -s /usr/share/man/man1/x86_64-linux-gnu-gfortran.1.gz gfortran.1.gz
locate gfortran.1.gz
ll gfortran.1.gz 

ll gfortran-13.1
locate gfortran-13.1
sudo rm gfortran-13.1
sudo ln -sf /usr/bin/gfortran-13 gfortran-13.1
locate  gfortran-13.1 
ll /usr/share/man/man1/gfortran-13.1 
`