ipbus/ipbus-software #317: Uninstall of cactuscore-controlhub fails.

DavidCussans, 2024-08-14:
I recently installed cactus on a freshly build Alma-9 PC

I had forgotten to install the EPEL Repo , so the install failed.

Hence ( or perhaps otherwise ) systemctl won't control start/stop/status correctly:
```
[phdgc@it063642 ~]$ sudo systemctl status controlhub
Unit controlhub.service could not be found.
```

I tried to uninstall controlhub , but this also failed:

```
[phdgc@it063642 ~]$ sudo dnf erase cactuscore-controlhub
Dependencies resolved.
===========================================================================================================================================
 Package                                Architecture            Version                           Repository                          Size
===========================================================================================================================================
Removing:
 cactuscore-controlhub                  x86_64                  2.8.14-1.el9                      @ipbus-sw-updates                   35 M
Removing unused dependencies:
 SDL2                                   x86_64                  2.26.0-1.el9                      @appstream                         1.8 M
 erlang-compiler                        x86_64                  26.2.5-1.el9                      @epel                              5.2 M
 erlang-crypto                          x86_64                  26.2.5-1.el9                      @epel                              488 k
 erlang-erts                            x86_64                  26.2.5-1.el9                      @epel                              8.2 M
 erlang-kernel                          x86_64                  26.2.5-1.el9                      @epel                              4.8 M
 erlang-stdlib                          x86_64                  26.2.5-1.el9                      @epel                               10 M
 libdecor                               x86_64                  0.1.1-1.el9                       @appstream                          96 k
 lksctp-tools                           x86_64                  1.0.19-3.el9_4                    @baseos                            269 k
 mesa-libGLU                            x86_64                  9.0.1-6.el9                       @appstream                         345 k
 wxBase3                                x86_64                  3.0.5.1-6.el9                     @epel                              2.8 M
 wxGTK3                                 x86_64                  3.0.5.1-6.el9                     @epel                               15 M
 wxGTK3-gl                              x86_64                  3.0.5.1-6.el9                     @epel                              103 k
 wxGTK3-i18n                            noarch                  3.0.5.1-6.el9                     @epel                              2.5 M

Transaction Summary
===========================================================================================================================================
Remove  14 Packages

Freed space: 87 M
Is this ok [y/N]: y
Running transaction check
Transaction check succeeded.
Running transaction test
Transaction test succeeded.
Running transaction
  Preparing        :                                                                                                                   1/1
  Running scriptlet: cactuscore-controlhub-2.8.14-1.el9.x86_64                                                                        1/14
Stopping ControlHub ...... ok
/var/tmp/rpm-tmp.fzsLdX: line 3: /sbin/chkconfig: No such file or directory
/var/tmp/rpm-tmp.fzsLdX: line 4: /sbin/chkconfig: No such file or directory
error: %preun(cactuscore-controlhub-2.8.14-1.el9.x86_64) scriptlet failed, exit status 127

Error in PREUN scriptlet in rpm package cactuscore-controlhub
  Erasing          : wxGTK3-gl-3.0.5.1-6.el9.x86_64                                                                                   2/14
error: cactuscore-controlhub-2.8.14-1.el9.x86_64: erase failed

  Erasing          : wxGTK3-3.0.5.1-6.el9.x86_64                                                                                      3/14
  Erasing          : erlang-compiler-26.2.5-1.el9.x86_64                                                                              4/14
  Erasing          : erlang-kernel-26.2.5-1.el9.x86_64                                                                                5/14
  Erasing          : erlang-crypto-26.2.5-1.el9.x86_64                                                                                6/14
  Erasing          : erlang-stdlib-26.2.5-1.el9.x86_64                                                                                7/14
  Erasing          : erlang-erts-26.2.5-1.el9.x86_64                                                                                  8/14
  Erasing          : wxGTK3-i18n-3.0.5.1-6.el9.noarch                                                                                 9/14
  Erasing          : SDL2-2.26.0-1.el9.x86_64                                                                                        10/14
  Erasing          : libdecor-0.1.1-1.el9.x86_64                                                                                     11/14
  Erasing          : lksctp-tools-1.0.19-3.el9_4.x86_64                                                                              12/14
  Erasing          : wxBase3-3.0.5.1-6.el9.x86_64                                                                                    13/14
  Erasing          : mesa-libGLU-9.0.1-6.el9.x86_64                                                                                  14/14
  Running scriptlet: mesa-libGLU-9.0.1-6.el9.x86_64                                                                                  14/14
  Verifying        : SDL2-2.26.0-1.el9.x86_64                                                                                         1/14
  Verifying        : cactuscore-controlhub-2.8.14-1.el9.x86_64                                                                        2/14
  Verifying        : erlang-compiler-26.2.5-1.el9.x86_64                                                                              3/14
  Verifying        : erlang-crypto-26.2.5-1.el9.x86_64                                                                                4/14
  Verifying        : erlang-erts-26.2.5-1.el9.x86_64                                                                                  5/14
  Verifying        : erlang-kernel-26.2.5-1.el9.x86_64                                                                                6/14
  Verifying        : erlang-stdlib-26.2.5-1.el9.x86_64                                                                                7/14
  Verifying        : libdecor-0.1.1-1.el9.x86_64                                                                                      8/14
  Verifying        : lksctp-tools-1.0.19-3.el9_4.x86_64                                                                               9/14
  Verifying        : mesa-libGLU-9.0.1-6.el9.x86_64                                                                                  10/14
  Verifying        : wxBase3-3.0.5.1-6.el9.x86_64                                                                                    11/14
  Verifying        : wxGTK3-3.0.5.1-6.el9.x86_64                                                                                     12/14
  Verifying        : wxGTK3-gl-3.0.5.1-6.el9.x86_64                                                                                  13/14
  Verifying        : wxGTK3-i18n-3.0.5.1-6.el9.noarch                                                                                14/14

Removed:
  SDL2-2.26.0-1.el9.x86_64                    erlang-compiler-26.2.5-1.el9.x86_64            erlang-crypto-26.2.5-1.el9.x86_64
  erlang-erts-26.2.5-1.el9.x86_64             erlang-kernel-26.2.5-1.el9.x86_64              erlang-stdlib-26.2.5-1.el9.x86_64
  libdecor-0.1.1-1.el9.x86_64                 lksctp-tools-1.0.19-3.el9_4.x86_64             mesa-libGLU-9.0.1-6.el9.x86_64
  wxBase3-3.0.5.1-6.el9.x86_64                wxGTK3-3.0.5.1-6.el9.x86_64                    wxGTK3-gl-3.0.5.1-6.el9.x86_64
  wxGTK3-i18n-3.0.5.1-6.el9.noarch
Failed:
  cactuscore-controlhub-2.8.14-1.el9.x86_64

Error: Transaction failed
```

How can I proceed?

tswilliams, 2024-08-16:
Hi!

Thanks for raising this; the Alma 9 RPM scripts shouldn't be using `chkconfig`, so I'll fix that packaging bug in the coming days.

One potential solution is `dnf install chkconfig`, but unfortunately when I try that I see errors in some scenarios. The other solution is `rpm -e --noscripts cactuscore-controlhub` ; this should work even if you don't have `chkconfig` installed

tswilliams, 2025-01-06:
This bug was fixed in v2.8.15 - if a previous version of the SW is installed though, then you may still need to run `rpm -e --noscripts cactuscore-controlhub` to resolve the error (see above).

Closing since now fixed, but feel free to re-open if you see any further issues on this front.

closed 2025-01-06 by tswilliams