ipbus/ipbus-software #320: "ipbb ipbus gendecoders" command does not work

dddlcc, 2024-09-13:
I use ipbb to build the emp-firmware. command "ipbb ipbus gendecoders" shows there is no ipbus. My ipbb version is ipbb, version 0.5.2. How could I solve it? thx a lot

tswilliams, 2024-09-13:
Is uHAL installed on that machine? If it's not please follow the installation instructions at https://ipbus.web.cern.ch/doc/user/html/software/install/yum.html (Note: For FW builds you only need uHAL, not the ControlHub)

Once uHAL is installed, then you need to add `/opt/cactus/lib` to the `LD_LIBRARY_PATH` environment variable, and add `/opt/cactus/bin/uhal/tools` to `PATH` (also mentioned at https://serenity.web.cern.ch/serenity/emp-fwk/firmware/instructions.html#build-tool )

dddlcc, 2024-09-13:
Thx a lot for your reply. But I have installed the uhal several months ago. As the pictures show, the uahl is in /opt/cactus/bin/uhal/. And use ipbb -h ro check command. There is no "ipbus" command. Is there something I am wrong?
[image]
[image]
[image]