ipbus/ipbus-software #323: Build with Python 3.12 fails

rhauser, 2024-10-18:
Hi, 

we (ATLAS TDAQ) are seeing build failures of ipbus in experimental builds against Python 3.12. They seem to be mostly caused by the removal of distutils in this release (https://docs.python.org/3.12/whatsnew/3.12.html#distutils).

I worked around the build failures by adding an 'import setuptools' in various places before distutils is used (as setuptools contains an internal copy of distutils), see https://github.com/rhauser/ipbus-software/commit/7277980ec458ad8a13bdba4f69acff4b8490b0b0

The Python installation still fails, though, with the following warnings and error:

```
% make install prefix=/scratch/rhauser/ipbus/install/
OS Detected: el9
Using BUILD_HOME=/scratch/rhauser/ipbus/ipbus-software
mkdir -p /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD
cp /scratch/rhauser/ipbus/ipbus-software/uhal/config//setupTemplate.py /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD/cactuscore-uhal-python312.py
sed -i -e 's#__python_packages__#["uhal"]#' \
       -e 's#__packagename__#cactuscore-uhal-python312#' \
       -e 's#__version__#2.8.15#' \
       -e 's#__author__#Tom Williams#' \
       -e 's#__description__#Python bindings for the uHAL library#' \
       -e 's#__url__#https://ipbus.web.cern.ch/ipbus#' \
       -e 's#__project__#uhal#' \
       -e 's#__install_dir__#/opt/cactus#' \
       -e 's#__package_build_dir__#/scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD#' \
       /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD/cactuscore-uhal-python312.py
# Change directory into pkg and copy everything into rpm/pkg
cd pkg && \
  find . -name "*" -exec install -D \{\} /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD/\{\} \;
install: omitting directory '.'
install: omitting directory './uhal'
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
Regex version mismatch, expected: 10.42 2022-12-11 actual: 10.40 2022-04-14
# Add a manifest file
echo "include */*.so" > /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD/MANIFEST.in
# Change into rpm/pkg to finally run the customized setup.py
if [ -f setup.cfg ]; then cp setup.cfg /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD/ ; fi
cd /scratch/rhauser/ipbus/ipbus-software/uhal/python/rpm/RPMBUILD && \
  bindir=/scratch/rhauser/ipbus/install//bin python cactuscore-uhal-python312.py install --prefix=/scratch/rhauser/ipbus/install/ --exec-prefix=/scratch/rhauser/ipbus/install/
/cvmfs/sft-nightlies.cern.ch/lcg/views/dev3python312/Wed/x86_64-el9-gcc13-opt/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: SetuptoolsDeprecationWarning: setup.py install is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` directly.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://blog.ganssle.io/articles/2021/10/setup-py-deprecated.html for details.
        ********************************************************************************

!!
  self.initialize_options()
/cvmfs/sft-nightlies.cern.ch/lcg/views/dev3python312/Wed/x86_64-el9-gcc13-opt/lib/python3.12/site-packages/setuptools/_distutils/cmd.py:66: EasyInstallDeprecationWarning: easy_install command is deprecated.
!!

        ********************************************************************************
        Please avoid running ``setup.py`` and ``easy_install``.
        Instead, use pypa/build, pypa/installer or other
        standards-based tools.

        See https://github.com/pypa/setuptools/issues/917 for details.
        ********************************************************************************

!!
  self.initialize_options()
Traceback (most recent call last):
  File "/tmp/tmpq5713hro.py", line 1, in <module>
    from distutils.util import byte_compile
ModuleNotFoundError: No module named 'distutils'
error: command '/cvmfs/sft-nightlies.cern.ch/lcg/views/dev3python312/Wed/x86_64-el9-gcc13-opt/bin/python' failed with exit code 1
make: *** [/scratch/rhauser/ipbus/ipbus-software/uhal/config/mfPythonRPMRules.mk:97: install] Error 1
pc-tbed-pub-06:/scratch/rhauser/ipbus/ipbus-software/uhal/python % 

```

I'm testing this with a nightly build of LCG like this on an Alma 9 host:

```shell
source /cvmfs/sft-nightlies.cern.ch/lcg/views/dev3python312/Wed/x86_64-el9-gcc13-opt/setup.sh
make Set=uhal BUILD_PUGIXML=1 EXTERN_BOOST_INCLUDE_PREFIX=/cvmfs/sft-nightlies.cern.ch/lcg/nightlies/dev3python312/Wed/Boost/1.86.0/x86_64-el9-gcc13-opt/include/ EXTERN_BOOST_LIB_PREFIX=/cvmfs/sft-nightlies.cern.ch/lcg/nightlies/dev3python312/Wed/Boost/1.86.0/x86_64-el9-gcc13-opt/lib prefix=/scratch/rhauser/ipbus/install
```

thasti, 2025-01-31:
+1, I see similar issues with the release of Python 3.13.

chevillotn, 2025-02-17:
Hi,
I am using python3.12 on an ubuntu machine for ATLAS LAr phase 1/2 systems (LArC/LATOME/LATOURNETT).
I fail to build the uhal from sources. I don't get an error with `distutils` but rather on `PyFrameObject` which seems to have been removed in python v3.11:
```
Using BUILD_HOME=/home/chevillot/temp/ipbus-software-v2.8.15
g++ -c -g -Wall -pedantic -O3 -MMD -MP -fPIC -ftemplate-depth-128 -O0 -rdynamic -finline-functions -Wno-inline -DNDEBUG -I/home/chevillot/miniconda3/include/python3.12 -I/home/chevillot/miniconda3/include/python3.12 -DBOOST_TEST_DYN_LINK -Iinclude -Ipybind11/include -I/home/chevillot/miniconda3/include/python3.12 -I/home/chevillot/temp/ipbus-software-v2.8.15/uhal/log/include -I/home/chevillot/temp/ipbus-software-v2.8.15/uhal/grammars/include -I/home/chevillot/temp/ipbus-software-v2.8.15/uhal/uhal/include -I/home/chevillot/temp/ipbus-software-v2.8.15/uhal/tests/include -I/home/chevillot/temp/ipbus-software-v2.8.15/uhal/uhal/include/_static /home/chevillot/temp/ipbus-software-v2.8.15/uhal/python/src/common/cactus_pycohal.cpp -o /home/chevillot/temp/ipbus-software-v2.8.15/uhal/python/obj/cactus_pycohal.o
In file included from /usr/include/pybind11/cast.h:16,
                 from /usr/include/pybind11/attr.h:13,
                 from /usr/include/pybind11/pybind11.h:13,
                 from /usr/include/pybind11/stl.h:13,
                 from /home/chevillot/temp/ipbus-software-v2.8.15/uhal/python/src/common/cactus_pycohal.cpp:7:
/usr/include/pybind11/detail/type_caster_base.h: In function ‘std::string pybind11::detail::error_string()’:
/usr/include/pybind11/detail/type_caster_base.h:482:26: error: invalid use of incomplete type ‘PyFrameObject’ {aka ‘struct _frame’}
  482 |             frame = frame->f_back;
      |                          ^~
In file included from /home/chevillot/miniconda3/include/python3.12/Python.h:42,
                 from /usr/include/pybind11/detail/common.h:215,
                 from /usr/include/pybind11/stl.h:12,
                 from /home/chevillot/temp/ipbus-software-v2.8.15/uhal/python/src/common/cactus_pycohal.cpp:7:
/home/chevillot/miniconda3/include/python3.12/pytypedefs.h:22:16: note: forward declaration of ‘PyFrameObject’ {aka ‘struct _frame’}
```

Is there already a known fix to be able to compile uhal with python3.12?
Thanks a lot

chevillotn, 2025-02-17:
Hi,
As `distutils` is deprecated now with recent version of python, is there a plan to migrate to other installation tools?
thanks
Nicolas.