ipbus/ipbus-software #337: "Timeout" error when start controlhub service

jingyuluo, 2025-06-12:
Hi, 

I have been working on setting up the communication with FC7 using uhal/controlhub on Alma9 and Alma8.  I followed the instructions in https://ipbus.web.cern.ch/doc/user/html/software/install/yum.html to install uhal and controlhub packages.  The installation was successful, but I'm not able to start the controlhub service, which gave me a "timeout" error. 

Here is the output of "systemctl status controlhub.service": 

```
× controlhub.service - IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)
     Loaded: loaded (/usr/lib/systemd/system/controlhub.service; enabled; preset: disabled)
     Active: failed (Result: timeout) since Wed 2025-06-11 20:10:08 EDT; 13min ago
       Docs: http://ipbus.web.cern.ch/ipbus/doc/index.html
        CPU: 11.106s

Jun 11 20:08:50 controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1206.0> - Initialising the TCP listener (port 10203).
Jun 11 20:10:06  systemd[1]: controlhub.service: start operation timed out. Terminating.
Jun 11 20:10:06  controlhub_start[8945]: Starting ControlHub ........
Jun 11 20:10:06 controlhub[9130]: 2025-06-11 20:10:06.250 [notice] <0.1206.0> - TCP listener shutting down.
Jun 11 20:10:06 controlhub[9130]: 2025-06-11 20:10:06.250 [notice] <0.1205.0> - Device client registry shutting down.
Jun 11 20:10:06  controlhub[9130]: 2025-06-11 20:10:06.251 [notice] <0.1202.0> - Stopping the ControlHub application.
Jun 11 20:10:08 systemd[1]: controlhub.service: Failed with result 'timeout'.
Jun 11 20:10:08 systemd[1]: Failed to start IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications).
Jun 11 20:10:08 systemd[1]: controlhub.service: Consumed 11.106s CPU time.
Jun 11 20:10:08  systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
lines 1-16/16 (END)
```

And here is the ouput of "journalctl -xeu controlhub.service"

```
Jun 11 20:08:35 systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
Jun 11 20:08:35 systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
Jun 11 20:08:35 systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
Jun 11 20:08:35  systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
Jun 11 20:08:36  systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
Jun 11 20:08:36  systemd[1]: Starting IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)...
░░ Subject: A start job for unit controlhub.service has begun execution
░░ Defined-By: systemd
░░ Support: https://wiki.almalinux.org/Help-and-Support
░░ 
░░ A start job for unit controlhub.service has begun execution.
░░ 
░░ The job identifier is 3517.
Jun 11 20:08:50 controlhub_start[8945]: Using config file: /etc/controlhub/default/sys.config
Jun 11 20:08:50 controlhub[9125]: Starting up
Jun 11 20:08:50  controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1202.0> - Starting the ControlHub application (version 2.8.17).
Jun 11 20:08:50  controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1203.0> - ControlHub root supervisor starting now.
Jun 11 20:08:50  controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1204.0> - Initialising the stats server.
Jun 11 20:08:50  controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1205.0> - Initialising the device client registry.
Jun 11 20:08:50  controlhub[9130]: 2025-06-11 20:08:50.786 [notice] <0.1206.0> - Initialising the TCP listener (port 10203).
Jun 11 20:10:06  systemd[1]: controlhub.service: start operation timed out. Terminating.
Jun 11 20:10:06  controlhub_start[8945]: Starting ControlHub ........
Jun 11 20:10:06  controlhub[9130]: 2025-06-11 20:10:06.250 [notice] <0.1206.0> - TCP listener shutting down.
Jun 11 20:10:06  controlhub[9130]: 2025-06-11 20:10:06.250 [notice] <0.1205.0> - Device client registry shutting down.
Jun 11 20:10:06  controlhub[9130]: 2025-06-11 20:10:06.251 [notice] <0.1202.0> - Stopping the ControlHub application.
Jun 11 20:10:08  systemd[1]: controlhub.service: Failed with result 'timeout'.
░░ Subject: Unit failed
░░ Defined-By: systemd
░░ Support: https://wiki.almalinux.org/Help-and-Support
░░ 
░░ The unit controlhub.service has entered the 'failed' state with result 'timeout'.
Jun 11 20:10:08 systemd[1]: Failed to start IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications).
░░ Subject: A start job for unit controlhub.service has failed
░░ Defined-By: systemd
░░ Support: https://wiki.almalinux.org/Help-and-Support
░░ 
░░ A start job for unit controlhub.service has finished with a failure.
░░ 
░░ The job identifier is 3517 and the job result is failed.
Jun 11 20:10:08  systemd[1]: controlhub.service: Consumed 11.106s CPU time.
░░ Subject: Resources consumed by unit runtime
░░ Defined-By: systemd
░░ Support: https://wiki.almalinux.org/Help-and-Support
░░ 
░░ The unit controlhub.service completed and consumed the indicated resources.
Jun 11 20:10:08  systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run/controlhub.pid; please update the uni>
```

I'm wondering if you have any insights on what caused the error and what would be the solution?

Thanks a lot for your help!

tswilliams, 2025-06-12:
Hi!

Unfortunately I don't have any immediate insights - the service starts without errors for me. That said, the output that I see from journalctl on Alma 9 is different from what you've included above:
```
Jun 12 21:38:21 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
Jun 12 21:38:21 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
Jun 12 21:38:21 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
Jun 12 21:38:21 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
Jun 12 21:38:21 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
Jun 12 21:38:22 heplnw061.pp.rl.ac.uk systemd[1]: Starting IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications)...
░░ Subject: A start job for unit controlhub.service has begun execution
░░ Defined-By: systemd
░░ Support: https://wiki.rockylinux.org/rocky/support
░░ 
░░ A start job for unit controlhub.service has begun execution.
░░ 
░░ The job identifier is 87850.
Jun 12 21:38:22 heplnw061.pp.rl.ac.uk controlhub_start[204006]: Using config file: /etc/controlhub/default/sys.config
Jun 12 21:38:23 heplnw061.pp.rl.ac.uk controlhub_start[204006]: Starting ControlHub ... ok
Jun 12 21:38:26 heplnw061.pp.rl.ac.uk systemd[1]: Started IPbus ControlHub server daemon (mediates simultanous IPbus-based hardware access from multiple control applications).
░░ Subject: A start job for unit controlhub.service has finished successfully
░░ Defined-By: systemd
░░ Support: https://wiki.rockylinux.org/rocky/support
░░ 
░░ A start job for unit controlhub.service has finished successfully.
░░ 
░░ The job identifier is 87850.
Jun 12 21:38:26 heplnw061.pp.rl.ac.uk systemd[1]: /usr/lib/systemd/system/controlhub.service:10: PIDFile= references a path below legacy directory /var/run/, updating /var/run/controlhub.pid → /run>
```

Which OS is each of the above quotes from?

Cheers,
Tom

jingyuluo, 2025-06-12:
Thanks a lot for the feedback. Both are from Alma9. Previously I also tried Alma8 but met the same issue -- the controlhub service failed due to a "timeout" error...

mlj5j, 2025-06-17:
We just saw similar behavior after upgrading our OS from Alma 9.5 to 9.6.  It was working fine with 9.5, but after the upgrade it times out.  Haven't found a solution yet.

mlj5j, 2025-06-17:
@tswilliams , thanks for the quick reply.  Directly running `/opt/cactus/bin/controlhub_start` gives a similar timeout.  It shows the status bar "Starting ControlHub ......." and then after quite a while it says that it failed because timeout was exceeded.  I also ran the `/opt/cactus/bin/controlhub_status` script which returned "The ControlHub is down."

I also checked that nothing else was tying up the TCP port 10203 using `sudo lsof -i : 10203`.  There was a beam.smp process (was actually same pid as a previous attempt to start controlhub).  I killed the process and tried again with the same failed result.